mu-editor/mu #1340: Update documentation for changes in the Windows installer

carlosperate, 2021-03-02:
As the packaging system has changed, the new installer does not recognise the previous installation and previous users will end up with two Mu installations in their system.

We need to update the documentation about this, as discussed with @tmontes and @tjguk.

prcutler, 2022-07-17:
I was thinking of helping write this, but I don't know all the history.

Do you see this being included in this page:  https://codewith.mu/en/howto/1.1/install_windows

If so, what if I was to change Step 2 to be "Uninstall Mu" with some language about if Mu hasn't been installed yet, continue.  And then continue on with Step 3 being Run the Installer.

If not, would you prefer to see "Upgrading Mu" as its own section separate from installation?

I think with a little direction I might be able to help.

Thanks!

carlosperate, 2022-10-08:
Hi @prcutler, thanks for all your contributions and sorry it's taken long to reply to some of these.

Yes, I think there is three steps here:
- Update the current https://codewith.mu/en/howto/1.1/install_windows page with the steps and screenshots from the current installer (I assume the window looks different, if it is the same, then there isn't a need to do this)
- Add a yellow box at the top with a warning indicating that if the user has any 1.1-beta version installed, they should uninstall it first
    - I think for now it is okay to only include it as a warning, as as soon as v1.2 is out it won't be required any more.
- Add a new section at the bottom to include instructions on how to install Mu for "all users" (the default is only for the current user), as indicated in https://github.com/mu-editor/mu/issues/1843#issuecomment-961520506