mu-editor/mu #1343: Mu got stuck during startup

carlosperate, 2021-03-02:
- Running Mu from source code in current master
- Last time I opened Mu a few days earlier all worked fine (log looks clean)
- Today I try to open Mu with `python run.py` and it got "stuck" in the loading screen
    - Looking at the venvs the current one looks valid
    - While Mu was stuck there wasn't any new venv created
- After 4min I exited with ctrl+c
- The next time I run Mu (immediately after this) it worked fine again
   - This session didn't create any new venv, it run from the existing one

This is the log with the error, unfortunately it only shows the keyboard interrupt:
```
2021-03-02 14:50:15,603 - root:172(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.1
2021-03-02 14:50:15,605 - root:173(run) INFO: uname_result(system='Darwin', node='Carloss-MacBook-Pro.local', release='20.3.0', version='Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-03-02 14:50:15,605 - root:174(run) INFO: Python path: ['/Users/microbit-carlos/workspace/mine/mu-upstream', '/Users/microbit-carlos/.pyenv/versions/3.8.2/lib/python38.zip', '/Users/microbit-carlos/.pyenv/versions/3.8.2/lib/python3.8', '/Users/microbit-carlos/.pyenv/versions/3.8.2/lib/python3.8/lib-dynload', '/Users/microbit-carlos/workspace/mine/mu-upstream/.venv/lib/python3.8/site-packages', '/Users/microbit-carlos/workspace/mine/mu-upstream', '/Users/microbit-carlos/workspace/mine/mu-upstream/.venv/lib/python3.8/site-packages/IPython/extensions']
2021-03-02 14:50:15,605 - root:175(run) INFO: Language code: en_GB
2021-03-02 14:50:15,935 - mu.virtual_environment:379(ensure_and_create) DEBUG: Checking venv; attempt #1
2021-03-02 14:50:15,936 - mu.virtual_environment:416(ensure_path) INFO: Virtual Environment found at /Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210224-102310
2021-03-02 14:50:15,936 - mu.virtual_environment:427(ensure_interpreter) INFO: Interpreter found at /Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210224-102310/bin/python
2021-03-02 14:50:15,990 - mu.virtual_environment:456(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-03-02 14:50:15,992 - mu.virtual_environment:483(ensure_pip) INFO: Pip found at /Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210224-102310/bin/pip
2021-03-02 14:50:15,993 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import pygame
2021-03-02 14:50:16,377 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import numpy
2021-03-02 14:50:16,686 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import pgzero
2021-03-02 14:50:16,717 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import flask
2021-03-02 14:50:16,956 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import serial
2021-03-02 14:50:16,991 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import qtconsole
2021-03-02 14:50:17,021 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import nudatus
2021-03-02 14:50:17,068 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import esptool
2021-03-02 14:54:16,588 - root:106(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Users/microbit-carlos/workspace/mine/mu-upstream/mu/app.py", line 74, in set_frame
    def set_frame(self):
KeyboardInterrupt
2021-03-02 14:54:16,593 - mu.settings:169(save) DEBUG: Saving to /Users/microbit-carlos/Library/Application Support/mu/session.json
2021-03-02 14:54:16,595 - mu.settings:169(save) DEBUG: Saving to /Users/microbit-carlos/Library/Application Support/mu/venv.json
2021-03-02 14:54:19,905 - root:172(run) INFO: 
```

For comparison, the next session was fine:
```
2021-03-02 14:54:19,905 - root:172(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.1
2021-03-02 14:54:19,905 - root:173(run) INFO: uname_result(system='Darwin', node='Carloss-MacBook-Pro.local', release='20.3.0', version='Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-03-02 14:54:19,905 - root:174(run) INFO: Python path: ['/Users/microbit-carlos/workspace/mine/mu-upstream', '/Users/microbit-carlos/.pyenv/versions/3.8.2/lib/python38.zip', '/Users/microbit-carlos/.pyenv/versions/3.8.2/lib/python3.8', '/Users/microbit-carlos/.pyenv/versions/3.8.2/lib/python3.8/lib-dynload', '/Users/microbit-carlos/workspace/mine/mu-upstream/.venv/lib/python3.8/site-packages', '/Users/microbit-carlos/workspace/mine/mu-upstream', '/Users/microbit-carlos/workspace/mine/mu-upstream/.venv/lib/python3.8/site-packages/IPython/extensions']
2021-03-02 14:54:19,905 - root:175(run) INFO: Language code: en_GB
2021-03-02 14:54:20,142 - mu.virtual_environment:379(ensure_and_create) DEBUG: Checking venv; attempt #1
2021-03-02 14:54:20,143 - mu.virtual_environment:416(ensure_path) INFO: Virtual Environment found at /Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210224-102310
2021-03-02 14:54:20,143 - mu.virtual_environment:427(ensure_interpreter) INFO: Interpreter found at /Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210224-102310/bin/python
2021-03-02 14:54:20,182 - mu.virtual_environment:456(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-03-02 14:54:20,183 - mu.virtual_environment:483(ensure_pip) INFO: Pip found at /Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210224-102310/bin/pip
2021-03-02 14:54:20,183 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import pygame
2021-03-02 14:54:20,327 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import numpy
2021-03-02 14:54:20,570 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import pgzero
2021-03-02 14:54:20,609 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import flask
2021-03-02 14:54:20,853 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import serial
2021-03-02 14:54:20,890 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import qtconsole
2021-03-02 14:54:20,922 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import nudatus
2021-03-02 14:54:20,975 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import esptool
2021-03-02 14:54:22,172 - mu.logic:758(__init__) INFO: Setting up editor.
2021-03-02 14:54:22,173 - mu.logic:778(__init__) INFO: Log directory: /Users/microbit-carlos/Library/Logs/mu
2021-03-02 14:54:22,173 - mu.logic:779(__init__) INFO: Data directory: /Users/microbit-carlos/Library/Application Support/mu
2021-03-02 14:54:22,173 - mu.logic:793(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2021-03-02 14:54:22,173 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/microbit-carlos/mu_code from settings file
2021-03-02 14:54:22,359 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/microbit-carlos/mu_code from settings file
2021-03-02 14:54:22,359 - mu.logic:1523(change_mode) INFO: Workspace directory: /Users/microbit-carlos/mu_code
2021-03-02 14:54:22,359 - mu.logic:867(restore_session) DEBUG: <SessionSettings from /Users/microbit-carlos/Library/Application Support/mu/session.json>
```

It must have gotten stuck between these two logs:
```
2021-03-02 14:54:20,975 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import esptool
2021-03-02 14:54:22,172 - mu.logic:758(__init__) INFO: Setting up editor.
```

tjguk, 2021-03-02:
Well that is annoyingly weird. I *really* don't want to have to implement some sort of timeout on the import tests; that would add another layer to an already quite complicated process.

If there's a possibility that it's just the esptool (which is a bit of an added extra) which is misbehaving, we could just cut it out. The mode code already checks for its existence, so of all the baseline modules, its the one whose absence matters the least.

carlosperate, 2021-03-03:
Unfortunately, no matter how many times I've tried, I was able to replicate it only once more (trying with and without the esptool dependency):

```
2021-03-02 16:50:30,719 - root:172(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.1
2021-03-02 16:50:30,719 - root:173(run) INFO: uname_result(system='Darwin', node='Carloss-MacBook-Pro.local', release='20.3.0', version='Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-03-02 16:50:30,719 - root:174(run) INFO: Python path: ['/Users/microbit-carlos/workspace/mine/mu-upstream', '/Users/microbit-carlos/.pyenv/versions/3.8.2/lib/python38.zip', '/Users/microbit-carlos/.pyenv/versions/3.8.2/lib/python3.8', '/Users/microbit-carlos/.pyenv/versions/3.8.2/lib/python3.8/lib-dynload', '/Users/microbit-carlos/workspace/mine/mu-upstream/.venv/lib/python3.8/site-packages', '/Users/microbit-carlos/workspace/mine/mu-upstream', '/Users/microbit-carlos/workspace/mine/mu-upstream/.venv/lib/python3.8/site-packages/IPython/extensions']
2021-03-02 16:50:30,719 - root:175(run) INFO: Language code: en_GB
2021-03-02 16:50:30,936 - mu.virtual_environment:379(ensure_and_create) DEBUG: Checking venv; attempt #1
2021-03-02 16:50:30,936 - mu.virtual_environment:416(ensure_path) INFO: Virtual Environment found at /Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210224-102310
2021-03-02 16:50:30,936 - mu.virtual_environment:427(ensure_interpreter) INFO: Interpreter found at /Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210224-102310/bin/python
2021-03-02 16:50:30,972 - mu.virtual_environment:456(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-03-02 16:50:30,973 - mu.virtual_environment:483(ensure_pip) INFO: Pip found at /Users/microbit-carlos/Library/Application Support/mu/mu_venv-38-20210224-102310/bin/pip
2021-03-02 16:50:30,973 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import pygame
2021-03-02 16:50:31,080 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import numpy
2021-03-02 16:50:31,251 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import pgzero
2021-03-02 16:50:31,281 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import flask
2021-03-02 16:50:31,443 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import serial
2021-03-02 16:50:31,473 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import qtconsole
2021-03-02 16:50:31,499 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import nudatus
2021-03-02 16:50:31,539 - mu.virtual_environment:468(ensure_key_modules) DEBUG: Trying to import esptool
2021-03-02 16:52:58,328 - root:106(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Users/microbit-carlos/workspace/mine/mu-upstream/mu/app.py", line 74, in set_frame
    def set_frame(self):
KeyboardInterrupt
2021-03-02 16:52:58,335 - mu.settings:169(save) DEBUG: Saving to /Users/microbit-carlos/Library/Application Support/mu/session.json
2021-03-02 16:52:58,339 - mu.settings:169(save) DEBUG: Saving to /Users/microbit-carlos/Library/Application Support/mu/venv.json
```

However I don't think this is related to the venv, I think it got "stuck" somewhere else before the `mu.logic:758(__init__) INFO: Setting up editor.`.
One thing to notice is that the keyboard interrupt was trigger "in"  `AnimatedSplash.set_frame()` method, and if it was stuck anywhere in the venv init stuff the stack trace would have very likely point somewhere there. Ctrl+C while the app is running normally and it always point there, but during the init it shoudn't.

Just to give it a few quick tests, pressing ctrl+c during the startup process during a normal session shows the stack at different initialisation steps. Once I was able to stop the process exactly at `VirtualEnvironment.ensure_key_modules()`, which is the place that checks if the modules are available, so I think it's safe to assume this step had finished when the problem happened.

carlosperate, 2021-08-11:
This area has been updated a couple of times since I opened this issue and haven't been able to replicate since, so I'll close as fixed 👍

closed 2021-08-11 by carlosperate