mu-editor/mu #1346: Mu-editor won't launch in macOS Catalina with ADAFruit feather MO Express or GEMMA Mo Board

ritgrad83, 2021-03-03:
Tried launching MU-Editor with CircuitPython boards attached - fails to launch - Feather MO Express and Gemma MO from Adafruit. Both running latest Circuit Python versions. This machine is a 2012 MacBook Pro running latest Catalina

**Here is the log**

`Starting Mu 1.0.3
2021-03-03 14:16:16,759 - root:113(run) INFO: uname_result(system='Darwin', node='rpmmbp747.fios-router.home', release='19.6.0', version='Darwin Kernel Version 19.6.0: Tue Jan 12 22:13:05 PST 2021; root:xnu-6153.141.16~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-03-03 14:16:16,759 - root:114(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2021-03-03 14:16:16,759 - root:115(run) INFO: Language code: en_US
2021-03-03 14:16:17,060 - mu.logic:543(__init__) INFO: Setting up editor.
2021-03-03 14:16:17,061 - mu.logic:562(__init__) INFO: Settings path: /Users/richardmeagher/Library/Application Support/mu/settings.json
2021-03-03 14:16:17,061 - mu.logic:563(__init__) INFO: Session path: /Users/richardmeagher/Library/Application Support/mu/session.json
2021-03-03 14:16:17,061 - mu.logic:564(__init__) INFO: Log directory: /Users/richardmeagher/Library/Logs/mu
2021-03-03 14:16:17,061 - mu.logic:565(__init__) INFO: Data directory: /Users/richardmeagher/Library/Application Support/mu
2021-03-03 14:16:17,078 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2021-03-03 14:16:17,078 - mu.logic:590(setup) DEBUG: Creating directory: /Volumes/CIRCUITPY/images
2021-03-03 14:16:17,092 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.3', 'console_scripts', 'mu-editor')()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 145, in run
    editor.setup(setup_modes(editor, editor_window))
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/logic.py", line 591, in setup
    os.makedirs(images_path)
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/os.py", line 220, in makedirs
PermissionError: [Errno 1] Operation not permitted: '/Volumes/CIRCUITPY/images'
`
I checked Settings > Security and Privacy - and gave Mu-Editor All Disk Access

The same boards launched Mu-Editor without issues on my other Mac running Big Sur and was able to save python updates. 

**Here is the log file from the other machine (MacBook Pro - Big Sur)that launched Mu-editor**

Starting Mu 1.0.1
2021-03-03 13:56:56,372 - root:113(run) INFO: uname_result(system='Darwin', node='Richards-MBP.fios-router.home', release='20.3.0', version='Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-03-03 13:56:56,373 - root:114(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2021-03-03 13:56:56,373 - root:115(run) INFO: Language code: en
2021-03-03 13:56:56,679 - mu.logic:540(__init__) INFO: Setting up editor.
2021-03-03 13:56:56,680 - mu.logic:558(__init__) INFO: Settings path: /Users/richardmeagher/Library/Application Support/mu/settings.json
2021-03-03 13:56:56,681 - mu.logic:559(__init__) INFO: Session path: /Users/richardmeagher/Library/Application Support/mu/session.json
2021-03-03 13:56:56,681 - mu.logic:560(__init__) INFO: Log directory: /Users/richardmeagher/Library/Logs/mu
2021-03-03 13:56:56,681 - mu.logic:561(__init__) INFO: Data directory: /Users/richardmeagher/Library/Application Support/mu
2021-03-03 13:56:56,694 - mu.logic:574(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2021-03-03 13:56:56,825 - mu.logic:1112(change_mode) INFO: Workspace directory: /Users/richardmeagher/mu_code
2021-03-03 13:56:56,826 - mu.logic:623(restore_session) INFO: Restoring session from: /Users/richardmeagher/Library/Application Support/mu/session.json
2021-03-03 13:56:56,826 - mu.logic:624(restore_session) DEBUG: {'theme': 'day', 'mode': 'adafruit', 'paths': ['/Volumes/CIRCUITPY/code.py'], 'envars': [], 'minify': False, 'microbit_runtime': ''}
2021-03-03 13:56:56,826 - mu.logic:708(_load) INFO: Loading script from: /Volumes/CIRCUITPY/code.py
2021-03-03 13:56:56,830 - mu.logic:271(read_and_decode) DEBUG: Trying to decode with utf-8
2021-03-03 13:56:56,830 - mu.logic:274(read_and_decode) INFO: Decoded with utf-8
2021-03-03 13:56:56,831 - mu.logic:287(read_and_decode) DEBUG: Detected newline '\n'
2021-03-03 13:56:56,831 - mu.logic:782(_load) DEBUG: # Write your code here :-)
import board
import digitalio
import time
 
led = digitalio.DigitalInOut(board.D13)
led.direction = digitalio.Direction.OUTPUT
 
while True:
    led.value = True
    time.sleep(0.1)
    led.value = False
    time.sleep(0.1)
2021-03-03 13:56:56,924 - mu.logic:645(restore_session) INFO: Loaded files.
2021-03-03 13:56:56,924 - mu.logic:649(restore_session) INFO: User defined environment variables: []
2021-03-03 13:56:56,924 - mu.logic:653(restore_session) INFO: Minify scripts on micro:bit? False
2021-03-03 13:56:56,949 - mu.logic:1112(change_mode) INFO: Workspace directory: /Volumes/CIRCUITPY

carlosperate, 2021-03-04:
Hi @ritgrad83,
Have you try the "Disk Permissions in OSX Catalina" steps from the installation instructions?
https://codewith.mu/en/howto/1.0/install_macos#catalina-permissions.

ritgrad83, 2021-03-04:
Yes - I went into System Preferences > Security and Privacy, unlocked the padlock, scroll down to Full Disk Access 
[image]
 and made sure Mu-Editor

carlosperate, 2021-03-16:
This should be fixed in the latest release: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.2 🎉 

I'll close this as resolved, but please do reopen it if you are still having issues with the v1.1.0-beta.2.

closed 2021-03-16 by carlosperate

ritgrad83, 2021-04-09:
I just downloaded the latest 1.1.0-beta.3 - and still having issues. I created a bug report but here is the log file 
[mu.log](https://github.com/mu-editor/mu/files/6286721/mu.log)