mu-editor/mu #1353: CircuitPython REPL text poorly rendered on external monitor

BiffoBear, 2021-03-06:
Mac OS X 11.2.2, MacBook Pro (15inch, 2018)
Mu 1.1.0.alpha.2

The CircuitPython REPL text renders well on my laptop screen. When I move the window to an external monitor (2560 x 1440) then in the REPL, underscores are not rendered at all.

dybber, 2021-03-06:
This sounds like the same issue as #1098. Could you try if the issue persist between zoom-levels?

I've tried debugging it previously, but it seems more like a problem with the underlying libraries we use (Qt) or the font (Source Code Pro), as I can replicate it with very small Qt-programs.

BiffoBear, 2021-03-07:
Yes, it seems to be something to do with the way Mac OS X renders  fonts on different display resolutions. The REPL font renders poorly on my 2560 x 1440 external monitor at all sizes. The underscore is missing only at the default size on the external monitor. It is there in other sizes.

Would you prefer me to close this issue and comment on #1098?

xt0fer, 2021-06-15:
I have several other external monitors on macOS (M1 Mac mini), and yes, the underscores are a problem on all three. (But NOT on any native resolution of a M1 MacBook Air). 

Is there any kind of way to change the font used? we might try some others, and see if any of them are better in these cases?

carlosperate, 2021-06-15:
Thanks for the report @BiffoBear and @xt0fer, as this looks the same https://github.com/mu-editor/mu/issues/1098, I'll merge the this issue there to keep things tidy.
We can continue the conversation over there 👍

closed 2021-06-15 by carlosperate