mu-editor/mu #1355: Mu opens, but does not display a window / respond on Mac OS 11.2.2 Big Sur

whogben, 2021-03-08:
I downloaded the latest release of Mu, copied it into applications folder, and went through the process of opening it, going to security preferences to approve it, and re-opening it.

Mu application thinks it opens - it appears in the dock, but it never displays a window. When right-clicked, it says "Application Not Responding" and when force quit it says "You have forced python3 to quit."

I have tried restarting, and opening Mu with and without a circuitpython board connected.

carlosperate, 2021-03-08:
Hi @whogben, have you tried the steps listed in https://github.com/mu-editor/mu/discussions/1349?

carlosperate, 2021-04-11:
Hi @whogben, have you been able to resolve this issue? There are new releases as well if you'd like to try those.
https://github.com/mu-editor/mu/releases

carlosperate, 2021-08-11:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for the feedback!

closed 2021-08-11 by carlosperate