mu-editor/mu #1359: Win8.1 clean installation from b2 installer Crash Report

carlosperate, 2021-03-15:
First report from the new splash screen + crash reporter.

Using the current 32-bit (test) installer.

Version: 1.1.0.beta.2
Locale: en_US
Platform: Windows 8.1 6.3.9600 AMD64

```
Traceback (most recent call last):
  File "C:\Users\IEUser\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\IEUser\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\IEUser\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 460, in ensure_and_create
    "Unable to create a working virtual environment."
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.

```

carlosperate, 2021-03-15:
Full log with 2 attemps at opening Mu:

```
2021-03-15 00:09:20,242 - root:259(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.2
2021-03-15 00:09:20,242 - root:260(run) INFO: uname_result(system='Windows', node='IE11WIN8_1', release='8.1', version='6.3.9600', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 10, GenuineIntel')
2021-03-15 00:09:20,242 - root:261(run) INFO: Python path: ['C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\python37.zip', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.7.egg', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.7.egg', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-03-15 00:09:20,242 - root:262(run) INFO: Language code: en_US
2021-03-15 00:09:20,242 - mu.settings:220(load) WARNING: No settings file found at C:\Users\IEUser\AppData\Local\python\mu\settings.json; skipping
2021-03-15 00:09:20,242 - mu.settings:220(load) WARNING: No settings file found at C:\Users\IEUser\AppData\Local\python\mu\session.json; skipping
2021-03-15 00:09:20,570 - mu.virtual_environment:434(ensure_and_create) INFO: Added log handler.
2021-03-15 00:09:20,570 - mu.virtual_environment:439(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-03-15 00:09:22,211 - mu.virtual_environment:479(ensure_path) ERROR: C:\Users\IEUser\AppData\Local\python\mu\mu_venv-37-20210315-000905 does not exist.
2021-03-15 00:09:22,226 - mu.virtual_environment:443(ensure_and_create) DEBUG: Virtual environment not present or correct.
2021-03-15 00:09:22,226 - mu.virtual_environment:446(ensure_and_create) DEBUG: Creating new virtual environment at C:\Users\IEUser\AppData\Local\python\mu\mu_venv-37-20210315-000922.
2021-03-15 00:09:22,226 - mu.virtual_environment:378(relocate) DEBUG: Virtual environment set up mu_venv-37-20210315-000922 at C:\Users\IEUser\AppData\Local\python\mu\mu_venv-37-20210315-000922
2021-03-15 00:09:22,226 - mu.virtual_environment:578(create) INFO: Creating virtualenv: C:\Users\IEUser\AppData\Local\python\mu\mu_venv-37-20210315-000922
2021-03-15 00:09:22,226 - mu.virtual_environment:579(create) INFO: Virtualenv name: mu_venv-37-20210315-000922
2021-03-15 00:09:38,257 - mu.virtual_environment:631(install_baseline_packages) INFO: Installing baseline packages.
2021-03-15 00:09:38,257 - mu.virtual_environment:635(install_baseline_packages) INFO: C:\Users\IEUser\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels exists
2021-03-15 00:09:38,257 - mu.virtual_environment:229(install) DEBUG: About to pip install: ['C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\backcall-0.2.0-py2.py3-none-any.whl', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\bitstring-3.1.7-py3-none-any.whl', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\cffi-1.14.5-cp38-cp38-win_amd64.whl', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\click-7.1.2-py2.py3-none-any.whl', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\colorama-0.4.4-py2.py3-none-any.whl', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\cryptography-3.4.6-cp36-abi3-win_amd64.whl', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\decorator-4.4.2-py2.py3-none-any.whl', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\ecdsa-0.16.1-py2.py3-none-any.whl', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\esptool-3.0-py3-none-any.whl', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\Flask-1.1.2-py2.py3-none-any.whl', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\ipykernel-5.5.0-py3-none-any.whl', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\ipython-7.21.0-py3-none-any.whl', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\ipython_genutils-0.2.0-py2.py3-none-any.whl', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\itsdangerous-1.1.0-py2.py3-none-any.whl', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\jedi-0.18.0-py2.py3-none-any.whl', 'C:\\Users\\IEUser\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\mu\\wheels\\Jinja2-2.11.3-py2.py3-none-any.whl', 'C: [... cut ...]
  File "C:\Users\IEUser\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 460, in ensure_and_create
    "Unable to create a working virtual environment."
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.
2021-03-15 00:15:27,047 - mu.settings:169(save) DEBUG: Saving to C:\Users\IEUser\AppData\Local\python\mu\session.json
2021-03-15 00:15:27,048 - mu.settings:169(save) DEBUG: Saving to C:\Users\IEUser\AppData\Local\python\mu\venv.json

```

tjguk, 2021-03-15:
Well, good news that the crash reporting works :) Obviously not such good news that it had something to work on!

The problem is -- as @carlosperate says -- that we're building the Win32 package with win64 wheels. Cutting the Gordian Knot somewhat, I think we agreed a few weeks back that we could basically ditch the win32 installers since, even on reasonably old hardware, you're still likely to be running 64-bit.

carlosperate, 2021-03-15:
Oh yeah, that's right. This is also a good point to remove the 32 bit installer, so if we have people asking for it we can reconsider it.

ntoll, 2021-03-15:
@carlosperate I like that solution. Removing it from the `Makefile` is really all we need do, right..?

ntoll, 2021-03-15:
Also, yay for the crash reporter. :-D

tjguk, 2021-03-15:
> 
> 
> @carlosperate I like that solution. Removing it from the `Makefile` is really all we need do, right..?

... and the corresponding code in `make.py`

carlosperate, 2021-03-29:
Action: To remove the `win32` commands from make.py/Makefile