mu-editor/mu #1363: Splash screen stays in front of everything

tjguk, 2021-03-16:
The new splash screen stays in front of every other window... on every desktop! This is probably not best practice :)

carlosperate, 2021-03-16:
Do you mean you are getting a splash screen on both monitors?
You can still click on other windows to bring them to the front, no?

tjguk, 2021-03-17:
No, not on both monitor. I have two virtual desktops, each running across two monitors. The splash stays (in the same monitor) on top of all other windows, regardless of which desktop I switch to.

tjguk, 2021-03-20:
Just did a quick trial by the simple expedient of removing the `Qt.WindowStaysOnTopHint`. Unfortunately (at least one Windows) the Splash Screen Window isn't a "top-level" window. So, once you've put something else in front of it, it won't show up again in an Alt-Tab sequence.

carlosperate, 2021-04-08:
I can confirm as well that on Windows the beta.3 splash screen stays on top of everything, which can a be bit annoying.
This is not an issue in macOS, where clicking or alt-tab a different window places it on top of the splash.

carlosperate, 2021-08-04:
I can confirm this is also an issue in Ubuntu 20.04, and still not an issue on macOS 11.

@ntoll can you check if this is also a problem in Debian?

closed 2021-09-06 by carlosperate