mu-editor/mu #1366: Crash Report

PeetersY, 2021-03-16:
Version: 1.1.0.beta.2
Locale: en_US
Platform: Linux 5.8.0-45-generic #51~20.04.1-Ubuntu SMP Tue Feb 23 13:46:31 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/peetersy/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 441, in ensure_and_create
    self.ensure()
  File "/home/peetersy/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "/home/peetersy/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: /home/peetersy/.local/share/mu/mu_venv-38-20210316-203539 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/peetersy/.local/lib/python3.8/site-packages/mu/app.py", line 162, in run
    raise ex
  File "/home/peetersy/.local/lib/python3.8/site-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/peetersy/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "/home/peetersy/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 596, in create
    self.install_baseline_packages()
  File "/home/peetersy/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 649, in install_baseline_packages
    wheels.download()
  File "/home/peetersy/.local/lib/python3.8/site-packages/mu/wheels/__init__.py", line 82, in download
    process = subprocess.run(
  File "/usr/lib/python3.8/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'download', '--destination-directory', '/home/peetersy/.local/lib/python3.8/site-packages/mu/wheels', 'git+https://github.com/mu-editor/pgzero.git@mu-wheel']' returned non-zero exit status 1.

```

tjguk, 2021-03-16:
Thanks for reporting this @PeetersY -- can I ask: are you running this from a development virtual environment? You're on Linux, so it won't be a downloaded installer.

carlosperate, 2021-03-16:
Is git installed? If you run `git --version` in the terminal do you get a valid output?

@tjguk If it's not added to the log yet, would there be a way add the subprocess stdout/stderr?
@ntoll for these crash reports it would be nice to get the full log as well? Should we add it to the reporter? Or is it likely too much text?
If adding it is to feasible (or maybe not something we want to post automatically), maybe we should include instructions on how to find it and manually add it to the GH issue?

tjguk, 2021-03-16:
@carlosperate fairly sure the subprocess output is already there but I'll check. Second the notion of making it easy to add the logs to the GH issue.

carlosperate, 2021-03-29:
@PeetersY there is a new release that we think has fixed this issue. Would be able to install the latest version and try again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.3

carlosperate, 2021-04-06:
As we believe this issue has been solved in the latest beta release I'll close this. But please do feel free to reopen if it still doesn't work. Thanks again for the report!

closed 2021-04-06 by carlosperate