mu-editor/mu #1367: Crash Report

Erlan-w, 2021-03-16:
[image]
[image]

Version: 1.1.0.beta.2
Locale: en_ID
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Erlan W\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\Erlan W\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Erlan W\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 441, in ensure_and_create
    self.ensure()
  File "C:\Users\Erlan W\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 469, in ensure
    self.ensure_interpreter_version()
  File "C:\Users\Erlan W\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 522, in ensure_interpreter_version
    process = subprocess.run(
  File "C:\Users\Erlan W\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['C:\\Users\\Erlan W\\AppData\\Local\\python\\mu\\mu_venv-38-20210317-035053\\scripts\\python.exe', '-c', 'import sys; print("%s%s" % sys.version_info[:2])']' returned non-zero exit status 101.

```

Erlan-w, 2021-03-16:
hy, I am Erlan Wahyu, I am new to learning this programming, when I try to install Mu editor it crashes like this, is there a solution I can use?

tjguk, 2021-03-16:
Thanks for reporting, @Erlan-w. I'll try to reproduce it locally. My very initial guess is that we're not handling the fact that your username has a space in it ("Erlan W").

If you have the possibility, can you create/use a different user which has no space in the username?

Erlan-w, 2021-03-17:
okay, thank you, the solution is very helpful, now I can open mu editor.

tjguk, 2021-03-17:
Thanks for the feedback @Erlan-w. I'll close this report and raise an issue to track the failure to support user paths with embedded spaces.

closed 2021-04-05 by tjguk