mu-editor/mu #1369: Implicit dependency on git

carlosperate, 2021-03-16:
We currently have a dependency on git for installing the pgzero package, as we install it from our fork.

In the future when we change that to only fetch wheels from PyPI, we can still be depend on git if any of the packages tries to fetch a package from a repo. But to be fair, that is probably no different than a package trying to compile a C extension and not having the toolchain.

Technically we could configure the "pip install" to only use wheels, but that might be worse as we could be stopping some users from successfully installing dependencies if they have the toolchain required.

ntoll, 2021-03-17:
So, @lordmauve has made a new release of PGZero which includes a version of the patch I applied. We should be able to return to using PyPI as the source of that package. I would change things up right now, but I have an afternoon of meetings starting in a second.

ntoll, 2021-03-17:
Actually, #1376 (currently untested) should fix this.

tjguk, 2021-03-20:
I've just merged #1376 which should remove the immediate need. I don't know that there's much else we can do. As you say, @carlosperate, it's really now different from a user who hasn't, eg, installed `build-essentials` and need to build from source.

This was an outlandish case and I can't see that we'll often be specifying a git-based package, so I propose to close this for now.

carlosperate, 2021-03-22:
Yeah, we can consider this done. Thanks @lordmauve, @ntoll and @tjguk!

closed 2021-03-22 by carlosperate