mu-editor/mu #1372: Virtual Environment fails on Windows when user path contains a space

tjguk, 2021-03-17:
(First reported via https://github.com/mu-editor/mu/issues/1367)

On the surface it appears that the check for the Python interpreter in `VirtualEnvironment.ensure` fails if the user's installation path contains a space -- typically because their username contains a space. Not verified yet, but the OP in that report confirmed that re-installing under a different username solved the problem.

tjguk, 2021-03-18:
I can't reproduce this at the moment. I've tried creating a venv in a path with spaces in it, and it works fine. (And the underlying `subprocess.run` runs fine standalone). 

I'll try to find a machine where I can create another username altogether to see if it's reproducible that way. Two users have reported that switching to a spaceless username fixes the problem, so unless it's a coincidence...

tjguk, 2021-03-19:
[Reproducing the original trackback from #1367]

```
Traceback (most recent call last):
  File "C:\Users\Erlan W\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\Erlan W\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Erlan W\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 441, in ensure_and_create
    self.ensure()
  File "C:\Users\Erlan W\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 469, in ensure
    self.ensure_interpreter_version()
  File "C:\Users\Erlan W\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 522, in ensure_interpreter_version
    process = subprocess.run(
  File "C:\Users\Erlan W\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['C:\\Users\\Erlan W\\AppData\\Local\\python\\mu\\mu_venv-38-20210317-035053\\scripts\\python.exe', '-c', 'import sys; print("%s%s" % sys.version_info[:2])']' returned non-zero exit status 101.
```

tjguk, 2021-03-19:
I'm completely unable to reproduce this error. I've created a username with spaces in and run, both from dev and from the installer. And each time it works fine. I can't believe it's just a coincidence that two users, each with a space in their username, receive the same error, which is fixed when they install from a user without a space.

But I can't explain how it's getting there...

tjguk, 2021-03-21:
The only small ray of light here is that both of the two reports show an exit error code of 101, rather than just 1 which would be usual "something went wrong" exitcode. I'll try to research where that 101 is coming from.

tjguk, 2021-03-26:
Re-opening as it's not yet clearly fixed

tjguk, 2021-04-09:
Just a note: even assuming that my theory is correct -- that the behaviour changed in later Windows releases -- we're hampered by a bug in the platform module which is misreporting the Windows version. My platform.uname reports build 19041 (as does the user in #1452) but I'm actually running 19042, as reported by `ver` and elsewhere.

I know why this bug is present but it's not clear how soon it will be fixed in Python. I'll probably slap a PR together to parse the output from `ver` to get the correct version number. At least the water will be very slightly less muddy then :)

carlosperate, 2021-04-09:
I cannot replicate the same issue in a Windows 8.1 VM:

```
C:\Users\test user>ver

Microsoft Windows [Version 6.3.9600]
```

Do you think this an issue in Python or Windows?

tjguk, 2021-04-09:
It's Python. I can't lay my hands on the actual bug report right now, but basically the `platform` module, rather than parsing the output of `ver` or whatever, relies on the version number of a key DLL (probably `kernel32` or something). But now that Windows is doing much more frequent "build version" updates, they've actually released a build-version update which didn't touch that DLL. So its version is still 19041 even though the overall Windows version is 19042. At least, that's my understanding.

So it's not that the `platform.uname` is always wrong or whatever. It depends on an unexpected discrepancy between the overall Windows version and the version of a specific DLL.

carlosperate, 2021-04-09:
Sorry, I meant the issue with the spaces in the path? 😅
Should we be able to replicate it with an older version of Windows? If it's Windows related, is it related to the "windows version" or the "installed updated" (maybe it's the same thing and this question is moot)?

tjguk, 2021-04-09:
> Sorry, I meant the issue with the spaces in the path?

Oh, I see. Fundamentally I *think* it's a Windows thing. At least: the way Python `subprocess` is invoking Win API `CreateProcess` might be triggering a known issue when spaces are in the path. Since I couldn't reproduce it on either of my Win10 boxes -- both of which are fully patched -- I assumed it was maybe some behaviour which had changed in recent Win10 versions.

But if you're *not* able to reproduce on Win8.1 then that makes that theory very unlikely.

When I get some time I propose to put together a mini-app which should test the underlying command and whicih users can run on their boxes to see if it reproduces the issue. Beyond that, I'm not sure what to do.

carlosperate, 2021-04-09:
> But if you're not able to reproduce on Win8.1 then that makes that theory very unlikely.

It's quite possible this virtual machines have been patched recently. Today when I tested it actually force a windows update on shutdown, so I wouldn't be surprised if they are more or less up-to-date in that respect.

I can try replication the vm from the original source and stopping any windows updates to see if that triggers it.

tjguk, 2021-04-09:
Hmmmm. If my theory holds any water (and it's looking increasingly unlikely) then the updates would probably be *well* after this date. I'll try to put together a set of test cases and see if users can run them.

carlosperate, 2021-12-17:
Issue https://github.com/mu-editor/mu/issues/1926 contains more information (only noticed the relationship by the end of putting that together), so let's continue the conversation over there to keep the milestones slim.

PR #1390 (2021-03-21): Space in username 
(Fixes #1372)

Attempt to workaround the non-reproducible bug where a username containing a space causes the venv interpreter to fail
closed 2021-12-17 by carlosperate