mu-editor/mu #1377: Crash Report

RandoSY, 2021-03-17:
Version: 1.1.0.beta.2
Locale: en_US
Platform: Linux 5.8.1-050801-generic #202008111432 SMP Tue Aug 11 14:34:42 UTC 2020 x86_64

```
Traceback (most recent call last):
  File "/home/rasyoung/.local/lib/python3.8/site-packages/mu/app.py", line 162, in run
    raise ex
  File "/home/rasyoung/.local/lib/python3.8/site-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/rasyoung/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 459, in ensure_and_create
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.

```

tjguk, 2021-03-17:
Thanks for the crash report, @RandoSY -- are you in a position to retrieve the log files for us? For a Linux user they should be in:

`~/.cache/mu/log/mu.log`

RandoSY, 2021-03-17:
> This message was sent from a non-IU address. Please exercise caution when
> clicking links or opening attachments from external sources.

Hi Tim,

Here u go!

Traceback (most recent call last):
  File "/home/rasyoung/.local/lib/python3.8/site-packages/mu/app.py", line
162, in run
    raise ex
  File "/home/rasyoung/.local/lib/python3.8/site-packages/mu/app.py", line
149, in run
    venv.ensure_and_create(self.display_text)
  File
"/home/rasyoung/.local/lib/python3.8/site-packages/mu/virtual_environment.py",
line 459, in ensure_and_create
    raise VirtualEnvironmentError(

tjguk, 2021-03-17:
@RandoSY -- is that all that's in the file? It looks like what you've provided is just the information from the crash screen, which only gives us the very immediate context of the error. I'm looking for the contents of the file in `~/.cache/mu/log/mu.log`. Are you able to get hold of that?

RandoSY, 2021-03-17:
I attached the log file, "mu.log", and just was highlighting the error
message ... :)

tjguk, 2021-03-17:
Well that's annoying because Github appears to have discarded the attachment. (Unless I'm looking in the wrong place...). Are you able to see it from the GH web interface at https://github.com/mu-editor/mu/issues/1377?

RandoSY, 2021-03-17:
Let me try again ... 
[mu.log](https://github.com/mu-editor/mu/files/6158979/mu.log)

RandoSY, 2021-03-17:
Did u finally get access to this, wondering?

tjguk, 2021-03-17:
Thanks, @RandoSY  -- yes, just looking at it now. There's a pip dependency conflict, but presumably one that only occurs in your particular environment since we have buildbots building several Python versions on Ubuntu (as does the main developer @ntoll). 

What distro/architecture are you running, please?

RandoSY, 2021-03-17:
uname -a
Linux PCIeSSD 5.8.1-050801-generic #202008111432 SMP Tue Aug 11 14:34:42
UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

and

lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
*Description: Ubuntu 20.10*
Release: 20.10
Codename: groovy

tjguk, 2021-03-18:
Thanks -- I'll see if we can get a Ubuntu Groovy installation running with Python 3.8; we should be able to reproduce and then pin the right combination of versions. I'm a little surprised we haven't hit this already, though, given just how many times we've rebuilt these venvs in the course of testing!

carlosperate, 2021-03-18:
Pasting the error message here so that it helps with searches (from users with the same problem, or for us to find this again in the future):
```
ERROR: Cannot install reedsolo 1.5.4 (from /home/rasyoung/.local/lib/python3.8/site-packages/mu/wheels/reedsolo-1.5.4-cp38-cp38-linux_x86_64.whl) and reedsolo 1.5.4 (from /home/rasyoung/.local/lib/python3.8/site-packages/mu/wheels/reedsolo-1.5.4-py3-none-any.whl) because these package versions have conflicting dependencies.

The conflict is caused by:
    The user requested reedsolo 1.5.4 (from /home/rasyoung/.local/lib/python3.8/site-packages/mu/wheels/reedsolo-1.5.4-py3-none-any.whl)
    The user requested reedsolo 1.5.4 (from /home/rasyoung/.local/lib/python3.8/site-packages/mu/wheels/reedsolo-1.5.4-cp38-cp38-linux_x86_64.whl)

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/user_guide/#fixing-conflicting-dependencies

2021-03-17 12:29:53,183 - mu.virtual_environment:198(run) DEBUG: About to run blocking: /home/rasyoung/.local/share/mu/mu_venv-38-20210317-122951/bin/pip, ['list', '--disable-pip-version-check'], 30.0
2021-03-17 12:29:53,475 - mu.virtual_environment:606(install_jupyter_kernel) INFO: Installing Jupyter Kernel: "Python/Mu (mu_venv-38-20210317-122951)"
2021-03-17 12:29:53,502 - mu.virtual_environment:456(ensure_and_create) ERROR: Unable to create a working virtual environment.
2021-03-17 12:30:00,512 - root:174(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/home/rasyoung/.local/lib/python3.8/site-packages/mu/app.py", line 162, in run
    raise ex
  File "/home/rasyoung/.local/lib/python3.8/site-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/rasyoung/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 459, in ensure_and_create
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.
```

Looks like this version of reedsolo only has a wheel for Windows Python 3.7, so the two conflicting wheels must have been created by Mu? `reedsolo-1.5.4-cp38-cp38-linux_x86_64.whl` and `reedsolo-1.5.4-py3-none-any.whl`.

https://pypi.org/project/reedsolo/1.5.4/

tjguk, 2021-03-18:
Thanks @carlosperate -- yes; we download whatever pip gives us for that platform and then build wheels so we're only left with wheels to install. If I get a chance later I'll run up Ubuntu 20.10 on WSL and try to reproduce. If anyone wants to get there first, please feel free :)

carlosperate, 2021-03-18:
Right, I guess what i meant is that I'd expect Mu to create `reedsolo-1.5.4-cp38-cp38-linux_x86_64.whl`, but not `reedsolo-1.5.4-py3-none-any.whl` (without additional flags it woudn't create a `py3-none-any` wheel, no?), and I wouldn't know where that wheel came from since it's not hosted in PyPI.

tjguk, 2021-03-20:
Another one I can't reproduce. I'm running on Kubuntu 20.10; did a fresh install from PyPI; started the editor; hit the "git not installed" bug; installed git; everything went through smoothly. 

Checked the logs -- Mu clearly downloading the tgz for reedsolo 1.5.4 and, later, generating a ".any" wheel with no conflicts.

I'll go back to the @RandoSY logs to look for any clues there, and see if I can come at it a different way; but there's nothing obvious.

tjguk, 2021-03-20:
Just looking at the tarzip, it looks like it'll build with Cython if that's available. Otherwise it's a pure-python lib. I might try to install Cython on this box to see if that changes anything

tjguk, 2021-03-20:
Bingo. Installing Cython -> the same crash report which @RandoSY experienced. I'm a little surprised as we're only creating wheels from sdists after everything's downloaded. (So I don't think we're creating a pure wheel from one set of downloads and then a Cython-generated binary wheel from another)

But clearly something of the sort is happening: I'll try to understand the sequence of events here

tjguk, 2021-03-20:
Uploading mu own log file
[mu.log](https://github.com/mu-editor/mu/files/6175368/mu.log)

tjguk, 2021-03-20:
I've put forward a PR at #1386 -- assuming it builds & tests correctly I'll merge that some time today

PR #1386 (2021-03-20): Only try to rebuild the specific sdist, not its dependencies
Fixes #1377 

When we use `pip wheel` to convert an sdist to a wheel on the local machine, it defaults to pulling in the dependencies again. For reasons I haven't entirely understood, this results in an odd effect when one of the sdist projects includes one of the others as a dependency; this is compounded when the sdist has an option to build a binary when Cython is present, or a pure Python package when it isn't

To cut the Gordian Knot we can simply build a wheel from that sdist alone
closed 2021-03-20 by tjguk