mu-editor/mu #1380: "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-3jbwjd_l/PyQtChart/'

cdesqaz, 2021-03-17:
Hi everybody,
My system uname -all:
Linux w2l-pc.masua.com 4.15.0-137-generic #141-Ubuntu SMP Fri Feb 19 13:46:27 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

masua@w2l-pc:~/0ai0/SysAdmin$  python
Python 2.7.17 (default, Feb 27 2021, 15:10:58) 
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 

masua@w2l-pc:~/0ai0/SysAdmin$  python3
Python 3.6.9 (default, Jan 26 2021, 15:33:00) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
masua@w2l-pc:~/0ai0/SysAdmin$  pip -V
pip 21.0.1 from /usr/local/lib/python3.6/dist-packages/pip (python 3.6)

I was installing mu-editor first time ever. 

This It were the first failed terminal output:
pip3 install mu-editor ==>

Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-3jbwjd_l/PyQtChart/'

after that:

I just wondering about this issues, already read every answer, and at the end put attention on my terminal output en run the command with sudo. 

That's it. Installation run perfectly.

virtual environment created with python3.6: python3 -m venv mu
source ~/myvenv/mu/bin/activate
sudo pip3 install mu-editor

Successfully installed Pillow-8.1.2 PyQt5-sip-12.8.1 PyQtChart-5.14.0 appdirs-1.4.4 colorzero-1.1 cycler-0.10.0 gpiozero-1.5.1 guizero-1.1.1 ipykernel-5.5.0 jupyter-client-6.1.12 jupyter-core-4.7.1 kiwisolver-1.3.1 matplotlib-2.2.2 mu-editor-1.0.3 nudatus-0.0.5 numpy-1.19.5 pgzero-1.2 pigpio-1.78 pycodestyle-2.4.0 pyflakes-2.0.0 pygame-1.9.6 pyparsing-2.4.7 pyqt5-5.14.1 pyserial-3.4 pytz-2021.1 qscintilla-2.11.4 qtconsole-4.3.1 semver-2.13.0 tornado-6.1
(mu) masua@w2l-pc:/0l2p0/Python/ide/mu$ mu-editor
Registrando en /home/masua/.cache/mu/log/mu.log
QApplication: invalid style override 'gtk' passed, ignoring it.
Available styles: Windows, Fusion
Gtk-Message: 09:39:53.450: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: 09:40:11.517: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: 09:47:19.852: GtkDialog mapped without a transient parent. This is discouraged.
QObject::killTimer: Timers cannot be stopped from another thread
QObject::QObject: Timers cannot be stopped from another thread
QThread: Destroyed while thread is still running
Abortado (`core' generado)
(mu) masua@w2l-pc:/0l2p0/Python/ide/mu$ [IPKernelApp] WARNING | Parent appears to have exited, shutting down.
^C
(mu) masua@w2l-pc:/0l2p0/Python/ide/mu$

Current erminal output:
masua@w2l-pc:~/0ai0/SysAdmin$   pip install mu-editor
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: mu-editor in /usr/local/lib/python3.6/dist-packages (1.0.3)
Requirement already satisfied: requests>=2.19.1 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (2.25.0)
Requirement already satisfied: qscintilla==2.11.4 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (2.11.4)
Requirement already satisfied: guizero>=1.1.0 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (1.1.1)
Requirement already satisfied: semver>=2.8.0 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (2.13.0)
Requirement already satisfied: PyQtChart==5.14.0 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (5.14.0)
Requirement already satisfied: nudatus>=0.0.3 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (0.0.5)
Requirement already satisfied: gpiozero>=1.4.1 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (1.5.1)
Requirement already satisfied: pyserial==3.4 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (3.4)
Requirement already satisfied: matplotlib==2.2.2 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (2.2.2)
Requirement already satisfied: qtconsole==4.3.1 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (4.3.1)
Requirement already satisfied: appdirs>=1.4.3 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (1.4.4)
Requirement already satisfied: pycodestyle==2.4.0 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (2.4.0)
Requirement already satisfied: pyflakes==2.0.0 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (2.0.0)
Requirement already satisfied: pigpio>=1.40.post1 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (1.78)
Requirement already satisfied: pgzero==1.2 in /usr/local/lib/python3.6/dist-packages (from mu-editor) (1.2)
[... 116 lines cut ...]
Requirement already satisfied: python-dateutil>=2.1 in /usr/local/lib/python3.6/dist-packages (from matplotlib==2.2.2->mu-editor) (2.8.1)
Requirement already satisfied: six>=1.10 in /home/masua/.local/lib/python3.6/site-packages (from matplotlib==2.2.2->mu-editor) (1.15.0)
Requirement already satisfied: numpy>=1.7.1 in /usr/local/lib/python3.6/dist-packages (from matplotlib==2.2.2->mu-editor) (1.19.5)
Requirement already satisfied: pygame<2.0,>=1.9.2 in /usr/local/lib/python3.6/dist-packages (from pgzero==1.2->mu-editor) (1.9.6)
Requirement already satisfied: PyQt5-sip<13,>=12.7 in /usr/local/lib/python3.6/dist-packages (from pyqt5==5.14.1->mu-editor) (12.8.1)
Requirement already satisfied: ipykernel>=4.1 in /usr/local/lib/python3.6/dist-packages (from qtconsole==4.3.1->mu-editor) (5.5.0)
Requirement already satisfied: traitlets in /usr/lib/python3/dist-packages (from qtconsole==4.3.1->mu-editor) (4.3.2)
Requirement already satisfied: pygments in /usr/local/lib/python3.6/dist-packages (from qtconsole==4.3.1->mu-editor) (2.0.2)
Requirement already satisfied: jupyter-core in /usr/local/lib/python3.6/dist-packages (from qtconsole==4.3.1->mu-editor) (4.7.1)
Requirement already satisfied: jupyter-client>=4.1 in /usr/local/lib/python3.6/dist-packages (from qtconsole==4.3.1->mu-editor) (6.1.12)
Requirement already satisfied: ipython-genutils in /usr/lib/python3/dist-packages (from qtconsole==4.3.1->mu-editor) (0.2.0)
Requirement already satisfied: colorzero in /usr/local/lib/python3.6/dist-packages (from gpiozero>=1.4.1->mu-editor) (1.1)
Requirement already satisfied: tornado>=4.2 in /usr/local/lib/python3.6/dist-packages (from ipykernel>=4.1->qtconsole==4.3.1->mu-editor) (6.1)
Requirement already satisfied: ipython>=5.0.0 in /usr/lib/python3/dist-packages (from ipykernel>=4.1->qtconsole==4.3.1->mu-editor) (5.5.0)
Requirement already satisfied: pexpect in /usr/lib/python3/dist-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor) (4.2.1)
Requirement already satisfied: pyzmq>=13 in /usr/lib/python3/dist-packages (from jupyter-client>=4.1->qtconsole==4.3.1->mu-editor) (16.0.2)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python3.6/dist-packages (from requests>=2.19.1->mu-editor) (2020.12.5)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /usr/local/lib/python3.6/dist-packages (from requests>=2.19.1->mu-editor) (1.26.2)
Requirement already satisfied: idna<3,>=2.5 in /usr/local/lib/python3.6/dist-packages (from requests>=2.19.1->mu-editor) (2.10)
Requirement already satisfied: chardet<4,>=3.0.2 in /usr/lib/python3/dist-packages (from requests>=2.19.1->mu-editor) (3.0.4)
(ipy) masua@w2l-pc:~/0ai0/SysAdmin$ 

Resuming, I must have to use, 'sudo' in order to get install mu-editor.
I did the installation without and with virtual environment, same happens.
Thanks for reading, I apologize in advance, mistakes, omissions on my own bad

carlosperate, 2021-03-18:
Hi @cdesqaz,

I hope you don't mind I've edited your post to make it a bit more readable. Wrapping the terminal output with code formatting helps lot when reading it: ` ``` content ``` `

using `sudo pip`, even inside a virtualenv, is highly discouraged as it can end up using the system python pip instead of the virtual environment.

For example, in this output we see that `sudo pip` is confirming things are already installed in `/usr/local/lib/python3.6/dist-packages`, which is your system python installation, not the virtual environment.
```
(ipy) masua@w2l-pc:~/0ai0/SysAdmin$  sudo pip install mu-editor
[sudo] contraseña para masua: 
WARNING: The directory '/home/masua/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already satisfied: mu-editor in /usr/local/lib/python3.6/dist-packages (1.0.3)
```

When you use `pip` without sudo in the virtualenvironment it fails to install PyQtChart, which I must admit cannot explain from the top of head.

Based on the output, pip is collecting the source distribution and tries to build it:

```
Collecting PyQtChart==5.14.0 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/c8/6e/08e68bb5c3c64fbb56f1755c0b4748bb82775fa1b7c9fbaa61572eeaa76a/PyQtChart-5.14.0.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.6/tokenize.py", line 452, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-build-6igjjfqn/PyQtChart/setup.py'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-6igjjfqn/PyQtChart/
```

It should instead pick up the `PyQtChart-5.14.0-5.14.2-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl` wheel from: https://pypi.org/project/PyQtChart/5.14.0/#files

What linux distribution, version and platform (32/64 bits, Arm, other?) are you using?

cdesqaz, 2021-03-20:
Hi! It's fine, thanks for your answer. 
I am an Ubuntu user long time ago, in this machine, it is a old one:
masua@w2l-pc:~/0ai0/SysAdmin$  cat arqCPU
modo(s) de operación de las CPUs:    32-bit, 64-bit
CPU(s):                              2
Lista de la(s) CPU(s) en línea:      0,1
Familia de CPU:                      6
Nombre del modelo:                   Intel(R) Celeron(R) CPU  N2805  @ 1.46GHz
CPU MHz:                             968.582
CPU MHz máx.:                        1466,3000
CPU MHz mín.:                        533,2000
CPU(s) del nodo NUMA 0:              0,1

masua@w2l-pc:~/0ai0/SysAdmin$  cat hostnamectl_output.txt
Static hostname: w2l-pc.masua.com
Icon name: computer-handset
Chassis: handset
Operating System: Ubuntu 18.04.5 LTS ---> Lubuntu flavor l
Kernel: Linux 4.15.0-136-generic
Architecture: x86-64

masua@w2l-pc:~/0ai0/SysAdmin$  apt show lubuntu-core -a
Package: lubuntu-core
Version: 0.94.1
Priority: optional
Section: universe/metapackages
Source: lubuntu-meta
Origin: Ubuntu
Maintainer: Lubuntu Developers <lubuntu-devel@lists.ubuntu.com>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 16,4 kB
Depends: apport-gtk, dbus-x11, lightdm, lightdm-gtk-greeter, lubuntu-artwork, lubuntu-default-settings, lubuntu-icon-theme, lxpanel, lxsession, lxterminal, openbox, pcmanfm, plymouth-theme-lubuntu-logo, plymouth-theme-lubuntu-text, python3-launchpadlib, ubuntu-mono, xserver-xorg-input-all, xserver-xorg-input-synaptics, xserver-xorg-video-all, xserver-xorg-video-intel, xserver-xorg-video-qxl
Recommends: ubuntu-report
Task: lubuntu-core
Supported: 3y
Download-Size: 3.004 B
APT-Manual-Installed: yes
APT-Sources: http://la-mirrors.evowise.com/ubuntu bionic-updates/universe amd64 Packages
Description: Lubuntu Desktop environment - minimal installation
 This metapackage package depends on the minimal components for an
 installation of Lubuntu.
 .
 También se usa para ayudar a asegurar actualizaciones correctas, así que
 se recomienda no eliminarlo.

Package: lubuntu-core
Version: 0.94
Priority: optional
Section: universe/metapackages
Source: lubuntu-meta
Origin: Ubuntu
Maintainer: Lubuntu Developers <lubuntu-devel@lists.ubuntu.com>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 16,4 kB
Depends: apport-gtk, dbus-x11, lightdm, lightdm-gtk-greeter, lubuntu-artwork, lubuntu-default-settings, lubuntu-icon-theme, lxpanel, lxsession, lxterminal, openbox, pcmanfm, plymouth-theme-lubuntu-logo, plymouth-theme-lubuntu-text, ubuntu-mono, xserver-xorg-input-all, xserver-xorg-input-synaptics, xserver-xorg-video-all, xserver-xorg-video-intel, xserver-xorg-video-qxl
Recommends: ubuntu-report
Task: lubuntu-core
Supported: 3y
Download-Size: 2.932 B
APT-Sources: http://la-mirrors.evowise.com/ubuntu bionic/universe amd64 Packages
Description: Lubuntu Desktop environment - minimal installation
 This metapackage package depends on the minimal components for an
 installation of Lubuntu.
 .
 También se usa para ayudar a asegurar actualizaciones correctas, así que
 se recomienda no eliminarlo.

Thank for reading.

carlosperate, 2021-03-24:
Okay, so the PyQtCharts linux wheel now uses `manylinux2014`, and something in your system thinks it's not compatible.

Could you check what glibc version do you have? Hopefully one of the options in this article should work: https://dev.to/0xbf/how-to-get-glibc-version-c-lang-26he

Can you also try this in this order:
- Check what pip version you have inside the virtualenv
- Update the virtualenv pip to the latest version `pip install --upgrade pip`
- Try to install again `pip install PyQtChart==5.14.0`

cdesqaz, 2021-03-24:
> Okay, so the PyQtCharts linux wheel now uses `manylinux2014`, and something in your system thinks it's not compatible.
> 
> Could you check what glibc version do you have? Hopefully one of the options in this article should work: https://dev.to/0xbf/how-to-get-glibc-version-c-lang-26he
> 
> Can you also try this in this order:
> 
> * Check what pip version you have inside the virtual environment `pip --version`
> * Update pip inside the virtual environment to the latest version `pip install --upgrade pip`
> * Try to install again `pip install PyQtChart==5.14.0`

hi!, let's go
> Could you check what glibc version do you have? Hopefully one of the options in this article should work: https://dev.to/0xbf/how-to-get-glibc-version-c-lang-26he

masua@w2l-pc:~$  ldd --version
ldd (Ubuntu GLIBC 2.27-3ubuntu1.4) 2.27
Copyright (C) 2018 Free Software Foundation, Inc.
Esto es software libre; vea el código fuente para las condiciones de copia.
No hay NINGUNA garantía; ni siquiera de COMERCIABILIDAD o IDONEIDAD PARA UN
FIN DETERMINADO.
Escrito por Roland McGrath y Ulrich Drepper.
masua@w2l-pc:~$  ldd `which ls` | grep libc
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fdf82ea1000)
masua@w2l-pc:~$  /lib/x86_64-linux-gnu/libc.so.6
GNU C Library (Ubuntu GLIBC 2.27-3ubuntu1.4) stable release version 2.27.
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
Compiled by GNU CC version 7.5.0.
libc ABIs: UNIQUE IFUNC
For bug reporting instructions, please see:
<https://bugs.launchpad.net/ubuntu/+source/glibc/+bugs>.
masua@w2l-pc:~$ 

masua@w2l-pc:~/0ai0/SysAdmin$  nano main.c
masua@w2l-pc:~/0ai0/SysAdmin$  gcc main.c -o main
masua@w2l-pc:~/0ai0/SysAdmin$  ./main
2.27
2.27
glibc 2.27
masua@w2l-pc:~/0ai0/SysAdmin$ 

okey,
> * Check what pip version you have inside the virtual environment `pip --version`

masua@w2l-pc:~$   source ~/myvenv/mu/bin/activate
(mu) masua@w2l-pc:~$  pip --version
pip 9.0.1 from /home/masua/myvenv/mu/lib/python3.6/site-packages (python 3.6)
(mu) masua@w2l-pc:~$ 

> * Update pip inside the virtual environment to the latest version `pip install --upgrade pip`

(mu) masua@w2l-pc:~$  pip install --upgrade pip
Cache entry deserialization failed, entry ignored
Collecting pip
  Downloading https://files.pythonhosted.org/packages/fe/ef/60d7ba03b5c442309ef42e7d69959f73aacccd0d86008362a681c4698e83/pip-21.0.1-py3-none-any.whl (1.5MB)
    100% |████████████████████████████████| 1.5MB 169kB/s 
Installing collected packages: pip
  Found existing installation: pip 9.0.1
    Uninstalling pip-9.0.1:
      Successfully uninstalled pip-9.0.1
Successfully installed pip-21.0.1
(mu) masua@w2l-pc:~$ 

> * Try to install again `pip install PyQtChart==5.14.0`

(mu) masua@w2l-pc:~$  pip install PyQtChart==5.14.0
Collecting PyQtChart==5.14.0
  Downloading PyQtChart-5.14.0-5.14.2-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.2 MB)
     |████████████████████████████████| 1.2 MB 246 kB/s 
[... 117 lines cut ...]
     |████████████████████████████████| 153 kB 351 kB/s 
Collecting certifi>=2017.4.17
  Using cached certifi-2020.12.5-py2.py3-none-any.whl (147 kB)
Collecting chardet<5,>=3.0.2
  Using cached chardet-4.0.0-py2.py3-none-any.whl (178 kB)
Collecting idna<3,>=2.5
  Using cached idna-2.10-py2.py3-none-any.whl (58 kB)
Collecting ptyprocess>=0.5
  Using cached ptyprocess-0.7.0-py2.py3-none-any.whl (13 kB)
Installing collected packages: six, ipython-genutils, decorator, wcwidth, traitlets, ptyprocess, parso, tornado, pyzmq, python-dateutil, pygments, prompt-toolkit, pickleshare, pexpect, jupyter-core, jedi, backcall, jupyter-client, ipython, urllib3, pytz, pyqt5, pyparsing, pygame, numpy, kiwisolver, ipykernel, idna, cycler, colorzero, chardet, certifi, semver, requests, qtconsole, qscintilla, pyserial, pyflakes, pycodestyle, Pillow, pigpio, pgzero, nudatus, matplotlib, guizero, gpiozero, appdirs, mu-editor
  Attempting uninstall: pyqt5
    Found existing installation: PyQt5 5.15.4
    Uninstalling PyQt5-5.15.4:
      Successfully uninstalled PyQt5-5.15.4
Successfully installed Pillow-8.1.2 appdirs-1.4.4 backcall-0.2.0 certifi-2020.12.5 chardet-4.0.0 colorzero-2.0 cycler-0.10.0 decorator-4.4.2 gpiozero-1.6.2 guizero-1.1.1 idna-2.10 ipykernel-5.5.0 ipython-7.16.1 ipython-genutils-0.2.0 jedi-0.18.0 jupyter-client-6.1.12 jupyter-core-4.7.1 kiwisolver-1.3.1 matplotlib-2.2.2 mu-editor-1.0.3 nudatus-0.0.5 numpy-1.19.5 parso-0.8.1 pexpect-4.8.0 pgzero-1.2 pickleshare-0.7.5 pigpio-1.78 prompt-toolkit-3.0.18 ptyprocess-0.7.0 pycodestyle-2.4.0 pyflakes-2.0.0 pygame-1.9.6 pygments-2.8.1 pyparsing-2.4.7 pyqt5-5.14.1 pyserial-3.4 python-dateutil-2.8.1 pytz-2021.1 pyzmq-22.0.3 qscintilla-2.11.4 qtconsole-4.3.1 requests-2.25.1 semver-2.13.0 six-1.15.0 tornado-6.1 traitlets-4.3.3 urllib3-1.26.4 wcwidth-0.2.5
(mu) masua@w2l-pc:~$  mu-editor
Registrando en /home/masua/.cache/mu/log/mu.log
QApplication: invalid style override 'gtk' passed, ignoring it.
	Available styles: Windows, Fusion

[image]

It 's  done. After pip install --upgrade pip, everything work very well. 

Thanks  a lot!

carlosperate, 2021-03-24:
Glad it worked!

closed 2021-03-24 by carlosperate