mu-editor/mu #1381: Crash Report

bhibb, 2021-03-17:
Version: 1.1.0.beta.2
Locale: en_US
Platform: Linux 5.8.0-45-generic #51~20.04.1-Ubuntu SMP Tue Feb 23 13:46:31 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/bhibb/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 441, in ensure_and_create
    self.ensure()
  File "/home/bhibb/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "/home/bhibb/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: /home/bhibb/.local/share/mu/mu_venv-38-20210317-145511 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/bhibb/.local/lib/python3.8/site-packages/mu/app.py", line 162, in run
    raise ex
  File "/home/bhibb/.local/lib/python3.8/site-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/bhibb/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "/home/bhibb/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 596, in create
    self.install_baseline_packages()
  File "/home/bhibb/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 649, in install_baseline_packages
    wheels.download()
  File "/home/bhibb/.local/lib/python3.8/site-packages/mu/wheels/__init__.py", line 82, in download
    process = subprocess.run(
  File "/usr/lib/python3.8/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'download', '--destination-directory', '/home/bhibb/.local/lib/python3.8/site-packages/mu/wheels', 'git+https://github.com/mu-editor/pgzero.git@mu-wheel']' returned non-zero exit status 1.

```

carlosperate, 2021-03-18:
Thanks for the report!

Can you run the following commands in the terminal and copy their output here? That will help us debug this.

```
git --version
```

```
/usr/bin/python3 -m pip --version
```

carlosperate, 2021-03-29:
@bhibb there is a new release that we think has fixed this issue. Would you be able to install the latest version and try again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.3

carlosperate, 2021-04-06:
As we believe this issue has been solved in the latest beta release I'll close this. But please do feel free to reopen if it still doesn't work. Thanks again for the report!

closed 2021-04-06 by carlosperate