mu-editor/mu #1383: Big Sur micro:bit with 1.1.0-beta.2

fhassancq, 2021-03-18:
Installed on a new laptop, still cannot flash to a BBC:microbit. Log below:

2021-03-18 08:09:49,446 - root:259(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.2
2021-03-18 08:09:49,449 - root:260(run) INFO: uname_result(system='Darwin', node='faisalhassans-MacBook-Pro.local', release='20.3.0', version='Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-03-18 08:09:49,449 - root:261(run) INFO: Python path: ['/', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python38.zip', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/lib-dynload', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/setuptools-49.6.0-py3.8.egg', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/pip-20.2.2-py3.8.egg', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/IPython/extensions']
2021-03-18 08:09:49,449 - root:262(run) INFO: Language code: en_GB
2021-03-18 08:09:49,879 - mu.virtual_environment:434(ensure_and_create) INFO: Added log handler.
2021-03-18 08:09:50,100 - mu.virtual_environment:438(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-03-18 08:09:50,121 - mu.virtual_environment:489(ensure_path) INFO: Virtual Environment found at: /Users/faisal.hassan/Library/Application Support/mu/mu_venv-38-20210317-093324
2021-03-18 08:09:50,124 - mu.virtual_environment:502(ensure_interpreter) INFO: Interpreter found at: /Users/faisal.hassan/Library/Application Support/mu/mu_venv-38-20210317-093324/bin/python
2021-03-18 08:09:50,169 - mu.virtual_environment:534(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-03-18 08:09:50,172 - mu.virtual_environment:566(ensure_pip) INFO: Pip found at: /Users/faisal.hassan/Library/Application Support/mu/mu_venv-38-20210317-093324/bin/pip
2021-03-18 08:09:50,175 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: pygame
2021-03-18 08:09:50,941 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: numpy
2021-03-18 08:09:51,780 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-03-18 08:09:51,820 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: flask
2021-03-18 08:09:52,129 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: serial
2021-03-18 08:09:52,173 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: qtconsole
2021-03-18 08:09:52,215 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: nudatus
2021-03-18 08:09:52,270 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: esptool
2021-03-18 08:09:52,346 - mu.virtual_environment:451(ensure_and_create) INFO: Virtual environment already exists.
2021-03-18 08:09:53,485 - mu.logic:758(__init__) INFO: Setting up editor.
2021-03-18 08:09:53,485 - mu.logic:778(__init__) INFO: Log directory: /Users/faisal.hassan/Library/Logs/mu
2021-03-18 08:09:53,486 - mu.logic:779(__init__) INFO: Data directory: /Users/faisal.hassan/Library/Application Support/mu
2021-03-18 08:09:53,486 - mu.logic:793(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2021-03-18 08:09:53,486 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/faisal.hassan/mu_code from settings file
2021-03-18 08:09:53,582 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/faisal.hassan/mu_code from settings file
2021-03-18 08:09:53,583 - mu.logic:1523(change_mode) INFO: Workspace directory: /Users/faisal.hassan/mu_code
2021-03-18 08:09:53,583 - mu.logic:867(restore_session) DEBUG: <SessionSettings from /Users/faisal.hassan/Library/Application Support/mu/session.json>
2021-03-18 08:09:53,588 - mu.logic:978(_load) INFO: Loading script from: /Users/faisal.hassan/OneDrive - Yew Chung International School HK/YCIS/KS3/Year 8/4 - BBC microbits - Done/Lesson 3 - Iteration and Selection/homeworkcode2.py
2021-03-18 08:09:53,589 - mu.logic:316(read_and_decode) DEBUG: Trying to decode with utf-8
2021-03-18 08:09:53,589 - mu.logic:319(read_and_decode) INFO: Decoded with utf-8
2021-03-18 08:09:53,590 - mu.logic:332(read_and_decode) DEBUG: Detected newline '\n'
2021-03-18 08:09:53,590 - mu.logic:1077(_load) DEBUG: from microbit import *

x=0
y=0

while True:
    for i in range (1,9):
        display.set_pixel(x, y, i)
        y+=1
        sleep(250)
        
        if y==5:
            y=0
            x+=1
        if x==5:
            x=0
            display.clear()
2021-03-18 08:09:53,707 - mu.logic:889(restore_session) INFO: Loaded files.
2021-03-18 08:09:53,707 - mu.logic:892(restore_session) INFO: User defined environment variables: []
2021-03-18 08:09:53,707 - mu.logic:897(restore_session) INFO: Minify scripts on micro:bit? False
2021-03-18 08:09:53,730 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/faisal.hassan/mu_code from settings file
2021-03-18 08:09:53,730 - mu.logic:1523(change_mode) INFO: Workspace directory: /Users/faisal.hassan/mu_code
2021-03-18 08:10:19,111 - mu.logic:1396(show_admin) INFO: Showing admin with logs from /Users/faisal.hassan/Library/Logs/mu/mu.log
2021-03-18 08:10:19,111 - mu.virtual_environment:697(installed_packages) INFO: Discovering installed third party modules in venv.
2021-03-18 08:10:19,112 - mu.virtual_environment:198(run) DEBUG: About to run blocking: /Users/faisal.hassan/Library/Application Support/mu/mu_venv-38-20210317-093324/bin/pip, ['list', '--disable-pip-version-check'], 30.0
2021-03-18 08:10:19,934 - mu.virtual_environment:717(installed_packages) INFO: []

carlosperate, 2021-03-18:
Hi @fhassancq,

Thanks for the report. The log you pasted doesn't show an attempt to flash the micro:bit, could you open Mu again, click flash in the micro:bit mode (with a micro:bit plugged into the computer) and get the log again? That should contain more info that will help us identify the issue.

What happens when the flash button is pressed? does it show you an error message?

fhassancq, 2021-03-19:
2021-03-19 12:28:08,232 - root:259(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.2
2021-03-19 12:28:08,238 - root:260(run) INFO: uname_result(system='Darwin', node='faisalhassans-MacBook-Pro.local', release='20.3.0', version='Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-03-19 12:28:08,238 - root:261(run) INFO: Python path: ['/', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python38.zip', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/lib-dynload', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/setuptools-49.6.0-py3.8.egg', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/pip-20.2.2-py3.8.egg', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/IPython/extensions']
2021-03-19 12:28:08,238 - root:262(run) INFO: Language code: en_GB
2021-03-19 12:28:08,666 - mu.virtual_environment:434(ensure_and_create) INFO: Added log handler.
2021-03-19 12:28:08,893 - mu.virtual_environment:438(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-03-19 12:28:08,922 - mu.virtual_environment:489(ensure_path) INFO: Virtual Environment found at: /Users/faisal.hassan/Library/Application Support/mu/mu_venv-38-20210317-093324
2021-03-19 12:28:08,931 - mu.virtual_environment:502(ensure_interpreter) INFO: Interpreter found at: /Users/faisal.hassan/Library/Application Support/mu/mu_venv-38-20210317-093324/bin/python
2021-03-19 12:28:08,998 - mu.virtual_environment:534(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-03-19 12:28:09,002 - mu.virtual_environment:566(ensure_pip) INFO: Pip found at: /Users/faisal.hassan/Library/Application Support/mu/mu_venv-38-20210317-093324/bin/pip
2021-03-19 12:28:09,010 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: pygame
2021-03-19 12:28:09,749 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: numpy
2021-03-19 12:28:10,645 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-03-19 12:28:10,687 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: flask
2021-03-19 12:28:11,047 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: serial
2021-03-19 12:28:11,118 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: qtconsole
2021-03-19 12:28:11,173 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: nudatus
2021-03-19 12:28:11,228 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: esptool
2021-03-19 12:28:11,310 - mu.virtual_environment:451(ensure_and_create) INFO: Virtual environment already exists.
2021-03-19 12:28:12,496 - mu.logic:758(__init__) INFO: Setting up editor.
2021-03-19 12:28:12,506 - mu.logic:778(__init__) INFO: Log directory: /Users/faisal.hassan/Library/Logs/mu
2021-03-19 12:28:12,506 - mu.logic:779(__init__) INFO: Data directory: /Users/faisal.hassan/Library/Application Support/mu
2021-03-19 12:28:12,508 - mu.logic:793(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2021-03-19 12:28:12,510 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/faisal.hassan/mu_code from settings file
2021-03-19 12:28:12,717 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/faisal.hassan/mu_code from settings file
2021-03-19 12:28:12,718 - mu.logic:1523(change_mode) INFO: Workspace directory: /Users/faisal.hassan/mu_code
2021-03-19 12:28:12,718 - mu.logic:867(restore_session) DEBUG: <SessionSettings from /Users/faisal.hassan/Library/Application Support/mu/session.json>
2021-03-19 12:28:12,734 - mu.logic:978(_load) INFO: Loading script from: /Users/faisal.hassan/OneDrive - Yew Chung International School HK/YCIS/KS3/Year 8/4 - BBC microbits - Done/Lesson 3 - Iteration and Selection/homeworkcode2.py
2021-03-19 12:28:12,736 - mu.logic:316(read_and_decode) DEBUG: Trying to decode with utf-8
2021-03-19 12:28:12,736 - mu.logic:319(read_and_decode) INFO: Decoded with utf-8
2021-03-19 12:28:12,736 - mu.logic:332(read_and_decode) DEBUG: Detected newline '\n'
2021-03-19 12:28:12,736 - mu.logic:1077(_load) DEBUG: from microbit import *

x=0
y=0

while True:
    for i in range (1,9):
        display.set_pixel(x, y, i)
        y+=1
        sleep(250)
        
        if y==5:
            y=0
            x+=1
        if x==5:
            x=0
            display.clear()
2021-03-19 12:28:12,931 - mu.logic:889(restore_session) INFO: Loaded files.
2021-03-19 12:28:12,931 - mu.logic:892(restore_session) INFO: User defined environment variables: []
2021-03-19 12:28:12,932 - mu.logic:897(restore_session) INFO: Minify scripts on micro:bit? False
2021-03-19 12:28:12,960 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/faisal.hassan/mu_code from settings file
2021-03-19 12:28:12,960 - mu.logic:1523(change_mode) INFO: Workspace directory: /Users/faisal.hassan/mu_code
2021-03-19 12:28:31,531 - mu.logic:737(check_usb) INFO: microbit device connected on port: /dev/cu.usbmodem141402(VID: 0x0D28, PID: 0x0204, manufacturer: 'ARM')
[... cut ...]
2021-03-19 12:28:57,651 - mu.logic:1396(show_admin) INFO: Showing admin with logs from /Users/faisal.hassan/Library/Logs/mu/mu.log
2021-03-19 12:28:57,651 - mu.virtual_environment:697(installed_packages) INFO: Discovering installed third party modules in venv.
2021-03-19 12:28:57,651 - mu.virtual_environment:198(run) DEBUG: About to run blocking: /Users/faisal.hassan/Library/Application Support/mu/mu_venv-38-20210317-093324/bin/pip, ['list', '--disable-pip-version-check'], 30.0
2021-03-19 12:28:58,738 - mu.virtual_environment:717(installed_packages) INFO: []

carlosperate, 2021-03-19:
Thank you very much for the log, that helps a lot.

Oddly enough it looks like your Mu app is still not able to access the micro:bit drive. Could you try to manually give it access via the "Security and Privacy" settings?
[image]

I think you might have try this as well with older Mu versions, it likely didn't work back then because of how the app was signed. With the beta.2 release I'm hoping it should work this time. If the Mu app is already present in that list, can you try removing it, rebooting the computer, and then adding it again?
Thanks a lot for helping us debug this issue!

fhassancq, 2021-03-22:
The mu editor app doesn't appear in the list. I have tried uninstalling the app, rebooting the laptop and installing it again but have the same issue.

carlosperate, 2021-03-22:
Thanks for checking @fhassancq, after installing the beta.2 can you try adding Mu into that list in the preferences? That should give the mu application access to the micro:bit drive.

fhassancq, 2021-03-23:
I have tried that but still have the same error message when flashing to the BBC micro:bit. 

Starting Mu 1.1.0.beta.2
2021-03-23 08:43:34,207 - root:260(run) INFO: uname_result(system='Darwin', node='faisalhassans-MacBook-Pro.local', release='20.3.0', version='Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-03-23 08:43:34,207 - root:261(run) INFO: Python path: ['/', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python38.zip', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/lib-dynload', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/setuptools-49.6.0-py3.8.egg', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/pip-20.2.2-py3.8.egg', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/IPython/extensions']
2021-03-23 08:43:34,207 - root:262(run) INFO: Language code: en_GB
2021-03-23 08:43:34,631 - mu.virtual_environment:434(ensure_and_create) INFO: Added log handler.
2021-03-23 08:43:34,849 - mu.virtual_environment:438(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-03-23 08:43:34,880 - mu.virtual_environment:489(ensure_path) INFO: Virtual Environment found at: /Users/faisal.hassan/Library/Application Support/mu/mu_venv-38-20210317-093324
2021-03-23 08:43:34,886 - mu.virtual_environment:502(ensure_interpreter) INFO: Interpreter found at: /Users/faisal.hassan/Library/Application Support/mu/mu_venv-38-20210317-093324/bin/python
2021-03-23 08:43:34,951 - mu.virtual_environment:534(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-03-23 08:43:34,956 - mu.virtual_environment:566(ensure_pip) INFO: Pip found at: /Users/faisal.hassan/Library/Application Support/mu/mu_venv-38-20210317-093324/bin/pip
2021-03-23 08:43:34,960 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: pygame
2021-03-23 08:43:35,769 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: numpy
2021-03-23 08:43:36,603 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-03-23 08:43:36,640 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: flask
2021-03-23 08:43:37,082 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: serial
2021-03-23 08:43:37,138 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: qtconsole
2021-03-23 08:43:37,176 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: nudatus
2021-03-23 08:43:37,231 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: esptool
2021-03-23 08:43:37,301 - mu.virtual_environment:451(ensure_and_create) INFO: Virtual environment already exists.
2021-03-23 08:43:38,479 - mu.logic:758(__init__) INFO: Setting up editor.
2021-03-23 08:43:38,481 - mu.logic:778(__init__) INFO: Log directory: /Users/faisal.hassan/Library/Logs/mu
2021-03-23 08:43:38,481 - mu.logic:779(__init__) INFO: Data directory: /Users/faisal.hassan/Library/Application Support/mu
2021-03-23 08:43:38,482 - mu.logic:793(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2021-03-23 08:43:38,483 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/faisal.hassan/mu_code from settings file
2021-03-23 08:43:38,647 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/faisal.hassan/mu_code from settings file
2021-03-23 08:43:38,647 - mu.logic:1523(change_mode) INFO: Workspace directory: /Users/faisal.hassan/mu_code
2021-03-23 08:43:38,647 - mu.logic:867(restore_session) DEBUG: <SessionSettings from /Users/faisal.hassan/Library/Application Support/mu/session.json>
2021-03-23 08:43:38,658 - mu.logic:978(_load) INFO: Loading script from: /Users/faisal.hassan/OneDrive - Yew Chung International School HK/YCIS/KS3/Year 8/4 - BBC microbits - Done/Lesson 3 - Iteration and Selection/homeworkcode2.py
2021-03-23 08:43:38,660 - mu.logic:316(read_and_decode) DEBUG: Trying to decode with utf-8
2021-03-23 08:43:38,660 - mu.logic:319(read_and_decode) INFO: Decoded with utf-8
2021-03-23 08:43:38,660 - mu.logic:332(read_and_decode) DEBUG: Detected newline '\n'
2021-03-23 08:43:38,660 - mu.logic:1077(_load) DEBUG: from microbit import *

x=0
y=0

while True:
    for i in range (1,9):
        display.set_pixel(x, y, i)
        y+=1
        sleep(250)
        
        if y==5:
            y=0
            x+=1
        if x==5:
            x=0
            display.clear()
2021-03-23 08:43:38,790 - mu.logic:889(restore_session) INFO: Loaded files.
2021-03-23 08:43:38,790 - mu.logic:892(restore_session) INFO: User defined environment variables: []
2021-03-23 08:43:38,790 - mu.logic:897(restore_session) INFO: Minify scripts on micro:bit? False
2021-03-23 08:43:38,816 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/faisal.hassan/mu_code from settings file
2021-03-23 08:43:38,816 - mu.logic:1523(change_mode) INFO: Workspace directory: /Users/faisal.hassan/mu_code
2021-03-23 08:43:44,667 - mu.logic:737(check_usb) INFO: microbit device connected on port: /dev/cu.usbmodem141402(VID: 0x0D28, PID: 0x0204, manufacturer: 'ARM')
[... cut ...]
2021-03-23 08:45:52,584 - mu.interface.main:971(show_confirmation) DEBUG: None
2021-03-23 08:45:56,309 - mu.settings:169(save) DEBUG: Saving to /Users/faisal.hassan/Library/Application Support/mu/session.json
2021-03-23 08:45:56,311 - mu.logic:1387(quit) INFO: Quitting.

2021-03-23 08:45:56,311 - mu.settings:169(save) DEBUG: Saving to /Users/faisal.hassan/Library/Application Support/mu/session.json
2021-03-23 08:45:56,312 - mu.settings:169(save) DEBUG: Saving to /Users/faisal.hassan/Library/Application Support/mu/venv.json
2021-03-23 11:05:51,068 - root:259(run) INFO:

sunruoshi, 2021-03-23:
My version is Mu 1.1.0 beta.2 still have this problem. Cannot flash. OS version 11.2.3(20D91) Big Sur.

2021-03-23 09:27:52,476 - root:259(run) INFO:

Starting Mu 1.1.0.beta.2
2021-03-23 09:27:52,476 - root:260(run) INFO: uname_result(system='Darwin', node='Sams-MacBook-Pro.local', release='20.3.0', version='Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-03-23 09:27:52,476 - root:261(run) INFO: Python path: ['/', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python38.zip', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/lib-dynload', '/Users/sam/.local/lib/python3.8/site-packages', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/setuptools-49.6.0-py3.8.egg', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/pip-20.2.2-py3.8.egg', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/IPython/extensions']
2021-03-23 09:27:52,476 - root:262(run) INFO: Language code: en_US
2021-03-23 09:27:52,477 - mu.settings:220(load) WARNING: No settings file found at /Users/sam/Library/Application Support/mu/settings.json; skipping
2021-03-23 09:27:52,728 - mu.virtual_environment:434(ensure_and_create) INFO: Added log handler.
2021-03-23 09:27:52,728 - mu.virtual_environment:438(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-03-23 09:27:52,874 - mu.virtual_environment:489(ensure_path) INFO: Virtual Environment found at: /Users/sam/Library/Application Support/mu/mu_venv-38-20210322-154202
2021-03-23 09:27:52,880 - mu.virtual_environment:502(ensure_interpreter) INFO: Interpreter found at: /Users/sam/Library/Application Support/mu/mu_venv-38-20210322-154202/bin/python
2021-03-23 09:27:52,920 - mu.virtual_environment:534(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-03-23 09:27:52,925 - mu.virtual_environment:566(ensure_pip) INFO: Pip found at: /Users/sam/Library/Application Support/mu/mu_venv-38-20210322-154202/bin/pip
2021-03-23 09:27:52,925 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: pygame
2021-03-23 09:27:53,042 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: numpy
2021-03-23 09:27:53,272 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-03-23 09:27:53,309 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: flask
2021-03-23 09:27:53,584 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: serial
2021-03-23 09:27:53,624 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: qtconsole
2021-03-23 09:27:53,663 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: nudatus
2021-03-23 09:27:53,720 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: esptool
2021-03-23 09:27:53,791 - mu.virtual_environment:451(ensure_and_create) INFO: Virtual environment already exists.
2021-03-23 09:27:54,911 - mu.logic:758(init) INFO: Setting up editor.
2021-03-23 09:27:54,911 - mu.logic:778(init) INFO: Log directory: /Users/sam/Library/Logs/mu
2021-03-23 09:27:54,911 - mu.logic:779(init) INFO: Data directory: /Users/sam/Library/Application Support/mu
2021-03-23 09:27:54,912 - mu.logic:793(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2021-03-23 09:27:54,912 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/sam/mu_code from settings file
2021-03-23 09:27:54,956 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/sam/mu_code from settings file
2021-03-23 09:27:54,956 - mu.logic:1523(change_mode) INFO: Workspace directory: /Users/sam/mu_code
2021-03-23 09:27:54,956 - mu.logic:867(restore_session) DEBUG: <SessionSettings from /Users/sam/Library/Application Support/mu/session.json>
2021-03-23 09:27:54,961 - mu.logic:978(_load) INFO: Loading script from: /Users/sam/mu_code/compass.py
2021-03-23 09:27:54,962 - mu.logic:316(read_and_decode) DEBUG: Trying to decode with utf-8
2021-03-23 09:27:54,962 - mu.logic:319(read_and_decode) INFO: Decoded with utf-8
2021-03-23 09:27:54,962 - mu.logic:332(read_and_decode) DEBUG: Detected newline '\n'
2021-03-23 09:27:54,962 - mu.logic:1077(_load) DEBUG: # Write your code here :-)
from microbit import *

compass.calibrate()

while True:
bearing = compass.heading()
if bearing < 45 or bearing > 315:
display.show("N")
elif bearing > 45 and bearing < 135:
display.show("E")
elif bearing > 135 and bearing < 225:
display.show("S")
elif bearing > 225 and bearing < 315:
display.show("W")
else:
display.show(" ")

2021-03-23 09:27:55,069 - mu.logic:889(restore_session) INFO: Loaded files.
2021-03-23 09:27:55,070 - mu.logic:892(restore_session) INFO: User defined environment variables: []
2021-03-23 09:27:55,070 - mu.logic:897(restore_session) INFO: Minify scripts on micro:bit? False
2021-03-23 09:27:55,091 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/sam/mu_code from settings file
2021-03-23 09:27:55,091 - mu.logic:1523(change_mode) INFO: Workspace directory: /Users/sam/mu_code
[... cut ...]
2021-03-23 09:28:29,697 - mu.logic:1396(show_admin) INFO: Showing admin with logs from /Users/sam/Library/Logs/mu/mu.log
2021-03-23 09:28:29,697 - mu.virtual_environment:697(installed_packages) INFO: Discovering installed third party modules in venv.
2021-03-23 09:28:29,697 - mu.virtual_environment:198(run) DEBUG: About to run blocking: /Users/sam/Library/Application Support/mu/mu_venv-38-20210322-154202/bin/pip, ['list', '--disable-pip-version-check'], 30.0
2021-03-23 09:28:30,115 - mu.virtual_environment:717(installed_packages) INFO: []

npat860, 2021-03-24:
I have the same issue, super annoying :/
When I try to flash the error message "There was a problem flashing the micro:bit" comes up, log is like this:

2021-03-24 14:50:02,869 - mu.modes.microbit:273(flash) INFO: Preparing to flash script.
2021-03-24 14:50:02,870 - mu.modes.microbit:280(flash) DEBUG: Python script from 'exercise_0.py' tab:
2021-03-24 14:50:02,870 - mu.modes.microbit:281(flash) DEBUG: b'"""Exercise 0 - Hello, World!"""\n\n# An import statement - we are adding logic from the display module of the microbit library to this code\nfrom microbit import display\n\n# Our function that is called\ndef main():\n    # See https://microbit-micropython.readthedocs.io/en/v1.0.1/display.html\n    display.scroll("Nicholas was here")\n\n# This is a standard Python construction, and states that if this script is being called as the\n# primary script then execute the following commands\nif __name__ == "__main__":\n    # Call the above function\n    main()\n'
2021-03-24 14:50:02,878 - mu.modes.microbit:229(find_microbit) INFO: Path to micro:bit: /Volumes/MICROBIT
2021-03-24 14:50:02,879 - mu.modes.microbit:235(find_microbit) INFO: Serial port: /dev/cu.usbmodem141202
2021-03-24 14:50:02,879 - mu.modes.microbit:236(find_microbit) INFO: Device serial number: 9901000051114e450079800e000000440000000097969901
2021-03-24 14:50:02,879 - mu.modes.microbit:237(find_microbit) INFO: Board ID: 0x9901
2021-03-24 14:50:02,879 - mu.modes.microbit:358(flash) INFO: Checking target device.
2021-03-24 14:50:03,436 - mu.modes.microbit:246(get_device_micropython_version) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '1.0.1', 'version': 'micro:bit v1.0.1+b0bf4a9 on 2018-12-13; MicroPython v1.9.2-34-gd64154c73 on 2017-09-01', 'machine': 'micro:bit with nRF51822'}
2021-03-24 14:50:03,436 - mu.modes.microbit:254(get_device_micropython_version) INFO: Board MicroPython: 1.0.1
2021-03-24 14:50:03,436 - mu.modes.microbit:367(flash) INFO: Mu MicroPython: 1.0.1
2021-03-24 14:50:03,437 - mu.modes.microbit:474(copy_main) INFO: Copying main.py onto device
2021-03-24 14:50:03,437 - mu.modes.microbit:481(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', 'f(b\'"""Exercise 0 - Hello, World!"""\\n\\n# An import statement - we are\')', "f(b' adding logic from the display module of the microbit library to')", "f(b' this code\\nfrom microbit import display\\n\\n# Our function that is ')", "f(b'called\\ndef main():\\n    # See https://microbit-micropython.readth')", 'f(b\'edocs.io/en/v1.0.1/display.html\\n    display.scroll("Nicholas was\')', 'f(b\' here")\\n\\n# This is a standard Python construction, and states th\')', "f(b'at if this script is being called as the\\n# primary script then e')", 'f(b\'xecute the following commands\\nif __name__ == "__main__":\\n    # C\')', "f(b'all the above function\\n    main()\\n')", 'fd.close()']
2021-03-24 14:50:04,478 - mu.modes.microbit:408(flash) WARNING: Could not copy file to device.
2021-03-24 14:50:04,479 - mu.modes.microbit:409(flash) ERROR: Could not enter raw REPL.
2021-03-24 14:50:04,479 - mu.modes.microbit:410(flash) INFO: Falling back to old-style flashing.
2021-03-24 14:50:04,479 - mu.modes.microbit:444(flash_attached) INFO: Flashing new MicroPython runtime onto device
2021-03-24 14:50:04,487 - mu.modes.microbit:102(run) ERROR: [Errno 1] Operation not permitted: '/Volumes/MICROBIT/micropython.hex'
2021-03-24 14:50:04,520 - mu.modes.microbit:497(flash_failed) ERROR: [Errno 1] Operation not permitted: '/Volumes/MICROBIT/micropython.hex'
2021-03-24 14:50:04,526 - mu.interface.main:942(show_message) DEBUG: There was a problem flashing the micro:bit.
2021-03-24 14:50:04,526 - mu.interface.main:943(show_message) DEBUG: Please do not disconnect the device until flashing has completed. Please check the logs for more information.
2021-03-24 14:50:04,569 - mu.modes.microbit:459(flash_finished) INFO: Flashing successful.
2021-03-24 14:50:08,006 - mu.logic:1396(show_admin) INFO: Showing admin with logs from /Users/nicholaspatel/Library/Logs/mu/mu.log
2021-03-24 14:50:08,006 - mu.virtual_environment:697(installed_packages) INFO: Discovering installed third party modules in venv.
2021-03-24 14:50:08,006 - mu.virtual_environment:198(run) DEBUG: About to run blocking: /Users/nicholaspatel/Library/Application Support/mu/mu_venv-38-20210324-124837/bin/pip, ['list', '--disable-pip-version-check'], 30.0
2021-03-24 14:50:08,342 - mu.virtual_environment:717(installed_packages) INFO: []

carlosperate, 2021-03-24:
Can you check if the "terminal" app (located in `/System/Applications/Utilities/Terminal.app`) is in the Full Disk Access list, and if not, can you try adding it and rebooting the computer?

[image]

In my laptop I was able to replicate this issue after revoking terminal access to the full disk.

npat860, 2021-03-24:
Thanks for your help, unfortunately the flash error still comes up though (see log below)
Should I try enable full access for mu-editor or iCoreService?

------------

2021-03-25 00:35:04,702 - mu.logic:1442(show_admin) INFO: No admin settings changed.
2021-03-25 00:35:07,709 - mu.modes.microbit:273(flash) INFO: Preparing to flash script.
2021-03-25 00:35:07,710 - mu.modes.microbit:280(flash) DEBUG: Python script from 'exercise_0.py' tab:
2021-03-25 00:35:07,710 - mu.modes.microbit:281(flash) DEBUG: b'"""Exercise 0 - Hello, World!"""\n\n# An import statement - we are adding logic from the display module of the microbit library to this code\nfrom microbit import display\n\n# Our function that is called\ndef main():\n    # See https://microbit-micropython.readthedocs.io/en/v1.0.1/display.html\n    display.scroll("Nicholas was here")\n\n# This is a standard Python construction, and states that if this script is being called as the\n# primary script then execute the following commands\nif __name__ == "__main__":\n    # Call the above function\n    main()\n'
2021-03-25 00:35:07,719 - mu.modes.microbit:229(find_microbit) INFO: Path to micro:bit: /Volumes/MICROBIT
2021-03-25 00:35:07,719 - mu.modes.microbit:235(find_microbit) INFO: Serial port: /dev/cu.usbmodem141302
2021-03-25 00:35:07,719 - mu.modes.microbit:236(find_microbit) INFO: Device serial number: 9901000051114e450079800e000000440000000097969901
2021-03-25 00:35:07,719 - mu.modes.microbit:237(find_microbit) INFO: Board ID: 0x9901
2021-03-25 00:35:07,719 - mu.modes.microbit:358(flash) INFO: Checking target device.
2021-03-25 00:35:08,268 - mu.modes.microbit:246(get_device_micropython_version) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '1.0.1', 'version': 'micro:bit v1.0.1+b0bf4a9 on 2018-12-13; MicroPython v1.9.2-34-gd64154c73 on 2017-09-01', 'machine': 'micro:bit with nRF51822'}
2021-03-25 00:35:08,268 - mu.modes.microbit:254(get_device_micropython_version) INFO: Board MicroPython: 1.0.1
2021-03-25 00:35:08,269 - mu.modes.microbit:367(flash) INFO: Mu MicroPython: 1.0.1
2021-03-25 00:35:08,269 - mu.modes.microbit:474(copy_main) INFO: Copying main.py onto device
2021-03-25 00:35:08,269 - mu.modes.microbit:481(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', 'f(b\'"""Exercise 0 - Hello, World!"""\\n\\n# An import statement - we are\')', "f(b' adding logic from the display module of the microbit library to')", "f(b' this code\\nfrom microbit import display\\n\\n# Our function that is ')", "f(b'called\\ndef main():\\n    # See https://microbit-micropython.readth')", 'f(b\'edocs.io/en/v1.0.1/display.html\\n    display.scroll("Nicholas was\')', 'f(b\' here")\\n\\n# This is a standard Python construction, and states th\')', "f(b'at if this script is being called as the\\n# primary script then e')", 'f(b\'xecute the following commands\\nif __name__ == "__main__":\\n    # C\')', "f(b'all the above function\\n    main()\\n')", 'fd.close()']
2021-03-25 00:35:09,311 - mu.modes.microbit:408(flash) WARNING: Could not copy file to device.
2021-03-25 00:35:09,311 - mu.modes.microbit:409(flash) ERROR: Could not enter raw REPL.
2021-03-25 00:35:09,311 - mu.modes.microbit:410(flash) INFO: Falling back to old-style flashing.
2021-03-25 00:35:09,311 - mu.modes.microbit:444(flash_attached) INFO: Flashing new MicroPython runtime onto device
2021-03-25 00:35:09,319 - mu.modes.microbit:102(run) ERROR: [Errno 1] Operation not permitted: '/Volumes/MICROBIT/micropython.hex'
2021-03-25 00:35:09,351 - mu.modes.microbit:497(flash_failed) ERROR: [Errno 1] Operation not permitted: '/Volumes/MICROBIT/micropython.hex'
2021-03-25 00:35:09,356 - mu.interface.main:942(show_message) DEBUG: There was a problem flashing the micro:bit.
2021-03-25 00:35:09,357 - mu.interface.main:943(show_message) DEBUG: Please do not disconnect the device until flashing has completed. Please check the logs for more information.
2021-03-25 00:35:09,392 - mu.modes.microbit:459(flash_finished) INFO: Flashing successful.
2021-03-25 00:35:12,717 - mu.logic:1396(show_admin) INFO: Showing admin with logs from /Users/nicholaspatel/Library/Logs/mu/mu.log
2021-03-25 00:35:12,717 - mu.virtual_environment:697(installed_packages) INFO: Discovering installed third party modules in venv.
2021-03-25 00:35:12,717 - mu.virtual_environment:198(run) DEBUG: About to run blocking: /Users/nicholaspatel/Library/Application Support/mu/mu_venv-38-20210324-124837/bin/pip, ['list', '--disable-pip-version-check'], 30.0
2021-03-25 00:35:13,054 - mu.virtual_environment:717(installed_packages) INFO: []

blsz, 2021-03-24:
In MacOS 11.2.2 Big Sur, I was able to flash the micro:bit with Mu 1.1.0.beta.2 after giving Full Disk Access to "sh" in System Preferences Privacy tab.   I don't understand why I have  both "sh" and "Terminal".  

Mu flashes the micro:bit but still reports errors:

```
2021-03-24 13:10:29,175 - mu.modes.microbit:273(flash) INFO: Preparing to flash script.
2021-03-24 13:10:29,176 - mu.modes.microbit:280(flash) DEBUG: Python script from 'untitled' tab:
2021-03-24 13:10:29,176 - mu.modes.microbit:281(flash) DEBUG: b'from microbit import *\ndisplay.scroll("Hello, World!")\n'
2021-03-24 13:10:29,184 - mu.modes.microbit:229(find_microbit) INFO: Path to micro:bit: /Volumes/MICROBIT
2021-03-24 13:10:29,184 - mu.modes.microbit:235(find_microbit) INFO: Serial port: /dev/cu.usbmodem14102
2021-03-24 13:10:29,184 - mu.modes.microbit:236(find_microbit) INFO: Device serial number: 9904360258434e45001b400e0000004f000000009796990b
2021-03-24 13:10:29,184 - mu.modes.microbit:237(find_microbit) INFO: Board ID: 0x9904
2021-03-24 13:10:29,184 - mu.modes.microbit:358(flash) INFO: Checking target device.
2021-03-24 13:10:29,714 - mu.modes.microbit:246(get_device_micropython_version) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '2.0.0-beta.4', 'version': 'micro:bit v2.0.0-beta.4+7c8e3e8 on 2021-02-19; MicroPython v1.13 on 2021-02-19', 'machine': 'micro:bit with nRF52833'}
2021-03-24 13:10:29,715 - mu.modes.microbit:254(get_device_micropython_version) INFO: Board MicroPython: 2.0.0-beta.4
2021-03-24 13:10:29,715 - mu.modes.microbit:367(flash) INFO: Mu MicroPython: 2.0.0-beta.4
2021-03-24 13:10:29,715 - mu.modes.microbit:474(copy_main) INFO: Copying main.py onto device
2021-03-24 13:10:29,716 - mu.modes.microbit:481(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', 'f(b\'from microbit import *\\ndisplay.scroll("Hello, World!")\\n\')', 'fd.close()']
2021-03-24 13:10:30,753 - mu.modes.microbit:408(flash) WARNING: Could not copy file to device.
2021-03-24 13:10:30,753 - mu.modes.microbit:409(flash) ERROR: Could not enter raw REPL.
2021-03-24 13:10:30,753 - mu.modes.microbit:410(flash) INFO: Falling back to old-style flashing.
2021-03-24 13:10:30,754 - mu.modes.microbit:444(flash_attached) INFO: Flashing new MicroPython runtime onto device
2021-03-24 13:10:35,299 - mu.modes.microbit:459(flash_finished) INFO: Flashing successful.
```

blsz, 2021-03-24:
Mu 1.1.0.beta.2 on MacOS 11.2.3 Big Sur was failing to flash the micro:bit until Full Disk Access was also given to  "sh".  I don't know why I have both "Terminal" and "sh" in System Preferences > Privacy > Full Disk Access.

[image]

Mu flashes the micro:bit but still shows errors:

```
2021-03-24 13:10:29,175 - mu.modes.microbit:273(flash) INFO: Preparing to flash script.
2021-03-24 13:10:29,176 - mu.modes.microbit:280(flash) DEBUG: Python script from 'untitled' tab:
2021-03-24 13:10:29,176 - mu.modes.microbit:281(flash) DEBUG: b'from microbit import *\ndisplay.scroll("Hello, World!")\n'
2021-03-24 13:10:29,184 - mu.modes.microbit:229(find_microbit) INFO: Path to micro:bit: /Volumes/MICROBIT
2021-03-24 13:10:29,184 - mu.modes.microbit:235(find_microbit) INFO: Serial port: /dev/cu.usbmodem14102
2021-03-24 13:10:29,184 - mu.modes.microbit:236(find_microbit) INFO: Device serial number: 9904360258434e45001b400e0000004f000000009796990b
2021-03-24 13:10:29,184 - mu.modes.microbit:237(find_microbit) INFO: Board ID: 0x9904
2021-03-24 13:10:29,184 - mu.modes.microbit:358(flash) INFO: Checking target device.
2021-03-24 13:10:29,714 - mu.modes.microbit:246(get_device_micropython_version) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '2.0.0-beta.4', 'version': 'micro:bit v2.0.0-beta.4+7c8e3
e8 on 2021-02-19; MicroPython v1.13 on 2021-02-19', 'machine': 'micro:bit with nRF52833'}
2021-03-24 13:10:29,715 - mu.modes.microbit:254(get_device_micropython_version) INFO: Board MicroPython: 2.0.0-beta.4
2021-03-24 13:10:29,715 - mu.modes.microbit:367(flash) INFO: Mu MicroPython: 2.0.0-beta.4
2021-03-24 13:10:29,715 - mu.modes.microbit:474(copy_main) INFO: Copying main.py onto device
2021-03-24 13:10:29,716 - mu.modes.microbit:481(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', 'f(b\'from microbit import *\\ndisplay.scroll("Hello, World!")\\n\')', 'fd.close()']
2021-03-24 13:10:30,753 - mu.modes.microbit:408(flash) WARNING: Could not copy file to device.
2021-03-24 13:10:30,753 - mu.modes.microbit:409(flash) ERROR: Could not enter raw REPL.
2021-03-24 13:10:30,753 - mu.modes.microbit:410(flash) INFO: Falling back to old-style flashing.
2021-03-24 13:10:30,754 - mu.modes.microbit:444(flash_attached) INFO: Flashing new MicroPython runtime onto device
2021-03-24 13:10:35,299 - mu.modes.microbit:459(flash_finished) INFO: Flashing successful.
```

sunruoshi, 2021-03-25:
> Thank you very much for the log, that helps a lot.
>
> Oddly enough it looks like your Mu app is still not able to access the micro:bit drive. Could you try to manually give it access via the "Security [...]

I give Terminal the full access and reboot the computer, but still not working.

New logs:

2021-03-25 09:22:32,082 - mu.logic:889(restore_session) INFO: Loaded files.
2021-03-25 09:22:32,082 - mu.logic:892(restore_session) INFO: User defined environment variables: []
2021-03-25 09:22:32,083 - mu.logic:897(restore_session) INFO: Minify scripts on micro:bit? False
2021-03-25 09:22:32,143 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/sam/mu_code from settings file
2021-03-25 09:22:32,143 - mu.logic:1523(change_mode) INFO: Workspace directory: /Users/sam/mu_code
2021-03-25 09:22:32,712 - mu.logic:737(check_usb) INFO: microbit device connected on port: /dev/cu.usbmodem14402(VID: 0x0D28, PID: 0x0204, manufacturer: 'ARM')
2021-03-25 09:22:51,851 - mu.modes.microbit:273(flash) INFO: Preparing to flash script.
2021-03-25 09:22:51,851 - mu.modes.microbit:280(flash) DEBUG: Python script from 'compass.py' tab:
2021-03-25 09:22:51,851 - mu.modes.microbit:281(flash) DEBUG: b'# Write your code here :-)\nfrom microbit import *\n\ncompass.calibrate()\n\nwhile True:\n    bearing = compass.heading()\n    if bearing < 45 or bearing > 315:\n        display.show("N")\n    elif bearing > 45 and bearing < 135:\n        display.show("E")\n    elif bearing > 135 and bearing < 225:\n        display.show("S")\n    elif bearing > 225 and bearing < 315:\n        display.show("W")\n    else:\n        display.show(" ")\n'
2021-03-25 09:22:51,858 - mu.modes.microbit:229(find_microbit) INFO: Path to micro:bit: /Volumes/MICROBIT
2021-03-25 09:22:51,858 - mu.modes.microbit:235(find_microbit) INFO: Serial port: /dev/cu.usbmodem14402
2021-03-25 09:22:51,858 - mu.modes.microbit:236(find_microbit) INFO: Device serial number: 9904360258994e45003f900a0000006b000000009796990b
2021-03-25 09:22:51,858 - mu.modes.microbit:237(find_microbit) INFO: Board ID: 0x9904
2021-03-25 09:22:51,858 - mu.modes.microbit:358(flash) INFO: Checking target device.
2021-03-25 09:22:52,998 - mu.modes.microbit:376(flash) WARNING: Could not detect version of MicroPython.
2021-03-25 09:22:53,038 - mu.modes.microbit:423(flash_and_send) INFO: Flashing new MicroPython runtime onto device
2021-03-25 09:22:53,060 - mu.modes.microbit:102(run) ERROR: [Errno 1] Operation not permitted: '/Volumes/MICROBIT/micropython.hex'
2021-03-25 09:22:53,061 - mu.modes.microbit:497(flash_failed) ERROR: [Errno 1] Operation not permitted: '/Volumes/MICROBIT/micropython.hex'
2021-03-25 09:22:53,082 - mu.interface.main:942(show_message) DEBUG: There was a problem flashing the micro:bit.
2021-03-25 09:22:53,082 - mu.interface.main:943(show_message) DEBUG: Please do not disconnect the device until flashing has completed. Please check the logs for more information.
2021-03-25 09:22:53,123 - mu.modes.microbit:459(flash_finished) INFO: Flashing successful.
2021-03-25 09:22:53,123 - mu.modes.microbit:474(copy_main) INFO: Copying main.py onto device
2021-03-25 09:22:53,123 - mu.modes.microbit:481(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', "f(b'# Write your code here :-)\\nfrom microbit import *\\n\\ncompass.calib')", "f(b'rate()\\n\\nwhile True:\\n    bearing = compass.heading()\\n    if beari')", 'f(b\'ng < 45 or bearing > 315:\\n        display.show("N")\\n    elif bea\')', 'f(b\'ring > 45 and bearing < 135:\\n        display.show("E")\\n    elif \')', 'f(b\'bearing > 135 and bearing < 225:\\n        display.show("S")\\n    e\')', 'f(b\'lif bearing > 225 and bearing < 315:\\n        display.show("W")\\n \')', 'f(b\'   else:\\n        display.show(" ")\\n\')', 'fd.close()']
2021-03-25 09:22:54,160 - mu.modes.microbit:497(flash_failed) ERROR: Could not enter raw REPL.
2021-03-25 09:22:54,160 - mu.interface.main:942(show_message) DEBUG: There was a problem flashing the micro:bit.
2021-03-25 09:22:54,160 - mu.interface.main:943(show_message) DEBUG: Please do not disconnect the device until flashing has completed. Please check the logs for more information.
2021-03-25 09:22:56,745 - mu.logic:1396(show_admin) INFO: Showing admin with logs from /Users/sam/Library/Logs/mu/mu.log
2021-03-25 09:22:56,745 - mu.virtual_environment:697(installed_packages) INFO: Discovering installed third party modules in venv.
2021-03-25 09:22:56,745 - mu.virtual_environment:198(run) DEBUG: About to run blocking: /Users/sam/Library/Application Support/mu/mu_venv-38-20210322-154202/bin/pip, ['list', '--disable-pip-version-check'], 30.0
2021-03-25 09:22:57,319 - mu.virtual_environment:717(installed_packages) INFO: []

fhassancq, 2021-03-25:
> Can you check if the "terminal" app (located in `/System/Applications/Utilities/Terminal.app`) is in the Full Disk Access list, and if not, can you try adding it and rebooting the computer? (rebooting is important for this to work)
> 
> [image]
> 
> In my laptop I was able to replicate this issue after revoking terminal access to the full disk.

I tried this, but still doesn't flash.

[image]

marc0janssen, 2021-03-30:
I have exactly the same issue

marc0janssen, 2021-03-30:
How I fixed it:

1. in FINDER go to the folder /bin
2. open the system setting for security and privacy
3. select privacy
4. Unlock with the pad-lock the screen
5. drag and drop /bin/sh in the "right-side" part of the window where the apps are .
6. tick the box next to the just copied "SH".

This fixed it for me...

npat860, 2021-03-31:
Thanks, that worked for me!

fhassancq, 2021-03-31:
The default folder location in FINDER is OneDrive. I cannot find the /bin folder. If I search for it, the trash can folder open up with items I have deleted.

marc0janssen, 2021-03-31:
Hit Command+Shift+G from the Mac OS X desktop or a Finder window

Enter /bin

Look for sh

Drag and drop this to the privacy pane. 

Does this help?

fhassancq, 2021-03-31:
Eureka!!! It works!!! Thank you so much. :)

marc0janssen, 2021-03-31:
Glad to hear!

microbit-mark, 2021-06-22:
This issue reproduces in Mu 1.1.0.beta.4 on macOS 11.2.3

The workaround mentioned in https://github.com/mu-editor/mu/issues/1383#issuecomment-806196939 and https://github.com/mu-editor/mu/issues/1383#issuecomment-810537546 to allow full disk access to `/bin/sh` resolves the issue. Thanks @blsz @marc0janssen 👍🏻

mitch46syd, 2021-07-06:
As an amateur using Microbit and Github, I want to use Mu to develop microbit programs. I have the ElectricFreaks Nezha kit and would like to use Mu instead of MakeCode. 
Can I install the Nehza, PlanetX etc libraires using pip or some other tool.
Your advice most appreciated

closed 2021-10-06 by carlosperate