mu-editor/mu #1384: Crash Report

16Hyperwolf16, 2021-03-18:
Version: 1.1.0.beta.2
Locale: fr_FR
Platform: Windows 10 10.0.18362 AMD64

```
Traceback (most recent call last):
  File "C:\Users\adapo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\microbit.py", line 291, in flash
    path_to_microbit, port, board_id = self.find_microbit()
  File "C:\Users\adapo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\microbit.py", line 234, in find_microbit
    board_id = int(serial_number[:4], 16)
ValueError: invalid literal for int() with base 16: ''

```

tjguk, 2021-03-18:
@carlosperate  -- one for you to look at, I imagine?

16Hyperwolf16, 2021-03-19:
Hello
Finally it's good i have relaunch mu and it works perfectly

Sorry,
Have a nice day 😀

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

carlosperate, 2021-03-19:
Thanks for the report @16Hyperwolf16, I'm glad it is now working. If it happens again and let us know that would be great too, in case this is not a 1-off.

I'll update that part of the code to be a bit more resilient to missing USB serial strings

16Hyperwolf16, 2021-03-21:
Ok thanks 👍