mu-editor/mu #1387: MU will not start

jeskdm, 2021-03-20:
Mu used to be working fine on my computer but now it does not start correctly. When I launch Mu.launch.pyw I get the flashscreen and then I get MU in the taskbar but the window is not displaying and there is no way to maximize it. This is Win 10

tjguk, 2021-03-20:
@jeskdm can you upload a copy of the logs to this Issue please? You can find them in:

`%LOCALAPPDATA%\python\mu\logs\mu.log`

Also -- has anything changed from your end? eg did you install an updated copy of Mu? Or upgrade your OS?

jeskdm, 2021-03-20:
Before I send the logs etc. is there some way I can open MU and prevent it from opening previous open files etc. I suspect something is going wrong with one of the files it tries to open when I start the application.

tjguk, 2021-03-20:
Yes; there'll be some settings file which contain the list of saved files. Start by looking in:

`%LOCALAPPDATA%\python\mu`

for `.json` files. Depending on which version of Mu you're using they might be somewhere else. If you see a `session.json` file, delete it (or rename it) and try starting Mu again.

jeskdm, 2021-03-20:
Ok, I renamed the session.json file and it started fine now! Thanks for the advise 👍 🥇

tjguk, 2021-03-20:
Thanks for letting us know :)

closed 2021-03-20 by tjguk

carlosperate, 2021-03-22:
@jeskdm did you find out what file might have caused Mu to crash? If you could send it to us (or copy the contents in a response) that could help use replicate the issue and try to fix it.

jeskdm, 2021-03-23:
Ok I did some research and found the bug. The issue is that the path contained the Norwegian character "Å". In the json file this was converted to ...\\V\u00c5R 2021\\PYTHON KODE\\...  causing MU to crash on startup. I will assume the issue will happen with the other norwegian characters Æ and Ø

tjguk, 2021-03-23:
Brilliant. Thank you very much for looking into it @jeskdm. We've worked quite hard to make sure that the *code* files are unicode-happy but the more extensive use of JSON is newer. I'll raise a separate issue to track that.

jeskdm, 2021-03-23:
Glad I could help, I really like this great tool :-)