mu-editor/mu #1391: Crash Report

veryalien, 2021-03-21:
Version: 1.1.0.beta.2
Locale: en_US
Platform: Linux 4.15.0-139-generic #143-Ubuntu SMP Tue Mar 16 01:30:17 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/neil/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 441, in ensure_and_create
    self.ensure()
  File "/home/neil/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "/home/neil/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: /home/neil/.local/share/mu/mu_venv-36-20210321-214217 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/neil/.local/lib/python3.6/site-packages/mu/app.py", line 162, in run
    raise ex
  File "/home/neil/.local/lib/python3.6/site-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/neil/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "/home/neil/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 596, in create
    self.install_baseline_packages()
  File "/home/neil/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 649, in install_baseline_packages
    wheels.download()
  File "/home/neil/.local/lib/python3.6/site-packages/mu/wheels/__init__.py", line 95, in download
    check=True,
  File "/usr/lib/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'download', '--destination-directory', '/home/neil/.local/lib/python3.6/site-packages/mu/wheels', 'esptool==3.*']' returned non-zero exit status 1.

```

tjguk, 2021-03-22:
Thanks for the report @veryalien -- would you be able to attach the entire log file to this GH issue, please? (Feel free to review first to exclude anything you don't want public).

You should be able to find it under: `~/.cache/mu/`

carlosperate, 2021-04-12:
Hi @veryalien, if you could attach the log file that would be really useful.

Could you also update pip to the latest version and try again? Hopefully that should resolve your issue.

carlosperate, 2021-08-11:
Hi @veryalien,

Would you be able to check if beta.5 works for you and let us know the result?
`pip install mu-editor==1.1.0b5`

Thanks!

carlosperate, 2021-12-07:
Thanks for the solution @nikhilm, that sounds very similar to this crash report: https://github.com/mu-editor/mu/issues/1906#issuecomment-987083383