mu-editor/mu #1392: Crash Report

Henderson3, 2021-03-22:
Version: 1.1.0.beta.2
Locale: en_US
Platform: Linux 5.4.0-67-generic #75-Ubuntu SMP Fri Feb 19 18:03:38 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/arthur/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 441, in ensure_and_create
    self.ensure()
  File "/home/arthur/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "/home/arthur/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: /home/arthur/.local/share/mu/mu_venv-38-20210321-223021 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/arthur/.local/lib/python3.8/site-packages/mu/app.py", line 162, in run
    raise ex
  File "/home/arthur/.local/lib/python3.8/site-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/arthur/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "/home/arthur/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 596, in create
    self.install_baseline_packages()
  File "/home/arthur/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 649, in install_baseline_packages
    wheels.download()
  File "/home/arthur/.local/lib/python3.8/site-packages/mu/wheels/__init__.py", line 82, in download
    process = subprocess.run(
  File "/usr/lib/python3.8/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'download', '--destination-directory', '/home/arthur/.local/lib/python3.8/site-packages/mu/wheels', 'git+https://github.com/mu-editor/pgzero.git@mu-wheel']' returned non-zero exit status 1.

```

tjguk, 2021-03-22:
Thanks for the report, @Henderson3 -- this looks like https://github.com/mu-editor/mu/issues/1369 which has been addressed (at least for the interim) by https://github.com/mu-editor/mu/pull/1376.

So our next beta release (hopefully later this week) should get things working for you

closed 2021-03-22 by tjguk

carlosperate, 2021-03-22:
@Henderson3 in the meantime, to confirm that this is the same problem, could you try running this on a terminal and copying the output here?

```
git --version
```