mu-editor/mu #1396: Crash Report

coryjquirk, 2021-03-23:
thanks squad

Version: 1.1.0.beta.2
Locale: en_US
Platform: Windows 10 10.0.18362 AMD64

```
Traceback (most recent call last):
  File "C:\Users\coryq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 441, in ensure_and_create
    self.ensure()
  File "C:\Users\coryq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "C:\Users\coryq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201429 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\coryq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\coryq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\coryq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "C:\Users\coryq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 582, in create
    subprocess.run(
  File "C:\Users\coryq\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-m', 'virtualenv', '-p', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-q', 'C:\\Users\\coryq\\AppData\\Local\\python\\mu\\mu_venv-38-20210322-201430']' returned non-zero exit status 1.

```

ZanderBrown, 2021-03-23:
Interesting, can you fish out the full log? Should be somewhere like `%LOCALAPPDATA%\python\mu\logs\mu.log`

At first glance this looks similar to #1366

coryjquirk, 2021-03-23:
here is my log, it stops once I successfully opened Mu for the first time on this computer.
```
2021-03-22 20:14:29,102 - root:259(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.2
2021-03-22 20:14:29,102 - root:260(run) INFO: uname_result(system='Windows', node='cq-HP', release='10', version='10.0.18362', machine='AMD64', processor='Intel64 Family 6 Model 126 Stepping 5, GenuineIntel')
2021-03-22 20:14:29,102 - root:261(run) INFO: Python path: ['C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-03-22 20:14:29,103 - root:262(run) INFO: Language code: en_US
2021-03-22 20:14:29,103 - mu.settings:220(load) WARNING: No settings file found at C:\Users\coryq\AppData\Local\python\mu\settings.json; skipping
2021-03-22 20:14:29,103 - mu.settings:220(load) WARNING: No settings file found at C:\Users\coryq\AppData\Local\python\mu\session.json; skipping
2021-03-22 20:14:29,483 - mu.virtual_environment:434(ensure_and_create) INFO: Added log handler.
2021-03-22 20:14:29,986 - root:259(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.2
2021-03-22 20:14:29,986 - root:260(run) INFO: uname_result(system='Windows', node='cq-HP', release='10', version='10.0.18362', machine='AMD64', processor='Intel64 Family 6 Model 126 Stepping 5, GenuineIntel')
2021-03-22 20:14:29,986 - root:261(run) INFO: Python path: ['C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\coryq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-03-22 20:14:29,986 - root:262(run) INFO: Language code: en_US
2021-03-22 20:14:29,986 - mu.settings:220(load) WARNING: No settings file found at C:\Users\coryq\AppData\Local\python\mu\settings.json; skipping
2021-03-22 20:14:29,987 - mu.settings:220(load) WARNING: No settings file found at C:\Users\coryq\AppData\Local\python\mu\session.json; skipping
2021-03-22 20:14:30,045 - mu.virtual_environment:434(ensure_and_create) INFO: Added log handler.
2021-03-22 20:14:30,198 - mu.virtual_environment:438(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-03-22 20:14:30,222 - mu.virtual_environment:479(ensure_path) ERROR: C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201405 does not exist.
2021-03-22 20:14:30,223 - mu.virtual_environment:443(ensure_and_create) DEBUG: Virtual environment not present or correct.
2021-03-22 20:14:30,236 - mu.virtual_environment:445(ensure_and_create) DEBUG: Creating new virtual environment at C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201430.
2021-03-22 20:14:30,247 - mu.virtual_environment:377(relocate) DEBUG: Virtual environment set up mu_venv-38-20210322-201430 at C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201430
2021-03-22 20:14:30,248 - mu.virtual_environment:578(create) INFO: Creating virtualenv: C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201430
2021-03-22 20:14:30,248 - mu.virtual_environment:579(create) INFO: Virtualenv name: mu_venv-38-20210322-201430
2021-03-22 20:14:30,643 - mu.virtual_environment:438(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-03-22 20:14:30,650 - mu.virtual_environment:479(ensure_path) ERROR: C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201429 does not exist.
2021-03-22 20:14:30,656 - mu.virtual_environment:443(ensure_and_create) DEBUG: Virtual environment not present or correct.
2021-03-22 20:14:30,673 - mu.virtual_environment:445(ensure_and_create) DEBUG: Creating new virtual environment at C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201430.
[... cut ...]
2021-03-22 20:16:10,878 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: esptool
2021-03-22 20:16:11,190 - mu.virtual_environment:451(ensure_and_create) INFO: Virtual environment already exists.
2021-03-22 20:16:12,418 - mu.logic:758(__init__) INFO: Setting up editor.
2021-03-22 20:16:12,419 - mu.logic:778(__init__) INFO: Log directory: C:\Users\coryq\AppData\Local\python\mu\Logs
2021-03-22 20:16:12,419 - mu.logic:779(__init__) INFO: Data directory: C:\Users\coryq\AppData\Local\python\mu
```

ZanderBrown, 2021-03-23:
Thanks!

Looks like the relevant section is
```
2021-03-22 20:14:29,986 - mu.settings:220(load) WARNING: No settings file found at C:\Users\coryq\AppData\Local\python\mu\settings.json; skipping
2021-03-22 20:14:29,987 - mu.settings:220(load) WARNING: No settings file found at C:\Users\coryq\AppData\Local\python\mu\session.json; skipping
2021-03-22 20:14:30,045 - mu.virtual_environment:434(ensure_and_create) INFO: Added log handler.
2021-03-22 20:14:30,198 - mu.virtual_environment:438(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-03-22 20:14:30,222 - mu.virtual_environment:479(ensure_path) ERROR: C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201405 does not exist.
2021-03-22 20:14:30,223 - mu.virtual_environment:443(ensure_and_create) DEBUG: Virtual environment not present or correct.
2021-03-22 20:14:30,236 - mu.virtual_environment:445(ensure_and_create) DEBUG: Creating new virtual environment at C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201430.
2021-03-22 20:14:30,247 - mu.virtual_environment:377(relocate) DEBUG: Virtual environment set up mu_venv-38-20210322-201430 at C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201430
2021-03-22 20:14:30,248 - mu.virtual_environment:578(create) INFO: Creating virtualenv: C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201430
2021-03-22 20:14:30,248 - mu.virtual_environment:579(create) INFO: Virtualenv name: mu_venv-38-20210322-201430
2021-03-22 20:14:30,643 - mu.virtual_environment:438(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-03-22 20:14:30,650 - mu.virtual_environment:479(ensure_path) ERROR: C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201429 does not exist.
2021-03-22 20:14:30,656 - mu.virtual_environment:443(ensure_and_create) DEBUG: Virtual environment not present or correct.
2021-03-22 20:14:30,673 - mu.virtual_environment:445(ensure_and_create) DEBUG: Creating new virtual environment at C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201430.
2021-03-22 20:14:30,683 - mu.virtual_environment:377(relocate) DEBUG: Virtual environment set up mu_venv-38-20210322-201430 at C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201430
2021-03-22 20:14:30,692 - mu.virtual_environment:578(create) INFO: Creating virtualenv: C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201430
2021-03-22 20:14:30,692 - mu.virtual_environment:579(create) INFO: Virtualenv name: mu_venv-38-20210322-201430
2021-03-22 20:14:42,809 - root:174(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\coryq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 441, in ensure_and_create
    self.ensure()
  File "C:\Users\coryq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "C:\Users\coryq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201429 does not exist.
```
Which, at least to me, seems to be:
- Check for `mu_venv-38-20210322-201405` (doesn't exist)
- Create `mu_venv-38-20210322-201430`
- Check for `mu_venv-38-20210322-201429` (doesn't exist)
- Create `mu_venv-38-20210322-201430`
- Check for `mu_venv-38-20210322-201429` (doesn't exist)
- Give up

Combined with the timestamps it looks to me like you got very unlucky and time, as is often the case in programming, got the better of us

----

*time passes*

Okay I've ended up looking at this for far longer than expected and have looked at all sorts of things (all the way down to glibc, though of course that probably isn't involved on Windows, as well as pytime.c etc)

Then I noticed two `attempt #1` so I think my conclusion is StartupWorker is running twice (????). Some thread+time shenanigans going on here.

carlosperate, 2021-03-24:
Hi @coryjquirk,

Does that mean that Mu failed to open the first time, but then after trying again it successfully opened and worked?

coryjquirk, 2021-03-24:
@carlosperate I believe once Mu installation was complete, it failed to open automatically on install. I proceeded to open the application from the Windows start menu and began working with the program. 

Here are the logs from directly immediately after the first group of logs I posted.
```
2021-03-22 20:16:12,419 - mu.logic:793(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2021-03-22 20:16:12,420 - mu.modes.base:68(get_default_workspace) WARNING: Workspace C:\Users\coryq\mu_code in the settings file is not a valid directory; using default C:\Users\coryq\mu_code
2021-03-22 20:16:12,420 - mu.logic:797(setup) DEBUG: Creating directory: C:\Users\coryq\mu_code
2021-03-22 20:16:12,421 - mu.logic:805(setup) DEBUG: Creating directory: C:\Users\coryq\mu_code\images
2021-03-22 20:16:12,530 - mu.logic:812(setup) DEBUG: Creating directory: C:\Users\coryq\mu_code\fonts
2021-03-22 20:16:12,531 - mu.logic:815(setup) DEBUG: Creating directory: C:\Users\coryq\mu_code\sounds
2021-03-22 20:16:12,631 - mu.logic:822(setup) DEBUG: Creating directory: C:\Users\coryq\mu_code\music
2021-03-22 20:16:12,631 - mu.logic:828(setup) DEBUG: Creating directory: C:\Users\coryq\mu_code\templates
2021-03-22 20:16:12,666 - mu.logic:831(setup) DEBUG: Creating directory: C:\Users\coryq\mu_code\static
2021-03-22 20:16:12,953 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\coryq\mu_code from settings file
2021-03-22 20:16:12,953 - mu.logic:1523(change_mode) INFO: Workspace directory: C:\Users\coryq\mu_code
2021-03-22 20:16:12,954 - mu.logic:867(restore_session) DEBUG: <SessionSettings from C:\Users\coryq\AppData\Local\python\mu\session.json>
2021-03-22 20:16:12,964 - mu.logic:1470(select_mode) INFO: Showing available modes: ['python', 'circuitpython', 'microbit', 'esp', 'web', 'pyboard', 'debugger', 'pygamezero', 'lego', 'pico']
2021-03-22 20:17:01,329 - mu.logic:1477(select_mode) INFO: New mode selected: circuitpython
2021-03-22 20:17:01,365 - mu.logic:1523(change_mode) INFO: Workspace directory: D:\
2021-03-22 20:17:01,406 - mu.logic:1523(change_mode) INFO: Workspace directory: D:\
2021-03-22 20:17:01,772 - mu.logic:943(restore_session) INFO: Starting with blank file.
2021-03-22 20:17:10,994 - mu.logic:1115(get_dialog_directory) INFO: Using path for file dialog: D:\
2021-03-22 20:17:13,665 - mu.interface.main:405(get_load_path) DEBUG: Getting load path: D:/code.py
2021-03-22 20:17:13,665 - mu.logic:978(_load) INFO: Loading script from: D:/code.py
2021-03-22 20:17:13,665 - mu.logic:316(read_and_decode) DEBUG: Trying to decode with utf-8
2021-03-22 20:17:13,665 - mu.logic:319(read_and_decode) INFO: Decoded with utf-8
2021-03-22 20:17:13,666 - mu.logic:332(read_and_decode) DEBUG: Detected newline '\r\n'
2021-03-22 20:17:13,666 - mu.logic:1077(_load) DEBUG: import time
import board
from digitalio import DigitalInOut, Direction, Pull

led = DigitalInOut(board.P0_07)
led.direction = Direction.OUTPUT

button = DigitalInOut(board.P0_13)
button.direction = Direction.INPUT
button.pull = Pull.UP

while True:
    if not button.value:
        print("Button pressed!")
        led.value = True
    else:
        led.value = False

    time.sleep(0.01)

2021-03-22 20:20:38,574 - mu.logic:955(toggle_theme) INFO: Toggle theme to: night
2021-03-22 20:20:53,812 - mu.logic:1184(save_tab_to_file) INFO: Saving script to: D:/code.py
2021-03-22 20:20:53,812 - mu.logic:1185(save_tab_to_file) DEBUG: import time
import board
from digitalio import DigitalInOut, Direction, Pull

led = DigitalInOut(board.P0_07)
led.direction = Direction.OUTPUT

button = DigitalInOut(board.P0_13)
button.direction = Direction.INPUT
button.pull = Pull.UP

while True:
    if not button.value:
        print("Button pressed!")
        led.value = True
    else:
        led.value = False

    time.sleep(0.01)

2021-03-22 20:51:00,703 - mu.logic:1396(show_admin) INFO: Showing admin with logs from C:\Users\coryq\AppData\Local\python\mu\Logs\mu.log
2021-03-22 20:51:00,703 - mu.virtual_environment:697(installed_packages) INFO: Discovering installed third party modules in venv.
2021-03-22 20:51:00,703 - mu.virtual_environment:198(run) DEBUG: About to run blocking: C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-20210322-201430\scripts\pip.exe, ['list', '--disable-pip-version-check'], 30.0
2021-03-22 20:51:01,227 - mu.virtual_environment:717(installed_packages) INFO: []
2021-03-22 20:51:09,412 - mu.logic:1442(show_admin) INFO: No admin settings changed.
2021-03-22 20:55:13,392 - mu.logic:1184(save_tab_to_file) INFO: Saving script to: D:/code.py
2021-03-22 20:55:13,392 - mu.logic:1185(save_tab_to_file) DEBUG: import board
```

carlosperate, 2021-03-29:
Thanks for the clarification @coryjquirk.

> I believe once Mu installation was complete, it failed to open automatically on install. 

Do you mean you expected it to open automatically after the install, and that didn't happen?
Or do you mean that when Mu was opened after the installation it started to open and then that failed and raised the crash report?

As far as I know (but I might be wrong) the installer doesn't automatically open the Mu Editor, as this feature is not yet added: https://github.com/mu-editor/pup/pull/148

The first entry in your log looks like it might have crashed very early, and then it "crashed properly" showing the crash report website.

coryjquirk, 2021-04-04:
@carlosperate sorry my memory failed me there. I must have tried to ,manually open Mu just after installing it, rather than an automatic launch happening upon install.

carlosperate, 2021-04-20:
Thank you for all the testing @coryjquirk.
Could you try the following in a command prompt to see what virtualenv outputs?

```
"C:\Users\coryq\AppData\Local\Programs\Mu Editor\Python\pythonw.exe" -m virtualenv -p "C:\Users\coryq\AppData\Local\Programs\Mu Editor\Python\pythonw.exe" "C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-temp"
```

You can then remove `C:\Users\coryq\AppData\Local\python\mu\mu_venv-38-temp` if it has created it.

coryjquirk, 2021-04-27:
@carlosperate this command created a `mu_venv-38-temp` directory containing /Lib, /Scripts, .gitignore, and pyenv.cfg

Sorry I'm not more familiar with virtualenv to give you more details than that. I'm happy to continue testing since it helps me learn more about Python and my own computer.

carlosperate, 2021-05-11:
That's great, thanks @coryjquirk.
I was really expecting an error message to be outputted into the terminal, so if it successfully created the virtual environment that means the problem might be somewhere else.

Would you be able to try the latest beta and see if that one works? Might be better to uninstall beta.2 first.

carlosperate, 2021-08-10:
Hi @coryjquirk,
Would you be able to check if beta.5 works for you and let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

coryjquirk, 2021-08-10:
@carlosperate I installed and opened beta.5. it ran without crashing and no errors/exceptions arose in the Mu log. feel free to close issue if you'd like.

carlosperate, 2021-08-10:
I'm glad beta 5 worked, thanks for testing!

closed 2021-08-10 by carlosperate