mu-editor/mu #1397: Crash Report

pcsnow, 2021-03-23:
Initial Install of Mu 

Version: 1.1.0.beta.2
Locale: en_US
Platform: Windows 8.1 6.3.9600 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Paul\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\Paul\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Paul\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 459, in ensure_and_create
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.

```

pcsnow, 2021-03-23:
Initial Install of mu

Connected to itsybitsym4

I know I am down level on microcode. I will update and reattempt

carlosperate, 2021-03-24:
Hi @pcsnow, does that mean that you were able to successfully open Mu v1.1.0-beta.2 after the original issue?

carlosperate, 2021-04-06:
@pcsnow Would you be able to add your log file to this GH Issue? It will be quite useful for debugging. File should be located in `%LOCALAPPDATA%\python\mu\Logs`.

pcsnow, 2021-04-06:
upgrading to windows 10 seemed to solve the problems.
I have no other concerts.

"Trouble came clear when OS updated"

carlosperate, 2021-04-06:
I'm glad is working now!
Do you still have the log though? It would be useful to look at it in case there is something we can fix for other other windows 8.1 users.

carlosperate, 2021-04-20:
Ahhh, okay, I see, that's a useful page to keep in mind as well, thanks for the update!

Mu was able to run successfully after the original crash then? Is Mu currently working for you?

pcsnow, 2021-04-20:
Yes, it works fine.
Thanks

carlosperate, 2021-04-20:
Awesome, in that case I'll close this issue. Thanks for all the testing!

closed 2021-04-20 by carlosperate