mu-editor/mu #1399: JSON files not handling non-ASCII characters well

tjguk, 2021-03-23:
Ok I did some research and found the bug. The issue is that the path contained the Norwegian character "Å". In the json file this was converted to ...\\V\u00c5R 2021\\PYTHON KODE\\...  causing MU to crash on startup. I assume the same issue will happen for the other norwegian characters Æ and Ø as well.

_Originally posted by @jeskdm in https://github.com/mu-editor/mu/issues/1387#issuecomment-804805501_

ZanderBrown, 2021-03-23:
I've tested locally with Python 3.5 and 3.9, both encoude `Norwegian character "Å". I assume the same issue will happen for the other norwegian characters Æ and Ø as well.` as
```json
"Norwegian character \"\u00c5\". I assume the same issue will happen for the other norwegian characters \u00c6 and \u00d8 as well."
```
Is this Windows being Windows?

tjguk, 2021-03-23:
Thanks for testing, @ZanderBrown  -- I'd be surprised it this was to do with Windows as such: it's just whatever encoding the `json` module is doing internally. But anything's possible...

I'll get back to this when I've got some time later

tjguk, 2021-03-23:
@jeskdm -- would you be willing to upload the "corrupted" session file to this GH issue, please?

carlosperate, 2021-03-23:
This was definitely an issue in earlier versions of Mu, before the settings refactoring, as I had a tentative PR I never pushed, after a while I _think_ I tested it again (in macOS) after the refactor was merged and couldn't trigger it again. So, while I could be misremembering this, but I thought it was fixed when threw away my old branch.

I think at the time I was looking at https://github.com/mu-editor/mu/issues/635.