mu-editor/mu #1401: Crash Handler URI too large

carlosperate, 2021-03-23:
While running the tests in an Ubuntu 16.04 VM, the got stuck around `tests/virtual_environment/test_virtual_environment.py` and when I `Ctrl+C` to top the run I encountered the following issue:

```
...
tests/modes/test_pyboard.py::test_pyboard_mode_no_charts PASSED
tests/modes/test_pyboard.py::test_workspace_dir_nt_exists PASSED
tests/virtual_environment/test_virtual_environment.py::test_venv_second_try PASSED
^C^C

----------- coverage: platform linux, python 3.5.2-final-0 -----------
Name                         Stmts   Miss  Cover   Missing
----------------------------------------------------------
mu/__init__.py                   7      0   100%
mu/app.py                      154      1    99%   279
mu/config.py                     8      0   100%
mu/debugger/client.py          200      0   100%
mu/debugger/config.py            1      0   100%
mu/debugger/runner.py          257      0   100%
mu/debugger/utils.py            11      3    73%   30, 37, 40
mu/i18n.py                       6      0   100%
mu/interface/__init__.py         2      0   100%
mu/interface/dialogs.py        331     14    96%   318-319, 624-628, 634-635, 642-650
mu/interface/editor.py         273    122    55%   78-80, 112, 114-115, 119, 130-131, 140-167, 224-225, 311, 325, 331-334, 340-352, 360, 374-388, 394-400, 410-412, 420-429, 445, 465-467, 485, 491, 509-551, 583-596, 603-631
mu/interface/main.py           606      7    99%   152, 673, 725-726, 738, 797, 841
mu/interface/panes.py          820    118    86%   1307-1308, 1314-1316, 1351-1388, 1399-1437, 1446-1515, 1521-1526
mu/interface/themes.py         192      0   100%
mu/interface/widgets.py         79      0   100%
mu/logic.py                    831     24    97%   887, 1100-1104, 1516, 1775-1808
mu/modes/__init__.py            11      0   100%
mu/modes/api/__init__.py        11      0   100%
mu/modes/api/adafruit.py         1      0   100%
mu/modes/api/esp.py              1      0   100%
mu/modes/api/flask.py            1      0   100%
mu/modes/api/lego.py             1      0   100%
mu/modes/api/microbit.py         1      0   100%
mu/modes/api/pi.py               1      0   100%
mu/modes/api/pyboard.py          1      0   100%
mu/modes/api/pygamezero.py       1      0   100%
mu/modes/api/python3.py          1      0   100%
mu/modes/api/shared.py           1      0   100%
mu/modes/base.py               340      0   100%
mu/modes/circuitpython.py       69      0   100%
mu/modes/debugger.py           182      0   100%
mu/modes/esp.py                105      0   100%
mu/modes/lego.py                14      0   100%
mu/modes/microbit.py           313     16    95%   78-90, 325-333, 335-344, 390-397
mu/modes/pico.py                14      0   100%
mu/modes/pyboard.py             58      0   100%
mu/modes/pygamezero.py          65      0   100%
mu/modes/python3.py            180     10    94%   42-58
mu/modes/web.py                107     75    30%   73, 119, 125-139, 146-187, 193-206, 212, 218, 227-229, 238-241, 250-253, 260-272
mu/resources/__init__.py        17      0   100%
mu/settings.py                 122      0   100%
mu/virtual_environment.py      283    143    49%   47-53, 59, 113-118, 121-125, 128-139, 145-154, 157, 160, 163, 166, 212-217, 339, 393-402, 413-424, 433-435, 456-459, 467-471, 477-489, 501-509, 518-542, 548-560, 566-573, 579-599, 606-608, 632-658, 664-666, 672, 678-680, 690-692, 702-725
----------------------------------------------------------
TOTAL                         5679    533    91%

============================== 875 passed, 42 skipped in 338.57s (0:05:38) ==============================
ERROR:root:Unrecoverable error
Traceback (most recent call last):
  File "/home/carlos/workspace/mu/venv/lib/python3.5/site-packages/_pytest/main.py", line 257, in wrap_session
    session.exitstatus = doit(config, session) or 0
  File "/home/carlos/workspace/mu/venv/lib/python3.5/site-packages/_pytest/main.py", line 313, in _main
    config.hook.pytest_runtestloop(session=session)
  File "/home/carlos/workspace/mu/venv/lib/python3.5/site-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/home/carlos/workspace/mu/venv/lib/python3.5/site-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/home/carlos/workspace/mu/venv/lib/python3.5/site-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
[... 250 lines cut ...]
  File "/home/carlos/workspace/mu/venv/lib/python3.5/site-packages/_pytest/_code/code.py", line 769, in repr_traceback_entry
    source = self._getentrysource(entry)
  File "/home/carlos/workspace/mu/venv/lib/python3.5/site-packages/_pytest/_code/code.py", line 677, in _getentrysource
    source = entry.getsource(self.astcache)
  File "/home/carlos/workspace/mu/venv/lib/python3.5/site-packages/_pytest/_code/code.py", line 230, in getsource
    self.lineno, source, astnode=astnode
  File "/home/carlos/workspace/mu/venv/lib/python3.5/site-packages/_pytest/_code/source.py", line 181, in getstatementrange_ast
    start, end = get_statement_startend2(lineno, astnode)
  File "/home/carlos/workspace/mu/venv/lib/python3.5/site-packages/_pytest/_code/source.py", line 149, in get_statement_startend2
    for x in ast.walk(node):
  File "/usr/lib/python3.5/ast.py", line 217, in walk
    todo.extend(iter_child_nodes(node))
  File "/usr/lib/python3.5/ast.py", line 182, in iter_child_nodes
    for name, field in iter_fields(node):
  File "/usr/lib/python3.5/ast.py", line 172, in iter_fields
    yield field, getattr(node, field)
KeyboardInterrupt
Makefile:56: recipe for target 'coverage' failed
make: *** [coverage] Error 1

```

And it opened the following crash URL in the default browser:

https://codewith.mu/crash/?p=TGludXggNC4xNS4wLTExMi1nZW5lcmljICMxMTN%2BMTYuMDQuMS1VYnVudHUgU01QIEZyaSBKdWwgMTAgMDQ6Mzc6MDggVVRDIDIwMjAgeDg2XzY0&l=en_US&v=1.1.0.beta.2&e=VHJhY2ViYWNrIChtb3N0IHJlY2VudCBjYWxsIGxhc3QpOgogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9uMy41L3NpdGUtcGFja2FnZXMvX3B5dGVzdC9tYWluLnB5IiwgbGluZSAyNTcsIGluIHdyYXBfc2Vzc2lvbgogICAgc2Vzc2lvbi5leGl0c3RhdHVzID0gZG9pdChjb25maWcsIHNlc3Npb24pIG9yIDAKICBGaWxlICIvaG9tZS9jYXJsb3Mvd29ya3NwYWNlL211L3ZlbnYvbGliL3B5dGhvbjMuNS9zaXRlLXBhY2thZ2VzL19weXRlc3QvbWFpbi5weSIsIGxpbmUgMzEzLCBpbiBfbWFpbgogICAgY29uZmlnLmhvb2sucHl0ZXN0X3J1bnRlc3Rsb29wKHNlc3Npb249c2Vzc2lvbikKICBGaWxlICIvaG9tZS9jYXJsb3Mvd29ya3NwYWNlL211L3ZlbnYvbGliL3B5dGhvbjMuNS9zaXRlLXBhY2thZ2VzL3BsdWdneS9ob29rcy5weSIsIGxpbmUgMjg2LCBpbiBfX2NhbGxfXwogICAgcmV0dXJuIHNlbGYuX2hvb2tleGVjKHNlbGYsIHNlbGYuZ2V0X2hvb2tpbXBscygpLCBrd2FyZ3MpCiAgRmlsZSAiL2hvbWUvY2FybG9zL3dvcmtzcGFjZS9tdS92ZW52L2xpYi9weXRob24zLjUvc2l0ZS1wYWNrYWdlcy9wbHVnZ3kvbWFuYWdlci5weSIsIGxpbmUgOTMsIGluIF9ob29rZXhlYwogICAgcmV0dXJuIHNlbGYuX2lubmVyX2hvb2tleGVjKGhvb2ssIG1ldGhvZHMsIGt3YXJncykKICBGaWxlICIvaG9tZS9jYXJsb3Mvd29ya3NwYWNlL211L3ZlbnYvbGliL3B5dGhvbjMuNS9zaXRlLXBhY2thZ2VzL3BsdWdneS9tYW5hZ2VyLnB5IiwgbGluZSA4NywgaW4gPGxhbWJkYT4KICAgIGZpcnN0cmVzdWx0PWhvb2suc3BlYy5vcHRzLmdldCgiZmlyc3RyZXN1bHQiKSBpZiBob29rLnNwZWMgZWxzZSBGYWxzZSwKICBGaWxlICIvaG9tZS9jYXJsb3Mvd29ya3NwYWNlL211L3ZlbnYvbGliL3B5dGhvbjMuNS9zaXRlLXBhY2thZ2VzL3BsdWdneS9jYWxsZXJzLnB5IiwgbGluZSAyMDMsIGluIF9tdWx0aWNhbGwKICAgIGdlbi5zZW5kKG91dGNvbWUpCiAgRmlsZSAiL2hvbWUvY2FybG9zL3dvcmtzcGFjZS9tdS92ZW52L2xpYi9weXRob24zLjUvc2l0ZS1wYWNrYWdlcy9wbHVnZ3kvY2FsbGVycy5weSIsIGxpbmUgODAsIGluIGdldF9yZXN1bHQKICAgIHJhaXNlIGV4WzFdLndpdGhfdHJhY2ViYWNrKGV4WzJdKQogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9uMy41L3NpdGUtcGFja2FnZXMvcGx1Z2d5L2NhbGxlcnMucHkiLCBsaW5lIDE4NywgaW4gX211bHRpY2FsbAogICAgcmVzID0gaG9va19pbXBsLmZ1bmN0aW9uKCphcmdzKQogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9uMy41L3NpdGUtcGFja2FnZXMvX3B5dGVzdC9tYWluLnB5IiwgbGluZSAzMzgsIGluIHB5dGVzdF9ydW50ZXN0bG9vcAogICAgaXRlbS5jb25maWcuaG9vay5weXRlc3RfcnVudGVzdF9wcm90b2NvbChpdGVtPWl0ZW0sIG5leHRpdGVtPW5leHRpdGVtKQogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9uMy41L3NpdGUtcGFja2FnZXMvcGx1Z2d5L2hvb2tzLnB5IiwgbGluZSAyODYsIGluIF9fY2FsbF9fCiAgICByZXR1cm4gc2VsZi5faG9va2V4ZWMoc2VsZiwgc2VsZi5nZXRfaG9va2ltcGxzKCksIGt3YXJncykKICBGaWxlICIvaG9tZS9jYXJsb3Mvd29ya3NwYWNlL211L3ZlbnYvbGliL3B5dGhvbjMuNS9zaXRlLXBhY2thZ2VzL3BsdWdneS9tYW5hZ2VyLnB5IiwgbGluZSA5MywgaW4gX2hvb2tleGVjCiAgICByZXR1cm4gc2VsZi5faW5uZXJfaG9va2V4ZWMoaG9vaywgbWV0aG9kcywga3dhcmdzKQogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9 [... cut ...] dF9zdGF0ZW1lbnRfc3RhcnRlbmQyCiAgICBmb3IgeCBpbiBhc3Qud2Fsayhub2RlKToKICBGaWxlICIvdXNyL2xpYi9weXRob24zLjUvYXN0LnB5IiwgbGluZSAyMTcsIGluIHdhbGsKICAgIHRvZG8uZXh0ZW5kKGl0ZXJfY2hpbGRfbm9kZXMobm9kZSkpCiAgRmlsZSAiL3Vzci9saWIvcHl0aG9uMy41L2FzdC5weSIsIGxpbmUgMTgyLCBpbiBpdGVyX2NoaWxkX25vZGVzCiAgICBmb3IgbmFtZSwgZmllbGQgaW4gaXRlcl9maWVsZHMobm9kZSk6CiAgRmlsZSAiL3Vzci9saWIvcHl0aG9uMy41L2FzdC5weSIsIGxpbmUgMTcyLCBpbiBpdGVyX2ZpZWxkcwogICAgeWllbGQgZmllbGQsIGdldGF0dHIobm9kZSwgZmllbGQpCktleWJvYXJkSW50ZXJydXB0Cg%3D%3D

carlosperate, 2021-03-23:
From trial an error this is the max UIR that works in this nginx configuration, slightly above 8KBs (8196 characters):

https://codewith.mu/crash/?p=TGludXggNC4xNS4wLTExMi1nZW5lcmljICMxMTN%2BMTYuMDQuMS1VYnVudHUgU01QIEZyaSBKdWwgMTAgMDQ6Mzc6MDggVVRDIDIwMjAgeDg2XzY0&l=en_US&v=1.1.0.beta.2&e=VHJhY2ViYWNrIChtb3N0IHJlY2VudCBjYWxsIGxhc3QpOgogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9uMy41L3NpdGUtcGFja2FnZXMvX3B5dGVzdC9tYWluLnB5IiwgbGluZSAyNTcsIGluIHdyYXBfc2Vzc2lvbgogICAgc2Vzc2lvbi5leGl0c3RhdHVzID0gZG9pdChjb25maWcsIHNlc3Npb24pIG9yIDAKICBGaWxlICIvaG9tZS9jYXJsb3Mvd29ya3NwYWNlL211L3ZlbnYvbGliL3B5dGhvbjMuNS9zaXRlLXBhY2thZ2VzL19weXRlc3QvbWFpbi5weSIsIGxpbmUgMzEzLCBpbiBfbWFpbgogICAgY29uZmlnLmhvb2sucHl0ZXN0X3J1bnRlc3Rsb29wKHNlc3Npb249c2Vzc2lvbikKICBGaWxlICIvaG9tZS9jYXJsb3Mvd29ya3NwYWNlL211L3ZlbnYvbGliL3B5dGhvbjMuNS9zaXRlLXBhY2thZ2VzL3BsdWdneS9ob29rcy5weSIsIGxpbmUgMjg2LCBpbiBfX2NhbGxfXwogICAgcmV0dXJuIHNlbGYuX2hvb2tleGVjKHNlbGYsIHNlbGYuZ2V0X2hvb2tpbXBscygpLCBrd2FyZ3MpCiAgRmlsZSAiL2hvbWUvY2FybG9zL3dvcmtzcGFjZS9tdS92ZW52L2xpYi9weXRob24zLjUvc2l0ZS1wYWNrYWdlcy9wbHVnZ3kvbWFuYWdlci5weSIsIGxpbmUgOTMsIGluIF9ob29rZXhlYwogICAgcmV0dXJuIHNlbGYuX2lubmVyX2hvb2tleGVjKGhvb2ssIG1ldGhvZHMsIGt3YXJncykKICBGaWxlICIvaG9tZS9jYXJsb3Mvd29ya3NwYWNlL211L3ZlbnYvbGliL3B5dGhvbjMuNS9zaXRlLXBhY2thZ2VzL3BsdWdneS9tYW5hZ2VyLnB5IiwgbGluZSA4NywgaW4gPGxhbWJkYT4KICAgIGZpcnN0cmVzdWx0PWhvb2suc3BlYy5vcHRzLmdldCgiZmlyc3RyZXN1bHQiKSBpZiBob29rLnNwZWMgZWxzZSBGYWxzZSwKICBGaWxlICIvaG9tZS9jYXJsb3Mvd29ya3NwYWNlL211L3ZlbnYvbGliL3B5dGhvbjMuNS9zaXRlLXBhY2thZ2VzL3BsdWdneS9jYWxsZXJzLnB5IiwgbGluZSAyMDMsIGluIF9tdWx0aWNhbGwKICAgIGdlbi5zZW5kKG91dGNvbWUpCiAgRmlsZSAiL2hvbWUvY2FybG9zL3dvcmtzcGFjZS9tdS92ZW52L2xpYi9weXRob24zLjUvc2l0ZS1wYWNrYWdlcy9wbHVnZ3kvY2FsbGVycy5weSIsIGxpbmUgODAsIGluIGdldF9yZXN1bHQKICAgIHJhaXNlIGV4WzFdLndpdGhfdHJhY2ViYWNrKGV4WzJdKQogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9uMy41L3NpdGUtcGFja2FnZXMvcGx1Z2d5L2NhbGxlcnMucHkiLCBsaW5lIDE4NywgaW4gX211bHRpY2FsbAogICAgcmVzID0gaG9va19pbXBsLmZ1bmN0aW9uKCphcmdzKQogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9uMy41L3NpdGUtcGFja2FnZXMvX3B5dGVzdC9tYWluLnB5IiwgbGluZSAzMzgsIGluIHB5dGVzdF9ydW50ZXN0bG9vcAogICAgaXRlbS5jb25maWcuaG9vay5weXRlc3RfcnVudGVzdF9wcm90b2NvbChpdGVtPWl0ZW0sIG5leHRpdGVtPW5leHRpdGVtKQogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9uMy41L3NpdGUtcGFja2FnZXMvcGx1Z2d5L2hvb2tzLnB5IiwgbGluZSAyODYsIGluIF9fY2FsbF9fCiAgICByZXR1cm4gc2VsZi5faG9va2V4ZWMoc2VsZiwgc2VsZi5nZXRfaG9va2ltcGxzKCksIGt3YXJncykKICBGaWxlICIvaG9tZS9jYXJsb3Mvd29ya3NwYWNlL211L3ZlbnYvbGliL3B5dGhvbjMuNS9zaXRlLXBhY2thZ2VzL3BsdWdneS9tYW5hZ2VyLnB5IiwgbGluZSA5MywgaW4gX2hvb2tleGVjCiAgICByZXR1cm4gc2VsZi5faW5uZXJfaG9va2V4ZWMoaG9vaywgbWV0aG9kcywga3dhcmdzKQogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9uMy41L3NpdGUtcGFja2FnZXMvcGx1Z2d5L21hbmFnZXIucHkiLCBsaW5lIDg3LCBpbiA8bGFtYmRhPgogICAgZmlyc3RyZXN1bHQ9aG9vay5zcGVjLm9wdHMuZ2V0KCJmaXJzdHJlc3VsdCIpIGlmIGhvb2suc3BlYyBlbHNlIEZhbHNlLAogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9uMy41L3NpdGUtcGFja2FnZXMvcGx1Z2d5L2NhbGxlcnMucHkiLCBsaW5lIDIwOCwgaW4gX211bHRpY2FsbAogICAgcmV0dXJuIG91dGNvbWUuZ2V0X3Jlc3VsdCgpCiAgRmlsZSAiL2hvbWUvY2FybG9zL3dvcmtzcGFjZS9tdS92ZW52L2xpYi9weXRob24zLjUvc2l0ZS1wYWNrYWdlcy9wbHVnZ3kvY2FsbGVycy5weSIsIGxpbmUgODAsIGluIGdldF9yZXN1bHQKICAgIHJhaXNlIGV4WzFdLndpdGhfdHJhY2ViYWNrKGV4WzJdKQogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9uMy41L3NpdGUtcGFja2FnZXMvcGx1Z2d5L2NhbGxlcnMucHkiLCBsaW5lIDE4NywgaW4gX211bHRpY2FsbAogICAgcmVzID0gaG9va19pbXBsLmZ1bmN0aW9uKCphcmdzKQogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9uMy41L3NpdGUtcGFja2FnZXMvX3B5dGVzdC9ydW5uZXIucHkiLCBsaW5lIDExMCwgaW4gcHl0ZXN0X3J1bnRlc3RfcHJvdG9jb2wKICAgIHJ1bnRlc3Rwcm90b2NvbChpdGVtLCBuZXh0aXRlbT1uZXh0aXRlbSkKICBGaWxlICIvaG9tZS9jYXJsb3Mvd29ya3NwYWNlL211L3ZlbnYvbGliL3B5dGhvbjMuNS9zaXRlLXBhY2thZ2VzL19weXRlc3QvcnVubmVyLnB5IiwgbGluZSAxMjgsIGluIHJ1bnRlc3Rwcm90b2NvbAogICAgcmVwb3J0cy5hcHBlbmQoY2FsbF9hbmRfcmVwb3J0KGl0ZW0sICJ0ZWFyZG93biIsIGxvZywgbmV4dGl0ZW09bmV4dGl0ZW0pKQogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9uMy41L3NpdGUtcGFja2FnZXMvX3B5dGVzdC9ydW5uZXIucHkiLCBsaW5lIDIxNiwgaW4gY2FsbF9hbmRfcmVwb3J0CiAgICBjYWxsID0gY2FsbF9ydW50ZXN0X2hvb2soaXRlbSwgd2hlbiwgKiprd2RzKQogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9uMy41L3NpdGUtcGFja2FnZXMvX3B5dGVzdC9ydW5uZXIucHkiLCBsaW5lIDI1NiwgaW4gY2FsbF9ydW50ZXN0X2hvb2sKICAgIGxhbWJkYTogaWhvb2soaXRlbT1pdGVtLCAqKmt3ZHMpLCB3aGVuPXdoZW4sIHJlcmFpc2U9cmVyYWlzZQogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdmVudi9saWIvcHl0aG9uMy41L3NpdGUtcGFja2FnZXMvX3B5dGVzdC9ydW5uZXIucHkiLCBsaW5lIDMxMCwgaW4gZnJvbV9jYWxsCiAgICByZXN1bHQgPSBmdW5jKCkgICMgdHlwZTogT3B0aW9uYWxbVFJlc3VsdF0KICBGaWxlICIvaG9tZS9jYXJsb3Mvd29ya3NwYWNlL211L3ZlbnYvbGliL3B5dGhvbjMuNS9zaXRlLXBhY2thZ2VzL19weXRlc3QvcnVubmVyLnB5IiwgbGluZSAyNTYsIGluIDxsYW1iZGE%2BCiAgICBsYW1iZGE6IGlob29rKGl0ZW09aXRlbSwgKiprd2RzKSwgd2hlbj13aGVuLCByZXJhaXNlPXJlcmFpc2UKICBGaWxlICIvaG9tZS9jYXJsb3Mvd29ya3NwYWNlL211L3ZlbnYvbGliL3B5dGhvbjMuNS9zaXRlLXBhY2thZ2VzL3BsdWdneS9ob29rcy5weSIsIGxpbmUgMjg2LCBpbiBfX2NhbGxfXwogICAgcmV0dXJuIHNlbGYuX2hvb2tleGVjKHNlbGYsIHNlbGYuZ2V0X2hvb2tpbXBscygpLCBrd2FyZ3MpCiAgRmlsZSAiL2hvbWUvY2FybG9zL3dvcmtzcGFjZS9tdS92ZW52L2xpYi9weXRob24zLjUvc2l0ZS1wYWNrYWdlcy9wbHVnZ3kvbWFuYWdlci5weSIsIGxpbmUgOTMsIGluIF9ob29rZXhlYwogICAgcmV0dXJuIHNlbGYuX2lubmVyX2hvb2tleGVjKGhvb2ssIG1ldGhvZHMsIGt3YXJncykKICBGaWxlICIvaG9tZS9jYXJsb3Mvd29ya3NwYWNlL211L3ZlbnYvbGliL3B5dGhvbjMuNS9zaXRlLXBhY2thZ2VzL3BsdWdneS9tYW5hZ2VyLnB5Iiwgb [... cut ...] 2FybG9zL3dvcmtzcGFjZS9tdS92ZW52L2xpYi9weXRob24zLjUvc2l0ZS1wYWNrYWdlcy9fcHl0ZXN0L2ZpeHR1cmVzLnB5IiwgbGluZSA5MzksIGluIF90ZWFyZG93bl95aWVsZF9maXh0dXJlCiAgICBuZXh0KGl0KQogIEZpbGUgIi9ob21lL2Nhcmxvcy93b3Jrc3BhY2UvbXUvdGVzdHMvdmlydHVhbF9lbnZpcm9ubWVudC90ZXN0X3ZpcnR1YWxfZW52aXJvbm1lbnQucHkiLCBsaW5lIDc4LCBpbiB2ZW52CiAgICB3aGlsZSBsb2dnZXIuaGFzSGFuZGxlcnMoKSBhbmQgbG9nZ2VyLmhhbmRsZXJzOgpLZXlib2FyZEludGVycnVwdAoKRHVyaW5nIGhhbmRsaW5nIG9mIHRoZSBhYm92ZSBleGNlcHRpb24sIGFub3RoZXIgZXhjZXB0aW9uIG9jY3VycmVkOgoKVHJhY

carlosperate, 2021-04-16:
@ntoll something you might want to look at as well, we might need to cut down how much we add to the URI.

ntoll, 2021-07-10:
Ah yes... I wondered if/when we'd see this problem when very long stack traces were produced.

I think the best solution would be to ensure the **end** of the stack trace is appended (where the end is up to some size that is not too long for both our crash report page, and the subsequent linked-to bug report page on GitHub).

ntoll, 2021-07-10:
OK... looking into this I notice something in the generation of the error report that should mitigate this problem: https://github.com/mu-editor/mu/blob/master/mu/app.py#L179 introduced in #1625.

closed 2021-07-10 by ntoll