mu-editor/mu #1407: Crash Report

DCarri7962, 2021-03-25:
Version: 1.1.0.beta.2
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\DCarr\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\python3.py", line 109, in start_kernel
    self.repl_kernel_manager.start_kernel()
  File "C:\Users\DCarr\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\python3.py", line 42, in start_kernel
    kernel_cmd, kw = self.pre_start_kernel(**kw)
  File "C:\Users\DCarr\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\jupyter_client\manager.py", line 264, in pre_start_kernel
    kernel_cmd = self.format_kernel_cmd(extra_arguments=extra_arguments)
  File "C:\Users\DCarr\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\jupyter_client\manager.py", line 183, in format_kernel_cmd
    cmd = self.kernel_spec.argv + extra_arguments
  File "C:\Users\DCarr\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\jupyter_client\manager.py", line 89, in kernel_spec
    self._kernel_spec = self.kernel_spec_manager.get_kernel_spec(self.kernel_name)
  File "C:\Users\DCarr\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\jupyter_client\kernelspec.py", line 235, in get_kernel_spec
    raise NoSuchKernel(kernel_name)
jupyter_client.kernelspec.NoSuchKernel: No such kernel named mu_venv-38-20210325-131602

```

tjguk, 2021-03-25:
Thanks for the report @DCarri7962 -- that's the first time I've seen a failure to create the Jupyter Kernel. 

Would you be able to upload the full logs, please, to this issue? They should be in:

`%LOCALAPPDATA%\python\mu\Logs`

DCarri7962, 2021-03-25:
[mu.log](https://github.com/mu-editor/mu/files/6207968/mu.log)

carlosperate, 2021-08-11:
Hi @DCarri7962,

Would you be able to check if beta.5 works for you and let us know the result? The latest betas have reworked some of the Jupyter stuff:
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

carlosperate, 2021-09-21:
@DCarri7962 would you be able to test this again with beta.5? We think that might have resolved this issue: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

DCarri7962, 2021-09-23:
I just got my PC back from the shop. I will see if I can get working this weekend

carlosperate, 2021-12-17:
Hi @DCarri7962, were you able to try this again?

DCarri7962, 2022-03-23:
@carlosperate Thanks for you patience. PC crashed and had to replace the system. I have tested the python code again and all seems to be working well. I am using Python 3.10 and Mu 1.1.1 releases. If I see anything I will let you know!

DCarri7962, 2022-03-23:
[mu.log](https://github.com/mu-editor/mu/files/8333726/mu.log)

closed 2022-03-23 by DCarri7962