mu-editor/mu #1408: Crash Report

PascaleStE, 2021-03-25:
Hello, I do not know how to overcome this difficulty ... and need help. Thanks in advance !

Version: 1.1.0.beta.2
Locale: fr_FR
Platform: Linux 5.4.0-67-generic #75~18.04.1-Ubuntu SMP Tue Feb 23 19:17:50 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/cecile/.local/lib/python3.6/site-packages/mu/app.py", line 162, in run
    raise ex
  File "/home/cecile/.local/lib/python3.6/site-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/cecile/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 460, in ensure_and_create
    "Unable to create a working virtual environment."
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.

```

abixadamj, 2021-03-25:
I can confirm:

```

Traceback (most recent call last):
  File "/home/adasiek/mu/mu/virtual_environment.py", line 442, in ensure_and_create
    self.ensure()
  File "/home/adasiek/mu/mu/virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "/home/adasiek/mu/mu/virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: /home/adasiek/.local/share/mu/mu_venv-38-20210325-213304 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/adasiek/mu/mu/app.py", line 162, in run
    raise ex
  File "/home/adasiek/mu/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/adasiek/mu/mu/virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "/home/adasiek/mu/mu/virtual_environment.py", line 583, in create
    subprocess.run(
  File "/usr/lib/python3.8/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'virtualenv', '-p', '/usr/bin/python3', '-q', '/home/adasiek/.local/share/mu/mu_venv-38-20210325-213305']' returned non-zero exit status 1.
```

This is because of Debian - see: https://github.com/mu-editor/mu/issues/822#issuecomment-488058921
@ntoll - I think this can be a problem... because there are a lot of Debian/Ubuntu/Mint users...

abixadamj, 2021-03-25:
OK, I see that is something else. I have tried to run command and:

```
adasiek@free-desktop:~/mu$ python3 -m virtualenv -p /usr/bin/python3 -q /home/adasiek/.local/share/mu/mu_venv-38-20210325-213305
/usr/bin/python3: No module named virtualenv

```
So I install it:

```
adasiek@free-desktop:~/mu$ pip3 install virtualenv
Collecting virtualenv
  Downloading virtualenv-20.4.3-py2.py3-none-any.whl (7.2 MB)
     |████████████████████████████████| 7.2 MB 2.0 MB/s 
Requirement already satisfied: six<2,>=1.9.0 in /usr/lib/python3/dist-packages (from virtualenv) (1.14.0)
Collecting filelock<4,>=3.0.0
  Downloading filelock-3.0.12-py3-none-any.whl (7.6 kB)
Collecting distlib<1,>=0.3.1
  Downloading distlib-0.3.1-py2.py3-none-any.whl (335 kB)
     |████████████████████████████████| 335 kB 43.3 MB/s 
Requirement already satisfied: appdirs<2,>=1.4.3 in /usr/local/lib/python3.8/dist-packages (from virtualenv) (1.4.4)
Installing collected packages: filelock, distlib, virtualenv
Successfully installed distlib-0.3.1 filelock-3.0.12 virtualenv-20.4.3

```
But, after that there is another error:
```

Traceback (most recent call last):
  File "/home/adasiek/mu/mu/app.py", line 162, in run
    raise ex
  File "/home/adasiek/mu/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/adasiek/mu/mu/virtual_environment.py", line 459, in ensure_and_create
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.
```
And I cannot find out why...

tjguk, 2021-03-25:
@abixadamj I'm a little confused, because our `setup.py` includes a dependency on `virtualenv` so if you've `piip install`-ed I would expect `virtualenv` to be there already. If you're running from a git clone (or an unpacked sdist) then a `pip install -e .` should also have worked.

BTW I'm not sure that your error is the same as @PascaleStE  error -- they're both failures to create a venv but that can come from several directions.

tjguk, 2021-03-25:
@PascaleStE  -- are you in a position to attach the log file to this issue, please?

It should be in `~/.cache/mu/log/mu.log`

tjguk, 2021-03-25:
Also, thanks both of you for taking the trouble to report the crashes. It's great when a beta release actually gets useful feedback from users. We're looking to release the next beta after this weekend; hopefully we can spot what the issue is in your cases.

abixadamj, 2021-03-25:
@tjguk I did not run `pip install -e .` ;-( Shame on me.... I correct it and install all requirements.

That is FULL log:

```
2021-03-25 22:00:09,607 - root:264(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.2
2021-03-25 22:00:09,607 - root:265(run) INFO: uname_result(system='Linux', node='free-desktop', release='5.4.0-65-generic', version='#73-Ubuntu SMP Mon Jan 18 17:25:17 UTC 2021', machine='x86_64', processor='x86_64')
2021-03-25 22:00:09,607 - root:266(run) INFO: Python path: ['/home/adasiek/mu', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/adasiek/.local/lib/python3.8/site-packages', '/home/adasiek/mu', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3/dist-packages/IPython/extensions']
2021-03-25 22:00:09,607 - root:267(run) INFO: Language code: pl_PL
2021-03-25 22:00:09,607 - mu.settings:220(load) WARNING: No settings file found at /home/adasiek/.local/share/mu/settings.json; skipping
2021-03-25 22:00:09,607 - mu.settings:220(load) WARNING: No settings file found at /home/adasiek/.local/share/mu/session.json; skipping
2021-03-25 22:00:09,712 - mu.virtual_environment:435(ensure_and_create) INFO: Added log handler.
2021-03-25 22:00:09,713 - mu.virtual_environment:439(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-03-25 22:00:09,713 - mu.virtual_environment:479(ensure_path) ERROR: /home/adasiek/.local/share/mu/mu_venv-38-20210325-220009 does not exist.
2021-03-25 22:00:09,713 - mu.virtual_environment:445(ensure_and_create) DEBUG: Creating new virtual environment at /home/adasiek/.local/share/mu/mu_venv-38-20210325-220009.
2021-03-25 22:00:09,713 - mu.virtual_environment:379(relocate) DEBUG: Virtual environment set up mu_venv-38-20210325-220009 at /home/adasiek/.local/share/mu/mu_venv-38-20210325-220009
2021-03-25 22:00:09,713 - mu.virtual_environment:579(create) INFO: Creating virtualenv: /home/adasiek/.local/share/mu/mu_venv-38-20210325-220009
2021-03-25 22:00:09,713 - mu.virtual_environment:580(create) INFO: Virtualenv name: mu_venv-38-20210325-220009
2021-03-25 22:00:09,953 - mu.virtual_environment:632(install_baseline_packages) INFO: Installing baseline packages.
2021-03-25 22:00:09,954 - mu.virtual_environment:633(install_baseline_packages) INFO: /home/adasiek/mu/mu/wheels exists
2021-03-25 22:00:17,300 - mu.virtual_environment:658(install_baseline_packages) DEBUG: Processing ./mu/wheels/itsdangerous-1.1.0-py2.py3-none-any.whl
Processing ./mu/wheels/pyzmq-22.0.3-cp38-cp38-manylinux2010_x86_64.whl
Processing ./mu/wheels/ptyprocess-0.7.0-py2.py3-none-any.whl
Processing ./mu/wheels/Jinja2-2.11.3-py2.py3-none-any.whl
Processing ./mu/wheels/decorator-4.4.2-py2.py3-none-any.whl
Processing ./mu/wheels/pyserial-3.5-py2.py3-none-any.whl
Processing ./mu/wheels/backcall-0.2.0-py2.py3-none-any.whl
Processing ./mu/wheels/reedsolo-1.5.4-cp38-cp38-linux_x86_64.whl
Processing ./mu/wheels/setuptools-54.2.0-py3-none-any.whl
Processing ./mu/wheels/python_dateutil-2.8.1-py2.py3-none-any.whl
Processing ./mu/wheels/ipykernel-5.5.0-py3-none-any.whl
Processing ./mu/wheels/six-1.15.0-py2.py3-none-any.whl
Processing ./mu/wheels/jedi-0.18.0-py2.py3-none-any.whl
Processing ./mu/wheels/pygame-2.0.1-cp38-cp38-manylinux1_x86_64.whl
Processing ./mu/wheels/tornado-6.1-cp38-cp38-manylinux2010_x86_64.whl
Processing ./mu/wheels/Pygments-2.8.1-py3-none-any.whl
Processing ./mu/wheels/cryptography-3.4.7-cp36-abi3-manylinux2014_x86_64.whl
Processing ./mu/wheels/jupyter_core-4.7.1-py3-none-any.whl
Processing ./mu/wheels/Flask-1.1.2-py2.py3-none-any.whl
Processing ./mu/wheels/jupyter_client-6.1.12-py3-none-any.whl
Processing ./mu/wheels/pexpect-4.8.0-py2.py3-none-any.whl
Processing ./mu/wheels/pycparser-2.20-py2.py3-none-any.whl
Processing ./mu/wheels/QtPy-1.9.0-py2.py3-none-any.whl
Processing ./mu/wheels/ipython-7.21.0-py3-none-any.whl
Processing ./mu/wheels/pgzero-1.2.1-py3-none-any.whl
Processing ./mu/wheels/MarkupSafe-1.1.1-cp38-cp38-manylinux2010_x86_64.whl
Processing ./mu/wheels/prompt_toolkit-3.0.18-py3-none-any.whl
Processing ./mu/wheels/numpy-1.20.1-cp38-cp38-manylinux2010_x86_64.whl
Processing ./mu/wheels/ecdsa-0.16.1-py2.py3-none-any.whl
Processing ./mu/wheels/qtconsole-4.7.4-py2.py3-none-any.whl
Processing ./mu/wheels/esptool-3.0-py3-none-any.whl
Processing ./mu/wheels/wcwidth-0.2.5-py2.py3-none-any.whl
Processing ./mu/wheels/cffi-1.14.5-cp38-cp38-manylinux1_x86_64.whl
Processing ./mu/wheels/bitstring-3.1.7-py3-none-any.whl
Processing ./mu/wheels/ipython_genutils-0.2.0-py2.py3-none-any.whl
Processing ./mu/wheels/parso-0.8.1-py2.py3-none-any.whl
Processing ./mu/wheels/nudatus-0.0.5-py3-none-any.whl
Processing ./mu/wheels/pickleshare-0.7.5-py2.py3-none-any.whl
Processing ./mu/wheels/click-7.1.2-py2.py3-none-any.whl
Processing ./mu/wheels/traitlets-5.0.5-py3-none-any.whl
Processing ./mu/wheels/Werkzeug-1.0.1-py2.py3-none-any.whl
Installing collected packages: ipython-genutils, wcwidth, traitlets, six, ptyprocess, parso, tornado, setuptools, pyzmq, python-dateutil, Pygments, pycparser, prompt-toolkit, pickleshare, pexpect, jupyter-core, jedi, decorator, backcall, MarkupSafe, jupyter-client, ipython, cffi, Werkzeug, reedsolo, QtPy, pyserial, pygame, numpy, Jinja2, itsdangerous, ipykernel, ecdsa, cryptography, click, bitstring, qtconsole, pgzero, nudatus, Flask, esptool
  Attempting uninstall: setuptools
    Found existing installation: setuptools 54.1.2
    Uninstalling setuptools-54.1.2:
[... cut ...]
  File "/home/adasiek/mu/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/adasiek/mu/mu/virtual_environment.py", line 459, in ensure_and_create
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.
2021-03-25 22:00:41,766 - mu.settings:169(save) DEBUG: Saving to /home/adasiek/.local/share/mu/session.json
2021-03-25 22:00:41,766 - mu.settings:169(save) DEBUG: Saving to /home/adasiek/.local/share/mu/venv.json

```

abixadamj, 2021-03-25:
I have made tree of this venv - the whole file is at my Pcloud: http://e.pc.cd/HcjotalK

PascaleStE, 2021-03-26:
Here is "my" log file.

2021-03-25 21:03:02,830 - root:259(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.2
2021-03-25 21:03:02,830 - root:260(run) INFO: uname_result(system='Linux', node='cecile-N150ZU', release='5.4.0-67-generic', version='#75~18.04.1-Ubuntu SMP Tue Feb 23 19:17:50 UTC 2021', machine='x86_64', processor='x86_64')
2021-03-25 21:03:02,830 - root:261(run) INFO: Python path: ['/home/cecile/.local/bin', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/home/cecile/.local/lib/python3.6/site-packages', '/usr/local/lib/python3.6/dist-packages', '/usr/lib/python3/dist-packages', '/home/cecile/.local/lib/python3.6/site-packages/IPython/extensions']
2021-03-25 21:03:02,830 - root:262(run) INFO: Language code: fr_FR
2021-03-25 21:03:02,830 - mu.settings:220(load) WARNING: No settings file found at /home/cecile/.local/share/mu/settings.json; skipping
2021-03-25 21:03:02,831 - mu.settings:220(load) WARNING: No settings file found at /home/cecile/.local/share/mu/session.json; skipping
2021-03-25 21:03:02,912 - mu.virtual_environment:434(ensure_and_create) INFO: Added log handler.
2021-03-25 21:03:02,912 - mu.virtual_environment:439(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-03-25 21:03:02,912 - mu.virtual_environment:479(ensure_path) ERROR: /home/cecile/.local/share/mu/mu_venv-36-20210325-210302 does not exist.
2021-03-25 21:03:02,913 - mu.virtual_environment:443(ensure_and_create) DEBUG: Virtual environment not present or correct.
2021-03-25 21:03:02,913 - mu.virtual_environment:446(ensure_and_create) DEBUG: Creating new virtual environment at /home/cecile/.local/share/mu/mu_venv-36-20210325-210302.
2021-03-25 21:03:02,913 - mu.virtual_environment:378(relocate) DEBUG: Virtual environment set up mu_venv-36-20210325-210302 at /home/cecile/.local/share/mu/mu_venv-36-20210325-210302
2021-03-25 21:03:02,913 - mu.virtual_environment:578(create) INFO: Creating virtualenv: /home/cecile/.local/share/mu/mu_venv-36-20210325-210302
2021-03-25 21:03:02,913 - mu.virtual_environment:579(create) INFO: Virtualenv name: mu_venv-36-20210325-210302
2021-03-25 21:03:03,391 - mu.virtual_environment:631(install_baseline_packages) INFO: Installing baseline packages.
2021-03-25 21:03:03,393 - mu.virtual_environment:635(install_baseline_packages) INFO: /home/cecile/.local/lib/python3.6/site-packages/mu/wheels exists
2021-03-25 21:03:03,395 - mu.virtual_environment:647(install_baseline_packages) WARNING: No wheels found in /home/cecile/.local/lib/python3.6/site-packages/mu/wheels; downloading...
2021-03-25 21:03:03,397 - mu.wheels:58(download) DEBUG: dirpath: /home/cecile/.local/lib/python3.6/site-packages/mu/wheels
2021-03-25 21:03:03,397 - mu.wheels:74(download) DEBUG: mode_packages: [('pgzero', 'git+https://github.com/mu-editor/pgzero.git@mu-wheel'), ('flask', 'flask==1.1.2'), ('serial', 'pyserial>=3.0'), ('qtconsole', 'qtconsole==4.7.4'), ('nudatus', 'nudatus>=0.0.3'), ('esptool', 'esptool==3.*')]
2021-03-25 21:03:03,398 - mu.wheels:80(download) INFO: Running pip download for pgzero / git+https://github.com/mu-editor/pgzero.git@mu-wheel / []
2021-03-25 21:03:42,380 - mu.wheels:97(download) DEBUG: Collecting git+https://github.com/mu-editor/pgzero.git@mu-wheel
  Cloning https://github.com/mu-editor/pgzero.git (to mu-wheel) to /tmp/pip-ulearcli-build
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/pgzero-1.2.zip
Collecting pygame>=1.9.2 (from pgzero==1.2)
  Downloading https://files.pythonhosted.org/packages/01/da/4ff439558641a26dd29b04c25947e6c0ace041f56b2aa2ef1134edab06b8/pygame-2.0.1-cp36-cp36m-manylinux1_x86_64.whl (11.8MB)
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/pygame-2.0.1-cp36-cp36m-manylinux1_x86_64.whl
Collecting numpy (from pgzero==1.2)
  Downloading https://files.pythonhosted.org/packages/45/b2/6c7545bb7a38754d63048c7696804a0d947328125d81bf12beaa692c3ae3/numpy-1.19.5-cp36-cp36m-manylinux1_x86_64.whl (13.4MB)
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/numpy-1.19.5-cp36-cp36m-manylinux1_x86_64.whl
Successfully downloaded pgzero pygame numpy

2021-03-25 21:03:42,380 - mu.wheels:80(download) INFO: Running pip download for flask / flask==1.1.2 / []
2021-03-25 21:03:45,989 - mu.wheels:97(download) DEBUG: Collecting flask==1.1.2
  Downloading https://files.pythonhosted.org/packages/f2/28/2a03252dfb9ebf377f40fba6a7841b47083260bf8bd8e737b0c6952df83f/Flask-1.1.2-py2.py3-none-any.whl (94kB)
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/Flask-1.1.2-py2.py3-none-any.whl
Collecting itsdangerous>=0.24 (from flask==1.1.2)
  Downloading https://files.pythonhosted.org/packages/76/ae/44b03b253d6fade317f32c24d100b3b35c2239807046a4c953c7b89fa49e/itsdangerous-1.1.0-py2.py3-none-any.whl
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/itsdangerous-1.1.0-py2.py3-none-any.whl
Collecting Werkzeug>=0.15 (from flask==1.1.2)
  Downloading https://files.pythonhosted.org/packages/cc/94/5f7079a0e00bd6863ef8f1da638721e9da21e5bacee597595b318f71d62e/Werkzeug-1.0.1-py2.py3-none-any.whl (298kB)
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/Werkzeug-1.0.1-py2.py3-none-any.whl
Collecting Jinja2>=2.10.1 (from flask==1.1.2)
  Downloading https://files.pythonhosted.org/packages/7e/c2/1eece8c95ddbc9b1aeb64f5783a9e07a286de42191b7204d67b7496ddf35/Jinja2-2.11.3-py2.py3-none-any.whl (125kB)
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/Jinja2-2.11.3-py2.py3-none-any.whl
Collecting click>=5.1 (from flask==1.1.2)
[... cut ...]
    venv.ensure_and_create(self.display_text)
  File "/home/cecile/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 460, in ensure_and_create
    "Unable to create a working virtual environment."
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.
2021-03-25 21:50:57,809 - mu.settings:169(save) DEBUG: Saving to /home/cecile/.local/share/mu/session.json
2021-03-25 21:50:57,810 - mu.settings:169(save) DEBUG: Saving to /home/cecile/.local/share/mu/venv.json

carlosperate, 2021-03-29:
@abixadamj if you are installing Mu from source, can you pull the latest from upstream and trying again?
https://github.com/mu-editor/mu/pull/1415 might have fixed the issue you are having.

abixadamj, 2021-03-30:
@carlosperate 

I do:
```

dasiek@free-desktop:~/mu$ git pull
remote: Enumerating objects: 25, done.
remote: Counting objects: 100% (25/25), done.
remote: Compressing objects: 100% (12/12), done.
remote: Total 25 (delta 15), reused 21 (delta 13), pack-reused 0
Rozpakowywanie obiektów: 100% (25/25), 18.91 KiB | 2.10 MiB/s, gotowe.
Z https://github.com/mu-editor/mu
   daec504..f56bf64  master       -> origin/master
 * [nowy tag]        1.1.0-beta.3 -> 1.1.0-beta.3
Aktualizowanie daec504..f56bf64
Fast-forward
 CHANGES.rst                                 | 37 +++++++++++++++++++++++++++++++++++++
 docs/design.rst                             |  6 +++---
 docs/design/runtime_virtual_environment.rst | 13 +++++--------
 mu/__init__.py                              |  2 +-
 mu/app.py                                   |  2 ++
 mu/virtual_environment.py                   |  3 +--
 6 files changed, 49 insertions(+), 14 deletions(-)
```

It Works ;-)
[
[image]
](url)

You can Close.

carlosperate, 2021-03-30:
That's great @abixadamj, thanks for testing!

@PascaleStE can you check as well if beta.3 works for you?

PascaleStE, 2021-04-06:
Still not working...
Above the log file
2021-04-06 15:27:11,661 - root:259(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.2
2021-04-06 15:27:11,661 - root:260(run) INFO: uname_result(system='Linux', node='cecile-N150ZU', release='5.4.0-70-generic', version='#78~18.04.1-Ubuntu SMP Sat Mar 20 14:10:07 UTC 2021', machine='x86_64', processor='x86_64')
2021-04-06 15:27:11,661 - root:261(run) INFO: Python path: ['/home/cecile/.local/bin', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/home/cecile/.local/lib/python3.6/site-packages', '/usr/local/lib/python3.6/dist-packages', '/usr/lib/python3/dist-packages', '/home/cecile/.local/lib/python3.6/site-packages/IPython/extensions']
2021-04-06 15:27:11,661 - root:262(run) INFO: Language code: fr_FR
2021-04-06 15:27:11,661 - mu.settings:220(load) WARNING: No settings file found at /home/cecile/.local/share/mu/settings.json; skipping
2021-04-06 15:27:11,785 - mu.virtual_environment:434(ensure_and_create) INFO: Added log handler.
2021-04-06 15:27:11,785 - mu.virtual_environment:439(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-04-06 15:27:11,786 - mu.virtual_environment:489(ensure_path) INFO: Virtual Environment found at: /home/cecile/.local/share/mu/mu_venv-36-20210330-210610
2021-04-06 15:27:11,786 - mu.virtual_environment:502(ensure_interpreter) INFO: Interpreter found at: /home/cecile/.local/share/mu/mu_venv-36-20210330-210610/bin/python
2021-04-06 15:27:11,811 - mu.virtual_environment:534(ensure_interpreter_version) INFO: Both interpreters at version 36
2021-04-06 15:27:11,811 - mu.virtual_environment:566(ensure_pip) INFO: Pip found at: /home/cecile/.local/share/mu/mu_venv-36-20210330-210610/bin/pip
2021-04-06 15:27:11,812 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-04-06 15:27:11,831 - mu.virtual_environment:558(ensure_key_modules) ERROR: Failed to import: pgzero
2021-04-06 15:27:11,833 - mu.virtual_environment:443(ensure_and_create) DEBUG: Virtual environment not present or correct.
2021-04-06 15:27:11,833 - mu.virtual_environment:446(ensure_and_create) DEBUG: Creating new virtual environment at /home/cecile/.local/share/mu/mu_venv-36-20210406-152711.
2021-04-06 15:27:11,833 - mu.virtual_environment:378(relocate) DEBUG: Virtual environment set up mu_venv-36-20210406-152711 at /home/cecile/.local/share/mu/mu_venv-36-20210406-152711
2021-04-06 15:27:11,833 - mu.virtual_environment:578(create) INFO: Creating virtualenv: /home/cecile/.local/share/mu/mu_venv-36-20210406-152711
2021-04-06 15:27:11,833 - mu.virtual_environment:579(create) INFO: Virtualenv name: mu_venv-36-20210406-152711
2021-04-06 15:27:12,261 - mu.virtual_environment:631(install_baseline_packages) INFO: Installing baseline packages.
2021-04-06 15:27:12,261 - mu.virtual_environment:635(install_baseline_packages) INFO: /home/cecile/.local/lib/python3.6/site-packages/mu/wheels exists
2021-04-06 15:27:12,264 - mu.virtual_environment:229(install) DEBUG: About to pip install: ['/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/qtconsole-4.7.4-py2.py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/python_dateutil-2.8.1-py2.py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/itsdangerous-1.1.0-py2.py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/decorator-4.4.2-py2.py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/ptyprocess-0.7.0-py2.py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/jedi-0.18.0-py2.py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/pyzmq-22.0.3-cp36-cp36m-manylinux1_x86_64.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/Jinja2-2.11.3-py2.py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/ipython-7.16.1-py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/traitlets-4.3.3-py2.py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/jupyter_core-4.7.1-py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/nudatus-0.0.5-py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/prompt_toolkit-3.0.18-py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/setuptools-54.2.0-py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/pygame-2.0.1-cp36-cp36m-manylinux1_x86_64.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/MarkupSafe-1.1.1-cp36-cp36m-manylinux1_x86_64.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/pickleshare-0.7.5-py2.py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/wcwidth-0.2.5-py2.py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/backcall-0.2.0-py2.py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/QtPy-1.9.0-py2.py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/click-7.1.2-py2.py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/ipykernel-5.5.0-py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/pexpect-4.8.0-py2.py3-none-any.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/numpy-1.19.5-cp36-cp36m-manylinux1_x86_64.whl', '/home/cecile/.local/lib/python3.6/site-packages/mu/wheels/Werkzeug-1.0.1-py2.py3-none-an [... cut ...]
    venv.ensure_and_create(self.display_text)
  File "/home/cecile/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 460, in ensure_and_create
    "Unable to create a working virtual environment."
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.
2021-04-06 15:34:30,028 - mu.settings:169(save) DEBUG: Saving to /home/cecile/.local/share/mu/session.json
2021-04-06 15:34:30,029 - mu.settings:169(save) DEBUG: Saving to /home/cecile/.local/share/mu/venv.json

carlosperate, 2021-04-07:
Thanks for the checking again @PascaleStE.
Could you check what version of pip you are running?
Mu needs pip v19.0 or higher to install some of the dependencies.

If after updating pip that still doesn't work, can you try uninstalling the mu-editor package, and ensure the `/home/cecile/.local/lib/python3.6/site-packages/mu/` path is fully removed?
More than anything I'd like to make sure `/home/cecile/.local/lib/python3.6/site-packages/mu/wheels` doesn't have any .whl files left behind.
After that try installing `mu-editor` again, hopefully this time it should work.

These steps are manual worarounds until we resolve https://github.com/mu-editor/mu/issues/1441 and https://github.com/mu-editor/mu/issues/1421.

carlosperate, 2021-09-21:
This area has been heavily reworked and I believe this specific problem should be resolved so I'll close this issue.

However, if you are still experiencing this problem please reopen it, thanks!

closed 2021-09-21 by carlosperate