mu-editor/mu #1409: Crash Report

scoaa, 2021-03-26:
Version: 1.1.0.beta.2
Locale: en_HK
Platform: Linux 5.4.0-40-generic #44~18.04.1-Ubuntu SMP Wed Jun 24 23:13:08 UTC 2020 x86_64

```
Traceback (most recent call last):
  File "/home/a/.local/lib/python3.6/site-packages/mu/app.py", line 162, in run
    raise ex
  File "/home/a/.local/lib/python3.6/site-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/a/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 460, in ensure_and_create
    "Unable to create a working virtual environment."
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.

```

full log:

2021-03-26 13:08:43,277 - root:259(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.2
2021-03-26 13:08:43,277 - root:260(run) INFO: uname_result(system='Linux', node='reve', release='5.4.0-40-generic', version='#44~18.04.1-Ubuntu SMP Wed Jun 24 23:13:08 UTC 2020', machine='x86_64', processor='x86_64')
2021-03-26 13:08:43,277 - root:261(run) INFO: Python path: ['/home/a/.local/bin', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/home/a/.local/lib/python3.6/site-packages', '/usr/local/lib/python3.6/dist-packages', '/usr/lib/python3/dist-packages', '/home/a/.local/lib/python3.6/site-packages/IPython/extensions']
2021-03-26 13:08:43,277 - root:262(run) INFO: Language code: en_HK
2021-03-26 13:08:43,277 - mu.settings:220(load) WARNING: No settings file found at /home/a/.local/share/mu/settings.json; skipping
2021-03-26 13:08:43,278 - mu.settings:220(load) WARNING: No settings file found at /home/a/.local/share/mu/session.json; skipping
2021-03-26 13:08:45,828 - mu.virtual_environment:434(ensure_and_create) INFO: Added log handler.
2021-03-26 13:08:46,645 - mu.virtual_environment:439(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-03-26 13:08:46,656 - mu.virtual_environment:479(ensure_path) ERROR: /home/a/.local/share/mu/mu_venv-36-20210326-130839 does not exist.
2021-03-26 13:08:46,656 - mu.virtual_environment:443(ensure_and_create) DEBUG: Virtual environment not present or correct.
2021-03-26 13:08:46,657 - mu.virtual_environment:446(ensure_and_create) DEBUG: Creating new virtual environment at /home/a/.local/share/mu/mu_venv-36-20210326-130846.
2021-03-26 13:08:46,658 - mu.virtual_environment:378(relocate) DEBUG: Virtual environment set up mu_venv-36-20210326-130846 at /home/a/.local/share/mu/mu_venv-36-20210326-130846
2021-03-26 13:08:46,658 - mu.virtual_environment:578(create) INFO: Creating virtualenv: /home/a/.local/share/mu/mu_venv-36-20210326-130846
2021-03-26 13:08:46,658 - mu.virtual_environment:579(create) INFO: Virtualenv name: mu_venv-36-20210326-130846
2021-03-26 13:08:49,958 - mu.virtual_environment:631(install_baseline_packages) INFO: Installing baseline packages.
2021-03-26 13:08:49,959 - mu.virtual_environment:635(install_baseline_packages) INFO: /home/a/.local/lib/python3.6/site-packages/mu/wheels exists
2021-03-26 13:08:49,959 - mu.virtual_environment:647(install_baseline_packages) WARNING: No wheels found in /home/a/.local/lib/python3.6/site-packages/mu/wheels; downloading...
2021-03-26 13:08:49,960 - mu.wheels:58(download) DEBUG: dirpath: /home/a/.local/lib/python3.6/site-packages/mu/wheels
2021-03-26 13:08:49,960 - mu.wheels:74(download) DEBUG: mode_packages: [('pgzero', 'git+https://github.com/mu-editor/pgzero.git@mu-wheel'), ('flask', 'flask==1.1.2'), ('serial', 'pyserial>=3.0'), ('qtconsole', 'qtconsole==4.7.4'), ('nudatus', 'nudatus>=0.0.3'), ('esptool', 'esptool==3.*')]
2021-03-26 13:08:49,976 - mu.wheels:80(download) INFO: Running pip download for pgzero / git+https://github.com/mu-editor/pgzero.git@mu-wheel / []
2021-03-26 13:13:28,432 - mu.wheels:97(download) DEBUG: Collecting git+https://github.com/mu-editor/pgzero.git@mu-wheel
  Cloning https://github.com/mu-editor/pgzero.git (to mu-wheel) to /tmp/pip-kki9q6pd-build
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/pgzero-1.2.zip
Collecting pygame>=1.9.2 (from pgzero==1.2)
  Downloading https://files.pythonhosted.org/packages/01/da/4ff439558641a26dd29b04c25947e6c0ace041f56b2aa2ef1134edab06b8/pygame-2.0.1-cp36-cp36m-manylinux1_x86_64.whl (11.8MB)
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/pygame-2.0.1-cp36-cp36m-manylinux1_x86_64.whl
Collecting numpy (from pgzero==1.2)
  Downloading https://files.pythonhosted.org/packages/45/b2/6c7545bb7a38754d63048c7696804a0d947328125d81bf12beaa692c3ae3/numpy-1.19.5-cp36-cp36m-manylinux1_x86_64.whl (13.4MB)
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/numpy-1.19.5-cp36-cp36m-manylinux1_x86_64.whl
Successfully downloaded pgzero pygame numpy

2021-03-26 13:13:28,432 - mu.wheels:80(download) INFO: Running pip download for flask / flask==1.1.2 / []
2021-03-26 13:13:35,965 - mu.wheels:97(download) DEBUG: Collecting flask==1.1.2
  Downloading https://files.pythonhosted.org/packages/f2/28/2a03252dfb9ebf377f40fba6a7841b47083260bf8bd8e737b0c6952df83f/Flask-1.1.2-py2.py3-none-any.whl (94kB)
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/Flask-1.1.2-py2.py3-none-any.whl
Collecting Werkzeug>=0.15 (from flask==1.1.2)
  Downloading https://files.pythonhosted.org/packages/cc/94/5f7079a0e00bd6863ef8f1da638721e9da21e5bacee597595b318f71d62e/Werkzeug-1.0.1-py2.py3-none-any.whl (298kB)
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/Werkzeug-1.0.1-py2.py3-none-any.whl
Collecting itsdangerous>=0.24 (from flask==1.1.2)
  Downloading https://files.pythonhosted.org/packages/76/ae/44b03b253d6fade317f32c24d100b3b35c2239807046a4c953c7b89fa49e/itsdangerous-1.1.0-py2.py3-none-any.whl
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/itsdangerous-1.1.0-py2.py3-none-any.whl
Collecting Jinja2>=2.10.1 (from flask==1.1.2)
  Downloading https://files.pythonhosted.org/packages/7e/c2/1eece8c95ddbc9b1aeb64f5783a9e07a286de42191b7204d67b7496ddf35/Jinja2-2.11.3-py2.py3-none-any.whl (125kB)
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/Jinja2-2.11.3-py2.py3-none-any.whl
Collecting click>=5.1 (from flask==1.1.2)
  Using cached https://files.pythonhosted.org/packages/d2/3d/fa76db83bf75c4f8d338c2fd15c8d33fdd7ad23a9b5e57eb6c5de26b430e/click-7.1.2-py2.py3-none-any.whl
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/click-7.1.2-py2.py3-none-any.whl
Collecting MarkupSafe>=0.23 (from Jinja2>=2.10.1->flask==1.1.2)
[... 683 lines cut ...]
Processing ./.local/lib/python3.6/site-packages/mu/wheels/python_dateutil-2.8.1-py2.py3-none-any.whl
Processing ./.local/lib/python3.6/site-packages/mu/wheels/pyzmq-22.0.3-cp36-cp36m-manylinux1_x86_64.whl
Processing ./.local/lib/python3.6/site-packages/mu/wheels/click-7.1.2-py2.py3-none-any.whl
Processing ./.local/lib/python3.6/site-packages/mu/wheels/pygame-2.0.1-cp36-cp36m-manylinux1_x86_64.whl
Installing collected packages: six, ipython-genutils, decorator, wcwidth, traitlets, ptyprocess, parso, tornado, setuptools, pyzmq, python-dateutil, Pygments, prompt-toolkit, pickleshare, pexpect, jupyter-core, jedi, backcall, MarkupSafe, jupyter-client, ipython, Werkzeug, QtPy, Jinja2, itsdangerous, ipykernel, click, qtconsole, pyserial, pygame, numpy, nudatus, Flask
  Attempting uninstall: setuptools
    Found existing installation: setuptools 54.1.2
    Uninstalling setuptools-54.1.2:
      Successfully uninstalled setuptools-54.1.2
Successfully installed Flask-1.1.2 Jinja2-2.11.3 MarkupSafe-1.1.1 Pygments-2.8.1 QtPy-1.9.0 Werkzeug-1.0.1 backcall-0.2.0 click-7.1.2 decorator-4.4.2 ipykernel-5.5.0 ipython-7.16.1 ipython-genutils-0.2.0 itsdangerous-1.1.0 jedi-0.18.0 jupyter-client-6.1.12 jupyter-core-4.7.1 nudatus-0.0.5 numpy-1.19.5 parso-0.8.1 pexpect-4.8.0 pickleshare-0.7.5 prompt-toolkit-3.0.18 ptyprocess-0.7.0 pygame-2.0.1 pyserial-3.5 python-dateutil-2.8.1 pyzmq-22.0.3 qtconsole-4.7.4 setuptools-54.2.0 six-1.15.0 tornado-6.1 traitlets-4.3.3 wcwidth-0.2.5

2021-03-26 13:26:05,768 - mu.virtual_environment:202(run) DEBUG: About to run blocking: /home/a/.local/share/mu/mu_venv-36-20210326-132546/bin/pip, ['list', '--disable-pip-version-check'], 30.0
2021-03-26 13:26:06,179 - mu.virtual_environment:606(install_jupyter_kernel) INFO: Installing Jupyter Kernel: "Python/Mu (mu_venv-36-20210326-132546)"
2021-03-26 13:26:06,823 - mu.virtual_environment:456(ensure_and_create) ERROR: Unable to create a working virtual environment.
2021-03-26 13:26:13,848 - root:174(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
[... cut ...]
    venv.ensure_and_create(self.display_text)
  File "/home/a/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 460, in ensure_and_create
    "Unable to create a working virtual environment."
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.
2021-03-26 13:26:13,897 - mu.settings:169(save) DEBUG: Saving to /home/a/.local/share/mu/session.json
2021-03-26 13:26:13,897 - mu.settings:169(save) DEBUG: Saving to /home/a/.local/share/mu/venv.json

carlosperate, 2021-04-14:
Thank you for the report @scoaa. Do you still have this issue? If so, would you be able to:
- Uninstall mu-editor
- Ensure the `./.local/lib/python3.6/site-packages/mu/` directory is removed
- Update pip
- Install mu-editor again

Hopefully that should resolve the issue.

carlosperate, 2021-09-21:
This area has been heavily reworked and I believe this specific problem should be resolved so I'll close this issue due to lack of activity.

However, if you are still experiencing this problem please reopen it, thanks!

closed 2021-09-21 by carlosperate