mu-editor/mu #1410: Crash Report

brucebaranski, 2021-03-26:
Version: 1.1.0.beta.2
Locale: en_US
Platform: Darwin 19.6.0 Darwin Kernel Version 19.6.0: Tue Jan 12 22:13:05 PST 2021; root:xnu-6153.141.16~1/RELEASE_X86_64 x86_64

```
Traceback (most recent call last):
  File "/private/var/folders/xk/_rjk4wdj1rq4y8yrdk7bsthr0000gn/T/AppTranslocation/2D829160-E414-4519-981C-41B95929E0A3/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/private/var/folders/xk/_rjk4wdj1rq4y8yrdk7bsthr0000gn/T/AppTranslocation/2D829160-E414-4519-981C-41B95929E0A3/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/private/var/folders/xk/_rjk4wdj1rq4y8yrdk7bsthr0000gn/T/AppTranslocation/2D829160-E414-4519-981C-41B95929E0A3/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/__main__.py", line 9, in <module>
    main()
  File "/private/var/folders/xk/_rjk4wdj1rq4y8yrdk7bsthr0000gn/T/AppTranslocation/2D829160-E414-4519-981C-41B95929E0A3/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/__main__.py", line 5, in main
    run()
  File "/private/var/folders/xk/_rjk4wdj1rq4y8yrdk7bsthr0000gn/T/AppTranslocation/2D829160-E414-4519-981C-41B95929E0A3/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/app.py", line 339, in run
    editor.setup(setup_modes(editor, editor_window))
  File "/private/var/folders/xk/_rjk4wdj1rq4y8yrdk7bsthr0000gn/T/AppTranslocation/2D829160-E414-4519-981C-41B95929E0A3/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/logic.py", line 806, in setup
    os.makedirs(images_path)
  File "/private/var/folders/xk/_rjk4wdj1rq4y8yrdk7bsthr0000gn/T/AppTranslocation/2D829160-E414-4519-981C-41B95929E0A3/d/Mu Editor.app/Contents/MacOS/../Resources/Python/lib/python3.8/os.py", line 223, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 1] Operation not permitted: '/Volumes/CIRCUITPY/images'

```

carlosperate, 2021-04-06:
Thanks for the report @brucebaranski!
Could you try steps listed in this post? Should help with the "Operation not  permitted" error: https://github.com/mu-editor/mu/issues/1383#issuecomment-810537546

carlosperate, 2021-09-21:
This should be resolved with the steps listed in https://github.com/mu-editor/mu/issues/1383#issuecomment-810537546, so I'll close this issue due to lack of activity.

However, if you are still experiencing this problem please reopen it, thanks!