mu-editor/mu #1411: Crash Report

m20191201, 2021-03-27:
Version: 1.1.0.beta.2
Locale: en_US
Platform: Windows 7 6.1.7601 AMD64

```
Traceback (most recent call last):
  File "d:\program files\python38\lib\site-packages\mu\virtual_environment.py", line 550, in ensure_key_modules
    subprocess.run(
  File "d:\program files\python38\lib\subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['C:\\Users\\imin\\AppData\\Local\\python\\mu\\mu_venv-38-20210327-145840\\scripts\\python.exe', '-c', 'import pgzero']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "d:\program files\python38\lib\site-packages\mu\virtual_environment.py", line 441, in ensure_and_create
    self.ensure()
  File "d:\program files\python38\lib\site-packages\mu\virtual_environment.py", line 471, in ensure
    self.ensure_key_modules()
  File "d:\program files\python38\lib\site-packages\mu\virtual_environment.py", line 559, in ensure_key_modules
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: Failed to import: pgzero

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "d:\program files\python38\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "d:\program files\python38\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "d:\program files\python38\lib\site-packages\mu\virtual_environment.py", line 449, in en
```

tjguk, 2021-03-27:
Thanks for the report @m20191201  -- would you be able to attach the full log, please, to this GH issue? It should be at:

`%LOCALAPPDATA%\python\mu\logs`

carlosperate, 2021-03-29:
Thanks for the report @m20191201. If you could send the log file as mentioned by tjguk that would be great.
Do you know if git is installed in your computer?

carlosperate, 2021-03-29:
@m20191201 there is a new release that we think has fixed this issue. Would you be able to install the latest version and try again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.3

carlosperate, 2021-08-10:
Hi @m20191201,
Would you be able to check if beta.5 works for you and let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

carlosperate, 2021-09-21:
This area has been heavily reworked and I believe this specific problem should be resolved so I'll close this issue due to lack of activity.

However, if you are still experiencing this problem with the latest beta please reopen it, thanks!