mu-editor/mu #1413: OSX Beta.3 Virtual Env Crash Report

ntoll, 2021-03-27:
Version: 1.1.0.beta.3
Locale: en_GB
Platform: Darwin 19.5.0 Darwin Kernel Version 19.5.0: Tue May 26 20:41:44 PDT 2020; root:xnu-6153.121.2~2/RELEASE_X86_64 x86_64

```
Traceback (most recent call last):
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/app.py", line 162, in run
    raise ex
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 459, in ensure_and_create
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.

```

Here's what I did:

* Mounted the `dmg` and installed Mu by dragging it to the Applications folder.
* I started Mu by clicking its icon in the Applications folder, the new splash screen appeared as expected.
* After a while I noticed that it was going through the venv creation steps more than once.
* In the end, it dumped me into the crash report.
* Wondering if this was down to existing assets from a previous version of Mu, I deleted all the existing venvs and started Mu again, from a "fresh" start.
* Same thing happened. 
* I couldn't help but notice that it always restarted creation of the venvs after it reported it was trying to install Jupyter.
* Could this be down to the new way in which we kick off QProcesses..?

I've attached the log file, the first run logged in there being me running from source via `python run.py`. I suspect this is one for @tjguk and @carlosperate to look at initially. This only appears to be a problem when running the installed version of Mu, not the version run from source. I've also tried the following steps:

* Run Mu, from a clean state, from source. It just works (the venv is created).
* Run Mu, from a clean state, from installed version. I see the behaviour shown above.
* Run Mu, from the previous failed state (now with three venvs), from the installed version. I see the same behaviour as above, and now I have six venvs.
* Run Mu, from the previous failed state (now with six venvs), from source. It picks up the latest venv and works. I try to run the REPL (seeing as it appears that it failed *after* ensuring Jupyter), and THAT works.
* What is it about the installed version of Mu, that means it doesn't see there's a valid venv, so it keeps trying to recreate a new one..? @tmontes any guesses..?

**This blocks beta 3** :-/ (c'est la vie)

Something to discuss on Monday perhaps..?

[mu.log](https://github.com/mu-editor/mu/files/6216260/mu.log)

ntoll, 2021-03-27:
Worth pointing out that there's "no rush" to fix this. We're all volunteers, it's the weekend, and these things happen...

tjguk, 2021-03-28:
Well this is weird:

`ERROR: Mu interpreter at version 38; venv interpreter at version 27.`

tjguk, 2021-03-28:
Ok; so what I think is happening is this:

In `virtual_environment.VirtualEnvironment:ensure_interpreter_version` we purportedly run the interpreter from the virtual environment to make sure it's the same version as the one which Mu itself is running under. (This to avoid issues with wheel version mismatches etc.)

For reasons which might not still hold good, we use `subprocess.run` to do this, rather than QProcess.

We started to encounter users on Win10 with spaces in their usernames which failed this check. There are known issues around starting processes with spaces in the path on Windows (cf https://stackoverflow.com/questions/4053241/windows-api-createprocess-path-with-space).

The `subprocess.Popen` API doesn't give us enough control to implement the sort of solution mentioned in that SO article, so in an attempt to work around it a different way, I reimplemented that call so it uses an unqualified path name but sets the cwd to the venv's binary directory first.

On Windows inmy dev environment this works -- although I can't tell if it solves the original problem or not because I can't reproduce the error. Going by @ntoll explanation it looks like it works in a OSX dev env as well. But something about the installed version causes it to fail.

Most likely thing is whether the cwd is considered on the path or not for the purposes of an exe call. If it *isn't* then we're executing `python` with no full path -- and that'll probably translate to the 2.7 system Python on a Mac. On Win, it'll probably be captured by the App Store stub and produce a different kind of failure result.

Since I'm not even sure that it fixes the original problem, I propose to back that change out so that at least regular usernames don't encounter issues. I've got a couple of ideas about how we might work around the original problem, but I'll need to reproduce it first.

tjguk, 2021-03-29:
Fixed by #1415

closed 2021-03-29 by tjguk