mu-editor/mu #1414: Crash Report

Haifeng-Ding, 2021-03-28:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.2
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\dingh\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 441, in ensure_and_create
    self.ensure()
  File "C:\Users\dingh\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "C:\Users\dingh\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: C:\Users\dingh\AppData\Local\python\mu\mu_venv-38-20210327-172640 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\dingh\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\dingh\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\dingh\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "C:\Users\dingh\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 582, in create
    subprocess.run(
  File "C:\Users\dingh\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['C:\\Users\\dingh\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-m', 'virtualenv', '-p', 'C:\\Users\\dingh\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-q', 'C:\\Users\\dingh\\AppData\\Local\\python\\mu\\mu_venv-38-20210327-172646']' returned non-zero exit status 1.

```

carlosperate, 2021-03-29:
Thanks for the report @Haifeng-Ding, would you be able to attach the log to this issue as well? It'd really help us debug it. The log file should be located in `%LOCALAPPDATA%\python\mu\logs`.

carlosperate, 2021-08-10:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
Thanks for sending the crash report!

closed 2021-08-10 by carlosperate