mu-editor/mu #1416: Can't flash Microbitv2 on BigSur with latest MU-editor

marc0janssen, 2021-03-28:
I can't seem to flash a microbit V2 with MU-editor.

from microbit import *

boat = Image("05050:" "05050:" "05050:" "99999:" "09990")

display.show(boat)
2021-03-28 14:05:21,715 - root:259(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.2
2021-03-28 14:05:21,715 - root:260(run) INFO: uname_result(system='Darwin', node='MacBook-Pro-van-Marco.local', release='20.3.0', version='Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-03-28 14:05:21,715 - root:261(run) INFO: Python path: ['/', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python38.zip', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/lib-dynload', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/setuptools-49.6.0-py3.8.egg', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/pip-20.2.2-py3.8.egg', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/IPython/extensions']
2021-03-28 14:05:21,715 - root:262(run) INFO: Language code: nl_NL
2021-03-28 14:05:21,715 - mu.settings:220(load) WARNING: No settings file found at /Users/marco/Library/Application Support/mu/settings.json; skipping
2021-03-28 14:05:21,715 - mu.settings:220(load) WARNING: No settings file found at /Users/marco/Library/Application Support/mu/session.json; skipping
2021-03-28 14:05:22,020 - mu.virtual_environment:434(ensure_and_create) INFO: Added log handler.
2021-03-28 14:05:22,161 - mu.virtual_environment:438(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-03-28 14:05:22,166 - mu.virtual_environment:479(ensure_path) ERROR: /Users/marco/Library/Application Support/mu/mu_venv-38-20210328-140521 does not exist.
2021-03-28 14:05:22,166 - mu.virtual_environment:443(ensure_and_create) DEBUG: Virtual environment not present or correct.
2021-03-28 14:05:22,173 - mu.virtual_environment:445(ensure_and_create) DEBUG: Creating new virtual environment at /Users/marco/Library/Application Support/mu/mu_venv-38-20210328-140522.
2021-03-28 14:05:22,176 - mu.virtual_environment:377(relocate) DEBUG: Virtual environment set up mu_venv-38-20210328-140522 at /Users/marco/Library/Application Support/mu/mu_venv-38-20210328-140522
2021-03-28 14:05:22,176 - mu.virtual_environment:578(create) INFO: Creating virtualenv: /Users/marco/Library/Application Support/mu/mu_venv-38-20210328-140522
2021-03-28 14:05:22,176 - mu.virtual_environment:579(create) INFO: Virtualenv name: mu_venv-38-20210328-140522
2021-03-28 14:05:22,954 - mu.virtual_environment:631(install_baseline_packages) INFO: Installing baseline packages.
2021-03-28 14:05:22,959 - mu.virtual_environment:632(install_baseline_packages) INFO: /Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels exists
2021-03-28 14:05:22,966 - mu.virtual_environment:229(install) DEBUG: About to pip install: ['/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/esptool-3.0-py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/Jinja2-2.11.3-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/ipython-7.21.0-py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/MarkupSafe-1.1.1-cp38-cp38-macosx_10_9_x86_64.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/ecdsa-0.16.1-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/QtPy-1.9.0-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/pyzmq-22.0.3-cp38-cp38-macosx_10_9_x86_64.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/jedi-0.18.0-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/parso-0.8.1-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/pickleshare-0.7.5-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/traitlets-5.0.5-py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/ptyprocess-0.7.0-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/pygame-2.0.1-cp38-cp38-macosx_10_9_intel.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/Flask-1.1.2-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/pycparser-2.20-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/bitstring-3.1.7-py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/setuptools-54.1.1-py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/pexpect-4.8.0-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/jupyter_core-4.7.1-py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/cffi-1.14.5-cp38-cp38-macosx_10_9_x86_64.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/pgzero-1.2-py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/jupyter_client-6.1.11-py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/Werkzeug-1.0.1-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/prompt_toolkit-3.0.17-py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/pyserial-3.5-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/backcall-0.2.0-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/numpy-1.20.1-cp38-cp38-macosx_10_9_x86_64.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/python_dateutil-2.8.1-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/click-7.1.2-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/tornado-6.1-cp38-cp38-macosx_10_9_x86_64.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/qtconsole-4.7.4-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/itsdangerous-1.1.0-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/ipykernel-5.5.0-py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/reedsolo-1.5.4-py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/cryptography-3.4.6-cp36-abi3-macosx_10_10_x86_64.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/Pygments-2.8.1-py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/appnope-0.1.2-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/six-1.15.0-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/wcwidth-0.2.5-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/decorator-4.4.2-py2.py3-none-any.whl', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu [... cut ...]
2021-03-28 14:47:52,214 - mu.logic:1396(show_admin) INFO: Showing admin with logs from /Users/marco/Library/Logs/mu/mu.log
2021-03-28 14:47:52,215 - mu.virtual_environment:697(installed_packages) INFO: Discovering installed third party modules in venv.
2021-03-28 14:47:52,215 - mu.virtual_environment:198(run) DEBUG: About to run blocking: /Users/marco/Library/Application Support/mu/mu_venv-38-20210328-140522/bin/pip, ['list', '--disable-pip-version-check'], 30.0
2021-03-28 14:47:52,486 - mu.virtual_environment:717(installed_packages) INFO: []

marc0janssen, 2021-03-28:
OK... this is something weird.

1. It did not work for 2 hours of trying. --> nothing worked
2. Flashed other HEX-files from different editors --> all worked but MUeditor
3. Disconnected the MicroBit
4. Tried to save a HEX file from MU-editor without the MicroBit attached this time --> got a HEX-file
5.Attached the Microbit back and copied the mu-editor-HEX-file by hand on the Micro-bit --> this  failed.. got an error the HEX-file was not valid.
6. Flashed an HEX from another editor --> this worked fine (and the FAILED.TXT disappeared from the microbit)
7. started the mu-editor and flashed again --> THIS TIME IT WORKED and keeps working.

I can't explain this

marc0janssen, 2021-03-28:
Found out something new: 

After a while the problem was back.

When I deleted the "main.py" in files from the MicroBit, I could flash again.

marc0janssen, 2021-03-28:
After uploading a HEX from makecode.. I run into the same problems again... can't flash python-source to the microbit

Can someone help me what I'm doing wrong?

carlosperate, 2021-03-29:
Can you attach to this post (it might need  to be on a zip file) the hex that caused a fail.txt? A copy of the fail.txt file and  the details.txt file would be useful  too.

marc0janssen, 2021-03-29:
The fail.txt will not always appear,... If it appears it is when I disconnect the microbit, then generate a HEX on disk.
and copy this HEX by the hand on the microbit... then sometimes the fail.txt appears... at the moment is it not the case.
If I have it, I will share.

Weird stuff is that I can't reflash the microbit unless I delete the main.py from the microbit

marc0janssen, 2021-03-30:
FAIL.TXT
error: In application programming failed because the update sent was incomplete.
type: interface

DETAILS.TXT
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 9904360258824e450031800200000048000000009796990b
HIC ID: 9796990b
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Incompatible image detection: 1
Page erasing: 0
Daplink Mode: Interface
Interface Version: 0255
Bootloader Version: 0255
Git SHA: 1436bdcc67029fdfc0ff03b73e12045bb6a9f272
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0x828c6069
Interface CRC: 0x5b5cc0f5
Remount count: 5
URL: https://microbit.org/device/?id=9904&v=0255

carlosperate, 2021-03-30:
Thanks @marc0janssen do you also have the hex that has caused this? That file would be really useful to debug this.

marc0janssen, 2021-03-30:
How I fixed it:

1. in FINDER go to the folder /bin
2. open the system setting for security and privacy
3. select privacy
4. Unlock with the pad-lock the screen
5. drag and drop /bin/sh in the "right-side" part of the window where the apps are .
6. tick the box next to the just copied "SH".

This fixed it for me...

marc0janssen, 2021-03-30:
> Thanks @marc0janssen do you also have the hex that has caused this? That file would be really useful to debug this.

Sorry the was lost in the process and after the above procedure all is fine for me now...

The flash takes the aspected time now and the program runs,,,

(weird fact... with the problem there, if the occasional did work, the flash was really really short, compared to the normal flash)

carlosperate, 2021-03-30:
Thanks for sharing that solution! it's really appreciated :)

>Sorry that was lost in the process and after the above procedure all is fine for me now...
>
>The flash takes the aspected time now and the program runs,,,

I think there might be two problem in this issue. One was the permission issue that has been solved by adding `/bin/sh` to the security settings. The other seems to be that hex files generated by Mu might not be valid?

If that is the case I would expect the fail.txt issue might happen again in the future.

> (weird fact... with the problem there, if the occasional did work, the flash was really really short, compared to the normal flash)

Yeah, based on the error looks like the file is not complete.

marc0janssen, 2021-03-30:
Hi Carlos. 

That could be. But when the flashing was not working. In others words I hadn't added the sh to security, I could flash the program be removing the Main.py from the microbit and push the flash button. 

That's also weird isn't it?

carlosperate, 2021-03-31:
Right, generally the way that it works is that if Mu detects micro:bit already has MicroPython running on the device it will send the code via serial (serial works without adding the additional permissions to `/bin/sh`).
The main problem happens when Mu has issues detecting if MicroPython is in device, at that point it tries to copy a micropython.hex file (with just MicroPython) into the MICROBIT drive, and then send the files via serial. Is in this case that the permission issue is triggered and caused problems.

When clicking the "Flash" button without the micro:bit plugged in, Mu will ask where to save a (specially generated) .hex file with MicroPython and the user code attached inside. This method is only used in special cases, as it has several disadvantages, but is still available.  

If a generated hex file from Mu (with MicroPython and the user code) causes a fail.txt file in the micro:bit, it probably means there is a bug in the creation of this special hex file, which is why I was so interested to analyse one, and ideally get the Python source code as well to see if I could recreate the problem to debug it.

Btw, thanks again for all the testing and reports you've done, it's been incredibly useful!