mu-editor/mu #1417: Crash Report

Aristman, 2021-03-28:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.2
Locale: ru_RU
Platform: Linux 5.8.0-48-generic #54~20.04.1-Ubuntu SMP Sat Mar 20 13:40:25 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/robot1/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 550, in ensure_key_modules
    subprocess.run(
  File "/usr/lib/python3.8/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/home/robot1/.local/share/mu/mu_venv-38-20210328-205421/bin/python', '-c', 'import pgzero']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/robot1/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 441, in ensure_and_create
    self.ensure()
  File "/home/robot1/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 471, in ensure
    self.ensure_key_modules()
  File "/home/robot1/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 559, in ensure_key_modules
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: Failed to import: pgzero

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/robot1/.local/lib/python3.8/site-packages/mu/app.py", line 162, in run
    raise ex
  File "/home/robot1/.local/lib/python3.8/site-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/robot1/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "/home/robot1/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 596, in create
    self.install_baseline_packages()
  File "/home/robot1/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 649, in install_baseline_packages
    wheels.download()
  File "/home/robot1/.local/lib/python3.8/site-packages/mu/wheels/__init__.py", line 82, in download
    process = subprocess.run(
  File "/usr/lib/python3.8/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'download', '--destination-directory', '/home/robot1/.local/lib/python3.8/site-packages/mu/wheels', 'git+https://github.com/mu-editor/pgzero.git@mu-wheel']' returned non-zero exit status 1.

```

carlosperate, 2021-03-29:
Thanks for the report @Aristman.
Would you be able to attach the log into this ticket as well? It will be really useful to help us debug it. In linux is should be in : `~/.cache/mu/`

Apart from that could you run the following commands on terminal and copy/paste the output here as well?
```
git --version
```
```
/usr/bin/python3 -m pip --version
```

carlosperate, 2021-03-29:
@Aristman there is a new release that we think has fixed this issue. Would you be able to install the latest version and try again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.3

Aristman, 2021-03-29:
I installed a new version of your program in a new virtual environment and everything works fine. Thanks for solving the problem!

carlosperate, 2021-03-29:
That's fantastic, thanks for testing again @Aristman!

closed 2021-03-29 by carlosperate