mu-editor/mu #1418: Crash ReportTraceback (most recent call last):   File "C:\Users\hp\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run     raise ex   File "C:\Users\hp\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run     venv.ensure_and_create(self.display_text)   File "C:\Users\hp\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 459, in ensure_and_create     raise VirtualEnvironmentError( mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.

wes4545, 2021-03-28:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.2
Locale: en_US
Platform: Windows 10 10.0.15063 AMD64

```
Traceback (most recent call last):
  File "C:\Users\hp\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\hp\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\hp\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 459, in ensure_and_create
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.

```

carlosperate, 2021-03-29:
Hi @wes4545, would you be able to attach the log file to this ticket, it will help us diagnose the issue.
The log file can be found in: `%LOCALAPPDATA%\python\mu\Logs`

carlosperate, 2021-08-10:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
Thanks for sending the crash report!

closed 2021-08-10 by carlosperate