mu-editor/mu #1420: Crash Report

jesperstein, 2021-03-29:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: da_DK
Platform: Windows 10 10.0.18362 AMD64

```
Traceback (most recent call last):
  File "C:\Users\jespe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 442, in ensure_and_create
    self.ensure()
  File "C:\Users\jespe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 470, in ensure
    self.ensure_pip()
  File "C:\Users\jespe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 573, in ensure_pip
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: Pip not found where expected at: C:\Users\jespe\AppData\Local\python\mu\mu_venv-38-20210329-232155\scripts\pip.exe

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\jespe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 299, in installed
    next(iterlines)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\jespe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\jespe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\jespe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "C:\Users\jespe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 598, in create
    self.register_baseline_packages()
  File "C:\Users\jespe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 665, in register_baseline_packages
    packages = list(self.pip.installed())
  File "C:\Users\jespe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 305, in installed
    raise VirtualEnvironmentError("Unable to parse installed packages")
mu.virtual_environment.VirtualEnvironmentError: Unable to parse installed packages

```

carlosperate, 2021-04-06:
Thanks for the report @jesperstein.
Would you be able to also attach the Mu log into this GH Issue? it will be really helpful. File should be located in: `%LOCALAPPDATA%\python\mu\Logs`.

jesperstein, 2021-04-08:
Hi @carlosperate,

Yes of course, see the file below:
[Log.txt](https://github.com/mu-editor/mu/files/6280769/Log.txt)

Will you be able to help me, i really enjoyed using MU Editor for coding on Micro:Bit :)

Thanks for answering

Best regards
Jesper

carlosperate, 2021-04-08:
Thanks @jesperstein, thanks a lot for the log!

Could you run this in the command prompt (you can type `cmd.exe` in the windows menu) and copy/paste the output here?

```
dir C:\Users\jespe\AppData\Local\python\mu\mu_venv-38-20210329-232155\scripts\
```

@tjguk any thoughts on how pip might have been missing after the virtualenv modules creates the venv?

tjguk, 2021-04-09:
> any thoughts on how pip might have been missing after the virtualenv modules creates the venv?

None at all -- and what's surprising me is that it seems to go on to try to install the baseline packages, while I would have expected it to fail up front because of a lack of Pip.

jesperstein, 2021-04-18:
@carlosperate sorry for the late reply!

I have now entered it into cmd, and here's the result:

[image]

carlosperate, 2021-04-19:
Thanks for testing @jesperstein! It's a bit odd that pip.exe is not present in that folder 🤔 

We just released a new beta version, would you be able to install the latest release from this link and try again?
If it still doesn't work if you could send us the log again it should contain a bit more info that would help us debug it.
https://github.com/mu-editor/mu/releases

carlosperate, 2021-08-10:
Hi @jesperstein,

Would you be able to check if beta.5 works for you and let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

carlosperate, 2021-09-21:
@jesperstein  have you been able to test beta.5? There has been some fixes in this area that could have resolved the problem you were experiencing: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

carlosperate, 2021-10-14:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as this area has been reworks significantly since beta.3 and we believe this has been resolved: https://github.com/mu-editor/mu/releases/