mu-editor/mu #1421: venv doesn't verify if all required wheels are present in mu/wheels/

carlosperate, 2021-03-30:
It basically it (kind of) assumes that if any wheel is present, all required wheels are present.

https://github.com/mu-editor/mu/blob/ad16f421b091f82b50124a176df429398482011f/mu/virtual_environment.py#L644-L649

Of course, if a wheels is missing, it will then fail the `ensure_key_modules()` step, so it will fail (correctly) at that point during the venv creation. But other than that it will not try to download the wheels at any stage. This might be good, if we don't want to have the Mu from installers trying to download wheels during the start up process. But it also means it misses a chance to recover if an internet connection is present.

This will initially mostly affect developer environments, as that is the easiest way to have differences between  `mu.wheels.__init__` and the present wheels, in this case we "just" need to remember to manually remove all the wheels from the `mu/wheels/` directory.

However, I think it could also be a problem for users pip upgrading Mu from PyPI. Does pip remove the old mu folder before installing the new? would the old wheels stay and cause this issue?

Perhaps we should verify in `install_baseline_packages` (the code snippet above) that all the wheels required in `mu.wheels.__init__` are present?

carlosperate, 2021-04-19:
> However, I think it could also be a problem for users pip upgrading Mu from PyPI. Does pip remove the old mu folder before installing the new? would the old wheels stay and cause this issue?

Based on https://github.com/mu-editor/mu/issues/1488 I think this is an issue for pip upgrading Mu.

carlosperate, 2021-04-20:
Just to add a note here as well, based on https://github.com/mu-editor/mu/issues/1496, we can encounter the same issue when users update Mu via the Windows installer. It looks like the update overwrites existing files, but also leaves behind any files from the previous installation.

tjguk, 2021-05-31:
My current thinking -- which I don't have time to do tonight -- is to give each mu-version its own directory under `wheels/`. (Could be a zip file, but doesn't seem to add much).

That way, the download process for a given version writes the files it needs, and the venv create process reads the right ones, too.

carlosperate, 2021-06-20:
PR https://github.com/mu-editor/mu/pull/1650

carlosperate, 2021-08-04:
@SteveChoatie @prefontdon @parsko41 @Elikyals @dcarmean @soarek @mitag @webbbn @donxw

Hi everyone! 👋

The issues you were having with Mu should have been resolved with the beta.5 release: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Would you be able to try the installer from this release and check if it is working correctly in your environment?

Thanks everyone for your help testing the Mu Editor!

carlosperate, 2021-09-22:
We haven't received any reports about this issue since beta.5 was released, so I'll consider this resolved.
Thanks everybody!

closed 2021-09-22 by carlosperate