mu-editor/mu #1423: My script that breaks down GIF frames into images and plays them like a flipbook works in the terminal and other editors but not mu

MeadowofFlowers, 2021-03-30:
Mu won't produce a log so here is the script:

import sys
import os
import time
import pihole as ph
picdir = os.path.join(os.path.dirname(os.path.dirname(os.path.realpath(__file__))), 'pic')
libdir = os.path.join(os.path.dirname(os.path.dirname(os.path.realpath(__file__))), 'lib')
if os.path.exists(libdir):
	sys.path.append(libdir)

from waveshare_1in8_LCD import LCD_1in8
from waveshare_1in8_LCD import LCD_Config

from PIL  import Image
from PIL import ImageDraw
from PIL import ImageFont
from PIL import ImageColor

location = '/home/pi/waveshare_1in8_LCD/GIF'
files_in_dir = []
speed = .08
def analyseImage(path):
	'''
	Pre-process pass over the image to determine the mode (full or additive).
	Necessary as assessing single frames isn't reliable. Need to know the mode 
	before processing all frames.
	'''
	im = Image.open(path)
	results = {
		'size': im.size,
		'mode': 'full',
	}
	try:
		while True:
			if im.tile:
				tile = im.tile[0]
				update_region = tile[1]
				update_region_dimensions = update_region[2:]
				if update_region_dimensions != im.size:
					results['mode'] = 'partial'
					break
			im.seek(im.tell() + 1)
	except EOFError:
		pass
	return results

def processImage(path):    
	
	'''
	Iterate the GIF, extracting each frame.
	'''
	mode = analyseImage(path)['mode']
	
	im = Image.open(path)

	i = 0
	p = im.getpalette()
	last_frame = im.convert('RGBA')  
	
	try:
		while True:
			#print "saving %s (%s) frame %d, %s %s" % (path, mode, i, im.size, im.tile)
			
			'''
			If the GIF uses local colour tables, each frame will have its own palette.
			If not, we need to apply the global palette to the new frame.
			'''
			if not im.getpalette():
				im.putpalette(p)
			
			new_frame = Image.new('RGBA', im.size)
			
			'''
			Is this file a "partial"-mode GIF where frames update a region of a different size to the entire image?
			If so, we need to construct the new frame by pasting it on top of the preceding frames.
			'''
			if mode == 'partial':
				new_frame.paste(last_frame)
			
			new_frame.paste(im, (0,0), im.convert('RGBA'))
			new_frame.save('%s-%d.png' % (''.join(os.path.basename(path).split('.')[:-1]), i), 'PNG')
			i += 1
			last_frame = new_frame
			im.seek(im.tell() + 1)
	except EOFError:
		pass

def main():
	LCD = LCD_1in8.LCD()
	processImage('nyan.gif')
	
	print ("**********Init LCD**********")
	Lcd_ScanDir = LCD_1in8.SCAN_DIR_DFT  #SCAN_DIR_DFT = D2U_L2R
	LCD.LCD_Init(Lcd_ScanDir)
	LCD.LCD_Clear(0xffff);
	  
	for r, d, f in os.walk(location):
		for item in f:
			if '.png' in item:
				files_in_dir.append(os.path.join(r, item))
				
	sort = sorted(files_in_dir)
	while True:
		for file in sort:
			image = Image.new("RGB", (LCD.LCD_Dis_Column, LCD.LCD_Dis_Page), "BLACK")
			draw = ImageDraw.Draw(image) 
			font = ImageFont.truetype(picdir+'/universe.ttf', 20)
			
			base = Image.open(file).convert('RGBA') 
			
			txt = Image.new('RGBA', base.size, (255,255,255,0))
		
			d = ImageDraw.Draw(txt)
			
			X = 5
			Y = 20
			P = 48
			S = 3
		
			#d.text((X,P), "Nyan",font=font, fill="red")
			
			image = Image.alpha_composite(base,txt)
			
			LCD.LCD_ShowImage(image)
			time.sleep(speed)
	

try:
	if __name__ == '__main__':
		main()
except:
   print("except")

ntoll, 2021-03-30:
Hi @MeadowofFlowers ,

Could you please attach your log file..? You should be able to get it via the instructions found here: https://codewith.mu/en/howto/1.1/read_logs

Thank you..!

carlosperate, 2021-08-24:
@MeadowofFlowers are you still experincing this problem?

If so, would you be able to attach the logs?
And would you be able to reduce the example script to the bare minimum programme that illustrates the problem?
There is code there specific to an (raspberry pi?) LCD module that makes it really hard for us to replicate. If you can get the same the problem is just a few lines we might be able to look into it.

If you also remove the try...except part and copy paste any possible exception the programme is throwing that would be helpful too.

closed 2021-10-14 by carlosperate