mu-editor/mu #1427: Crash Report

sandstromd, 2021-03-31:
[mu.log](https://github.com/mu-editor/mu/files/6233413/mu.log)
[mu.log](https://github.com/mu-editor/mu/files/6233418/mu.log)

[mu.log](https://github.com/mu-editor/mu/files/6233423/mu.log)
[mu.log](https://github.com/mu-editor/mu/files/6233427/mu.log)
[mu.log](https://github.com/mu-editor/mu/files/6233430/mu.log)
[mu.log](https://github.com/mu-editor/mu/files/6233432/mu.log)

**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: en_US
Platform: Linux 5.4.0-70-generic #78~18.04.1-Ubuntu SMP Sat Mar 20 14:10:07 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/sandstr/.local/lib/python3.6/site-packages/mu/app.py", line 162, in run
    raise ex
  File "/home/sandstr/.local/lib/python3.6/site-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/sandstr/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 460, in ensure_and_create
    "Unable to create a working virtual environment."
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.

```

carlosperate, 2021-04-07:
Thanks for the report @sandstromd!
I believe this issue is caused by this: https://github.com/mu-editor/mu/issues/1441

Would you be able to update your system pip to the latest version? Or create a virtual environment and update pip there to `pip install mu-editor`?

sandstromd, 2021-05-11:
> Hi @sandstromd <https://github.com/sandstromd>, I was just wondering if
> you were able to fix the issue by updating pip and reinstalling the Mu
> editor?

Dear Carlos, Sorry, I gave up on mu. Feel free to close the ticket. Thanks
very much for your attention and followup.

Best - Don

closed 2021-07-05 by carlosperate