mu-editor/mu #1428: Crash Report

AlanLL, 2021-03-31:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: en_GB
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Alan_\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 750, in on_ls_fail
    self.disable()
  File "C:\Users\Alan_\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 689, in disable
    self.microbit_fs.setDisabled(True)
RuntimeError: wrapped C/C++ object of type MicroPythonDeviceFileList has been deleted

```

carlosperate, 2021-03-31:
Hi @AlanLL,

Thanks for the report!
Would you be able to attach the Mu log file here? it'll be really useful for us. File should be located in: `%LOCALAPPDATA%\python\mu\Logs`

AlanLL, 2021-03-31:
Hi Carlos,

Log attached

Regards Alan

carlosperate, 2021-03-31:
Thanks Alan, however I think maybe GitHub doesn't enable email attachments as GH issue responses and we didn't get the file.
Would you be able to add the log to the issue via the GitHub website? (it can be dragged and dropped into the text field)
https://github.com/mu-editor/mu/issues/1428

Sorry for the inconvenience, and thanks again for the bug report!

AlanLL, 2021-03-31:
Log content:
2021-03-31 15:45:41,937 - root:266(run) INFO:

-----------------

Starting Mu 1.1.0.beta.3
2021-03-31 15:45:41,937 - root:267(run) INFO: uname_result(system='Windows', node='HomePC', release='10', version='10.0.19041', machine='AMD64', processor='AMD64 Family 23 Model 17 Stepping 0, AuthenticAMD')
2021-03-31 15:45:41,937 - root:268(run) INFO: Python path: ['C:\\Users\\Alan_\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\Alan_\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\Alan_\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\Alan_\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\Alan_\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\Alan_\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\Alan_\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\Alan_\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\Alan_\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\Alan_\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\Alan_\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\Alan_\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-03-31 15:45:41,937 - root:269(run) INFO: Language code: en_GB
2021-03-31 15:45:41,937 - mu.settings:220(load) WARNING: No settings file found at C:\Users\Alan_\AppData\Local\python\mu\settings.json; skipping
2021-03-31 15:45:41,937 - mu.settings:220(load) WARNING: No settings file found at C:\Users\Alan_\AppData\Local\python\mu\session.json; skipping
2021-03-31 15:45:42,405 - mu.virtual_environment:435(ensure_and_create) INFO: Added log handler.
2021-03-31 15:45:43,280 - mu.virtual_environment:439(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-03-31 15:45:43,280 - mu.virtual_environment:479(ensure_path) ERROR: C:\Users\Alan_\AppData\Local\python\mu\mu_venv-38-20210331-154537 does not exist.
2021-03-31 15:45:43,280 - mu.virtual_environment:445(ensure_and_create) DEBUG: Creating new virtual environment at C:\Users\Alan_\AppData\Local\python\mu\mu_venv-38-20210331-154543.
2021-03-31 15:45:43,280 - mu.virtual_environment:379(relocate) DEBUG: Virtual environment set up mu_venv-38-20210331-154543 at C:\Users\Alan_\AppData\Local\python\mu\mu_venv-38-20210331-154543
2021-03-31 15:45:43,280 - mu.virtual_environment:579(create) INFO: Creating virtualenv: C:\Users\Alan_\AppData\Local\python\mu\mu_venv-38-20210331-154543
2021-03-31 15:45:43,296 - mu.virtual_environment:580(create) INFO: Virtualenv name: mu_venv-38-20210331-154543
2021-03-31 15:45:46,350 - mu.virtual_environment:632(install_baseline_packages) INFO: Installing baseline packages.
2021-03-31 15:45:46,353 - mu.virtual_environment:633(install_baseline_packages) INFO: C:\Users\Alan_\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels exists
2021-03-31 15:46:30,172 - mu.virtual_environment:658(install_baseline_packages) DEBUG: Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\backcall-0.2.0-py2.py3-none-any.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\bitstring-3.1.7-py3-none-any.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cffi-1.14.5-cp38-cp38-win_amd64.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\click-7.1.2-py2.py3-none-any.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\colorama-0.4.4-py2.py3-none-any.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cryptography-3.4.7-cp36-abi3-win_amd64.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\decorator-4.4.2-py2.py3-none-any.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ecdsa-0.16.1-py2.py3-none-any.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\esptool-3.0-py3-none-any.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\flask-1.1.2-py2.py3-none-any.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ipykernel-5.5.0-py3-none-any.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ipython-7.22.0-py3-none-any.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ipython_genutils-0.2.0-py2.py3-none-any.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\itsdangerous-1.1.0-py2.py3-none-any.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jedi-0.18.0-py2.py3-none-any.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jinja2-2.11.3-py2.py3-none-any.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jupyter_client-6.1.12-py3-none-any.whl
Processing c:\users\alan_\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jupyter_core-4.7.1-py3-none-any.whl
[... cut ...]
2021-03-31 16:37:41,987 - mu.logic:332(read_and_decode) DEBUG: Detected newline '\n'
2021-03-31 16:37:41,987 - mu.logic:1077(_load) DEBUG: """
NeoPixel example for Pico. Displays a rainbow on the NeoPixels.

REQUIRED HARDWARE:
* RGB NeoPixel LEDs connected to pin GP0.
"""
import time
import board
import neopixel
from _pixelbuf import colorwheel

# Update this to match the number of NeoPixel LEDs connected to your board.
num_pixels = 30

pixels = neopixel.NeoPixel(board.GP0, num_pixels, auto_write=F

carlosperate, 2021-05-05:
Merging another crash report here to track this issue in a single location: https://github.com/mu-editor/mu/issues/1533

> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: es_BO
> Platform: Windows 10 10.0.19041 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\Geff\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 754, in on_ls_fail
>     self.disable()
>   File "C:\Users\Geff\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 693, in disable
>     self.microbit_fs.setDisabled(True)
> RuntimeError: wrapped C/C++ object of type MicroPythonDeviceFileList has been deleted
> ```

Thank you very much for the crash report @zuna-boop!
If you could attach your log file to this ticket it would be quite useful too, as that will help us gather more information on the issue. The log file should be located in: %LOCALAPPDATA%\python\mu\Logs

AlanLL, 2021-05-05:
Logs as requested.

Regards
AlanLL

carlosperate, 2021-05-24:
Thank you for the log again @AlanLL.
Having a look at it we can find this as the interesting part:

```
2021-03-31 16:14:43,594 - mu.modes.esp:214(toggle_files) INFO: Toggle filesystem on.
2021-03-31 16:14:47,788 - mu.modes.base:686(ls) ERROR: Could not enter raw REPL.
Traceback (most recent call last):
  File "C:\Users\Alan_\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\base.py", line 683, in ls
    result = tuple(microfs.ls(self.serial))
  File "C:\Users\Alan_\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 185, in ls
    out, err = execute(["import os", "print(os.listdir())"], serial)
  File "C:\Users\Alan_\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 139, in execute
    raw_on(serial)
  File "C:\Users\Alan_\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 100, in raw_on
    flush_to_msg(serial, raw_repl_msg)
  File "C:\Users\Alan_\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 72, in flush_to_msg
    raise IOError("Could not enter raw REPL.")
OSError: Could not enter raw REPL.
2021-03-31 16:14:47,794 - mu.interface.main:942(show_message) DEBUG: There was a problem getting the list of files on the device. Please check Mu's logs for technical information. Alternatively, try unplugging/plugging-in your device and/or restarting Mu.
2021-03-31 16:14:47,794 - mu.interface.main:943(show_message) DEBUG: None
2021-03-31 16:15:00,315 - mu.interface.main:942(show_message) DEBUG: Could not find an attached Raspberry Pi Pico
2021-03-31 16:15:00,315 - mu.interface.main:943(show_message) DEBUG: Please make sure the device is plugged into this computer.

The device must have MicroPython flashed onto it before the file system will work.

Finally, press the device's reset button and wait a few seconds before trying again.
2021-03-31 16:15:14,905 - mu.logic:720(check_usb) INFO: circuitpython device disconnected on port: COM5(VID: 0x239A, PID: 0x80F4, manufacturer Microsoft)
2021-03-31 16:15:15,316 - mu.interface.main:970(show_confirmation) DEBUG: Detected new Adafruit CircuitPlayground device.
2021-03-31 16:15:15,316 - mu.interface.main:971(show_confirmation) DEBUG: Would you like to change Mu to the CircuitPython mode?
2021-03-31 16:15:20,768 - mu.logic:737(check_usb) INFO: circuitpython device connected on port: COM5(VID: 0x239A, PID: 0x80F4, manufacturer: 'Microsoft')
2021-03-31 16:15:23,039 - root:174(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\Alan_\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 750, in on_ls_fail
    self.disable()
  File "C:\Users\Alan_\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 689, in disable
    self.microbit_fs.setDisabled(True)
RuntimeError: wrapped C/C++ object of type MicroPythonDeviceFileList has been deleted
```

It looks like in the Raspberry Pico (or ESP, hard to tell) mode the file button was pressed, but Mu could not enter the raw-REPL. Then the pico with CircuitPython was detected and a few milliseconds later Mu crashes with the `wrapped C/C++ object of type MicroPythonDeviceFileList has been deleted` error.

carlosperate, 2021-05-24:
Merging issue here https://github.com/mu-editor/mu/issues/1586:

[mu.log](https://github.com/mu-editor/mu/files/6531429/mu.log)
[mu.log.2021-05-22.log](https://github.com/mu-editor/mu/files/6531430/mu.log.2021-05-22.log)
> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: en_US
> Platform: Windows 10 10.0.19041 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 754, in on_ls_fail
>     self.disable()
>   File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 693, in disable
>     self.microbit_fs.setDisabled(True)
> RuntimeError: wrapped C/C++ object of type MicroPythonDeviceFileList has been deleted
> ```

This is the interesting part from the logs:

```
2021-05-22 16:25:17,996 - mu.modes.esp:214(toggle_files) INFO: Toggle filesystem on.
2021-05-22 16:25:23,434 - mu.modes.base:686(ls) ERROR: Could not enter raw REPL.
Traceback (most recent call last):
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\base.py", line 683, in ls
    result = tuple(microfs.ls(self.serial))
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 185, in ls
    out, err = execute(["import os", "print(os.listdir())"], serial)
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 139, in execute
    raw_on(serial)
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 100, in raw_on
    flush_to_msg(serial, raw_repl_msg)
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 72, in flush_to_msg
    raise IOError("Could not enter raw REPL.")
OSError: Could not enter raw REPL.
2021-05-22 16:25:23,439 - mu.interface.main:942(show_message) DEBUG: There was a problem getting the list of files on the device. Please check Mu's logs for technical information. Alternatively, try unplugging/plugging-in your device and/or restarting Mu.
2021-05-22 16:25:23,439 - mu.interface.main:943(show_message) DEBUG: None
2021-05-22 16:25:46,506 - mu.interface.main:942(show_message) DEBUG: Could not find an attached ESP8266/ESP32
2021-05-22 16:25:46,506 - mu.interface.main:943(show_message) DEBUG: Please make sure the device is plugged into this computer.

The device must have MicroPython flashed onto it before the file system will work.

Finally, press the device's reset button and wait a few seconds before trying again.
2021-05-22 16:26:11,511 - mu.logic:720(check_usb) INFO: esp device disconnected on port: COM4(VID: 0x10C4, PID: 0xEA60, manufacturer Silicon Laboratories)
2021-05-22 16:26:11,693 - mu.logic:737(check_usb) INFO: esp device connected on port: COM4(VID: 0x10C4, PID: 0xEA60, manufacturer: 'Silicon Laboratories')
2021-05-22 16:26:13,289 - root:173(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 754, in on_ls_fail
    self.disable()
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 693, in disable
    self.microbit_fs.setDisabled(True)
RuntimeError: wrapped C/C++ object of type MicroPythonDeviceFileList has been deleted
```

Looks like it first tried to open the ESP model files panel, it could not entre the raw-REPL mode and asked the user to unplug and replug the device. Then after doing that Mu crashes with the `wrapped C/C++ object of type MicroPythonDeviceFileList has been deleted` error.

carlosperate, 2021-05-24:
So both logs show a very similar set of steps.
We might be able to replicate this by having a device that cannot get to the raw-REPL and then disconnecting it and reconnecting it to the computer.

carlosperate, 2021-09-27:
Merging issue https://github.com/mu-editor/mu/issues/1794 here:

> [errorlog.txt](https://github.com/mu-editor/mu/files/7138023/errorlog.txt)
> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.5
> Locale: en_NL
> Platform: Windows 10 10.0.19041 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\Jan_J\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 762, in on_ls_fail
>     self.disable()
>   File "C:\Users\Jan_J\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 701, in disable
>     self.microbit_fs.setDisabled(True)
> RuntimeError: wrapped C/C++ object of type MicroPythonDeviceFileList has been deleted
> ```
>
> This concerns an esp8266 put the micropyton bin first with the esptool and ubuntu and thereafter with windows and the MU tool.
> Mu works fine with esp32 ( micropyton) and with esp32S2 circuitpython. esp8266 REPL produces garbage . On Ubuntu the result (Thonny) is oke.

carlosperate, 2022-01-25:
Merging https://github.com/mu-editor/mu/issues/2004 here:

> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0b7 Locale: de_DE Platform: Windows 10 10.0.19043 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\Administrator\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 760, in on_ls_fail
>     self.disable()
>   File "C:\Users\Administrator\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 699, in disable
>     self.microbit_fs.setDisabled(True)
> RuntimeError: wrapped C/C++ object of type MicroPythonDeviceFileList has been deleted
> ```
> I have the problem to access on files on ESP32-WROOM  with MuEditor in microPython from Windows. When I work with ESP32 from Raspberry then all is okay.
> Thanks for hint in advance
> Best regards
> Jurgen
> [mu.log](https://github.com/mu-editor/mu/files/7932403/mu.log)
> 
>