mu-editor/mu #1431: Crash Report

Filip-Konkowski, 2021-03-31:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: en_GB
Platform: Linux 5.0.0-32-generic #34~18.04.2-Ubuntu SMP Thu Oct 10 10:36:02 UTC 2019 x86_64

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/mu/virtual_environment.py", line 442, in ensure_and_create
    self.ensure()
  File "/usr/local/lib/python3.6/dist-packages/mu/virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "/usr/local/lib/python3.6/dist-packages/mu/virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: /home/filip/.local/share/mu/mu_venv-36-20210331-184439 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/mu/app.py", line 162, in run
    raise ex
  File "/usr/local/lib/python3.6/dist-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/usr/local/lib/python3.6/dist-packages/mu/virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "/usr/local/lib/python3.6/dist-packages/mu/virtual_environment.py", line 596, in create
    self.install_baseline_packages()
  File "/usr/local/lib/python3.6/dist-packages/mu/virtual_environment.py", line 649, in install_baseline_packages
    wheels.download()
  File "/usr/local/lib/python3.6/dist-packages/mu/wheels/__init__.py", line 92, in download
    check=True,
  File "/usr/lib/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'download', '--destination-directory', '/usr/local/lib/python3.6/dist-packages/mu/wheels', 'pgzero>=1.2.1']' returned non-zero exit status 2.

```
[mu.log](https://github.com/mu-editor/mu/files/6238557/mu.log)

Filip-Konkowski, 2021-03-31:
[mu.log](https://github.com/mu-editor/mu/files/6238563/mu.log)

carlosperate, 2021-04-07:
Thanks for the report @Filip-Konkowski!

Would you be able to run this in a terminal and copy/paste the output here?

```
/usr/bin/python3 -m pip download --destination-directory /usr/local/lib/python3.6/dist-packages/mu/wheels pgzero>=1.2.1
``` 

If you could also run this to check the pip version, that'd be great too:

```
/usr/bin/python3 -m pip --version
```

Filip-Konkowski, 2021-04-07:
Hey here are the outputs, let me know if I can somehow help.

/usr/bin/python3 -m pip download --destination-directory
/usr/local/lib/python3.6/dist-packages/mu/wheels pgzero>=1.2.1

Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/download.py", line
198, in run
    requirement_set.prepare_files(finder)
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 380,
in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 620,
in _prepare_file
    session=self.session, hashes=hashes)
  File "/usr/lib/python3/dist-packages/pip/download.py", line 821, in unpack_url
    hashes=hashes
  File "/usr/lib/python3/dist-packages/pip/download.py", line 667, in
unpack_http_url
    _copy_file(from_path, download_dir, link)
  File "/usr/lib/python3/dist-packages/pip/download.py", line 631, in _copy_file
    shutil.copy(filename, download_location)
  File "/usr/lib/python3.6/shutil.py", line 245, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.6/shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied:
'/usr/local/lib/python3.6/dist-packages/mu/wheels/pgzero-1.2.1-py3-none-any.whl'

so  I run it with by sudo /usr/bin/python3 -m pip download
--destination-directory
/usr/local/lib/python3.6/dist-packages/mu/wheels pgzero>=1.2.1

error on mu-editor:

Traceback (most recent call last):
  File "/home/filip/.local/lib/python3.6/site-packages/mu/app.py",
line 162, in run
    raise ex
  File "/home/filip/.local/lib/python3.6/site-packages/mu/app.py",
line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/filip/.local/lib/python3.6/site-packages/mu/virtual_environment.py",
line 460, in ensure_and_create
    "Unable to create a working virtual environment."
mu.virtual_environment.VirtualEnvironmentError: Unable to create a
working virtual environment.

/usr/bin/python3 -m pip --version

pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.6)

carlosperate, 2021-04-07:
Thanks @Filip-Konkowski, I think I understand the issue a little bit better now.

> so  I run it with by sudo /usr/bin/python3 -m pip download
--destination-directory
/usr/local/lib/python3.6/dist-packages/mu/wheels pgzero>=1.2.1

I assume running `sudo /usr/bin/python3 -m pip download
--destination-directory
/usr/local/lib/python3.6/dist-packages/mu/wheels pgzero>=1.2.1` worked?

Filip-Konkowski, 2021-04-07:
Yeah looks like it worked. I also update all system libs on my Linux but
the same error.

carlosperate, 2021-04-07:
Okay, in that case if you do `sudo /usr/bin/python3 -m pip install mu-editor`, does that work?

carlosperate, 2021-04-07:
Oh wait, can you first uninstall mu: `sudo /usr/bin/python3 -m pip uninstall mu-editor`

And then ensure the `/usr/local/lib/python3.6/dist-packages/mu/` directory has been completely removed? If it's still there, specially if there are .whl files inside the `/usr/local/lib/python3.6/dist-packages/mu/wheels` folder, make sure you remove them.

Then try to install Mu again: `sudo /usr/bin/python3 -m pip install mu-editor`

Filip-Konkowski, 2021-04-25:
Removed mu-editor I checked that removed
/usr/local/lib/python3.6/dist-packages/mu/

did sudo /usr/bin/python3 -m pip install mu-editor

got:
Could not find a version that satisfies the requirement qscintilla==2.11.4
(from mu-editor) (from versions: 2.9.4, 2.10, 2.10.1, 2.10.2, 2.10.3,
2.10.4, 2.10.7, 2.10.8, 2.11, 2.11.1, 2.11.2, 2.11.3, 2.12.0, 2.12.1)
No matching distribution found for qscintilla==2.11.4 (from mu-editor)
tried once again I to install

got:
Command "python setup.py egg_info" failed with error code 1 in
/tmp/pip-build-1vn_e6y2/pyqt5/

Not sure but it looks like something is wrong with my python3 :(

carlosperate, 2021-08-10:
Hi @Filip-Konkowski,

Sorry it took me so long to reply.
` sudo /usr/bin/python3 -m pip install mu-editor` should have been ` sudo /usr/bin/python3 -m pip install --pre mu-editor`

Could you try to install it again, this time it should pick up the latest beta.5, that hopefully should have fixed some of these issues.

Filip-Konkowski, 2021-08-10:
Uh I completely forgot about my for Linux. Now I. Using it on Windows
machine. When I will have time I will try it and let you know.

carlosperate, 2021-08-10:
No problem, if you could try again when you have a chance that would be appreciated. Thanks!

carlosperate, 2021-09-24:
Hi @Filip-Konkowski, just wondering if you had a chance to give this a quick test? Thanks!

Filip-Konkowski, 2021-09-24:
Hey

I will maybe try next week

carlosperate, 2021-12-17:
Hi @Filip-Konkowski, are you still experiencing this issue?