mu-editor/mu #1434: can not install package via "pip

XcHero, 2021-04-02:
when i use "pip" to install package in "REPL",show info as picture.
[image]
OS info:Mac OS 10.15.6;
mu version :1.1.0-beta.3.

ajs256, 2021-04-02:
Looks like the `Application Support` directory name needs to be escaped.

carlosperate, 2021-04-08:
Thanks for the report @XcHero!
 
This is an issue in iPython, so I've opened this issue in their repository: https://github.com/ipython/ipython/issues/12896

Btw, the best way to install packages in Mu is via the settings:
[image]