mu-editor/mu #1435: Crash Report

apostate83, 2021-04-02:
[mu.log](https://github.com/mu-editor/mu/files/6251311/mu.log)
[mu.log](https://github.com/mu-editor/mu/files/6251318/mu.log)

**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: en_US
Platform: Windows 7 6.1.7601 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Owner\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\Owner\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Owner\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 459, in ensure_and_create
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.

```

apostate83, 2021-04-02:
I was trying to install the MU editor and it wouldn't install.  I had already successfully updated the bootloader and circuitpython to the latest revisions.  Also, since I'm running Windows 7, I installed the driver package.  But after all of that, I wasn't able to install the MU editor.  Thanks much :)

carlosperate, 2021-05-11:
Thank you for the report and the log @apostate83!

Looks like we haven't been able to capture the main cause of the problem in the logs. Can you try to run these commands in a "command prompt" (you can open one by searching for `cmd.exe` in the windows start menu) and copy/paste the result here?

```
dir C:\Users\Owner\AppData\Local\python\mu\
``` 

```
C:\Users\Owner\AppData\Local\python\mu\mu_venv-38-20210402-020531\scripts\python.exe -c import pgzero
```

```
C:\Users\Owner\AppData\Local\python\mu\mu_venv-38-20210402-020531\scripts\pip.exe --version
```

While Windows 7 is no longer supported by MS it looks like Python 3.8, the version used in the installers, should still work.  Windows 7 has only been dropped in 3.9: https://www.python.org/downloads/windows/

carlosperate, 2021-08-10:
Hi @apostate83,

Would you be able to check if beta.5 works for you and let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

carlosperate, 2021-12-17:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

closed 2021-12-17 by carlosperate