mu-editor/mu #1436: Crash Report

aniljaji, 2021-04-04:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: en_IN
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\anilj\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\anilj\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\anilj\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 459, in ensure_and_create
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.

```

carlosperate, 2021-04-06:
Hi @aniljaji,

Would you be able to also attach the log file to this GitHub issue? It will be really helpful, file should be located in: `%LOCALAPPDATA%\python\mu\Logs`

carlosperate, 2021-05-11:
Hi @aniljaji, just a friendly reminder in case you'd be able to attach the logs? They would be really helpful for us, and you should be able to find them in `%LOCALAPPDATA%\python\mu\logs`.

carlosperate, 2021-08-10:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
Thanks for sending the crash report!

closed 2021-08-10 by carlosperate