mu-editor/mu #1438: Crash Report

Jak7, 2021-04-05:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Mr. Gus Fring\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 442, in ensure_and_create
    self.ensure()
  File "C:\Users\Mr. Gus Fring\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "C:\Users\Mr. Gus Fring\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: C:\Users\Mr. Gus Fring\AppData\Local\python\mu\mu_venv-38-20210405-093754 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Mr. Gus Fring\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\Mr. Gus Fring\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Mr. Gus Fring\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "C:\Users\Mr. Gus Fring\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 583, in create
    subprocess.run(
  File "C:\Users\Mr. Gus Fring\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['C:\\Users\\Mr. Gus Fring\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-m', 'virtualenv', '-p', 'C:\\Users\\Mr. Gus Fring\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-q', 'C:\\Users\\Mr. Gus Fring\\AppData\\Local\\python\\mu\\mu_venv-38-20210405-094000']' returned non-zero exit status 1.

```

ZanderBrown, 2021-04-06:
I wonder if this is the spaces-in-username issue back again, though that shouldn't be a problem in beta.3

Looks like you forgot to attach your log, this often contains information we find rather helpful :-)

tjguk, 2021-04-06:
> I wonder if this is the spaces-in-username issue back again, though that shouldn't be a problem in beta.3

Unfortunately, the tentative fix for the spaces issue had to be backed out of the release late in the day as it caused other problems elsewhere. I would bet that this is the same issue -- although I'm frustratingly unable to reproduce it.

@Jak7 -- if you're in a position to do so, can you confirm that the issue is solved by using a different username without a space in it please?