mu-editor/mu #1439: Crash Report

cristinaolivotto, 2021-04-06:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: fr_CH
Platform: Linux 4.15.0-140-generic #144-Ubuntu SMP Fri Mar 19 14:12:35 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/onlfait/.local/lib/python3.6/site-packages/mu/app.py", line 162, in run
    raise ex
  File "/home/onlfait/.local/lib/python3.6/site-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/onlfait/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 460, in ensure_and_create
    "Unable to create a working virtual environment."
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.

```

carlosperate, 2021-04-06:
Thanks for the report @cristinaolivotto!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

carlosperate, 2021-05-11:
Hi @cristinaolivotto, just a friendly reminder in case you'd be able to attach the logs? They would be really helpful for us, and you should be able to find them in `~/.cache/mu`.

carlosperate, 2021-08-10:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
Thanks for sending the crash report!

closed 2021-08-10 by carlosperate