mu-editor/mu #1440: Crash Report

mael198, 2021-04-06:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: fr_CA
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\tauu1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\python3.py", line 197, in run_toggle
    self.stop_script()
  File "C:\Users\tauu1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\python3.py", line 259, in stop_script
    self.runner.process.kill()
RuntimeError: wrapped C/C++ object of type QProcess has been deleted

```

carlosperate, 2021-04-06:
Thanks for the report @mael198!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `%LOCALAPPDATA%\python\mu\Logs`

mael198, 2021-04-07:
I explain my problem to you I want to operate my Nfc copy cat of electronics cats. but i'm starting i'm a little lost with arduino circuitpython uf2 bootloader file adafruit playground express which is the card that we must finally take i believe. how to insert uf2 file in circuitpython or arduino ide please help me  Provenance : Courrier pour Windows 10 De : Carlos Pereira AtencioEnvoyé le :6 avril 2021 13:32À : mu-editor/muCc : mael198; MentionObjet :Re: [mu-editor/mu] Crash Report (#1440) Thanks for the report @mael198!Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: %LOCALAPPDATA%\python\mu\Logs

carlosperate, 2021-04-07:
Hi @mael198,

I'm not sure I understand your last email, I think it also got a little scrambled (you can see it in this URL https://github.com/mu-editor/mu/issues/1440#issuecomment-814519312 ).

Does the Mu editor work for you now?
If not, could you copy/paste the log file (`%LOCALAPPDATA%\python\mu\Logs`) into the GH issue? Unfortunately email attachments don't work, so it'll be best to reply from the GitHub website.

Thanks again for the report!

carlosperate, 2021-05-11:
Hi @mael198,

Did you still have this issue? Would you be able to clarify the last message and send us the logs? They would be really useful for us and you should be able to find them in `%LOCALAPPDATA%\python\mu\logs`.

carlosperate, 2022-06-10:
Closing due to lack of activity.

closed 2022-06-10 by carlosperate