mu-editor/mu #1441: runninng mu-editor with pip version <10 doesn't work (at least in Linux)

carlosperate, 2021-04-06:
Not 100% sure what feature that was added to pip v10 is required, but this minimum version is needed to be able to install the `cryptography` package  (which has recently moved to Rust) on Linux, which is needed for `esptool`:

With pip v9, which is the default in Ubuntu 18.04:
```
Collecting esptool==3.*
  File was already downloaded /home/carlos/workspace/mu/mu/wheels/esptool-3.0.tar.gz
Collecting bitstring>=3.1.6 (from esptool==3.*)
  File was already downloaded /home/carlos/workspace/mu/mu/wheels/bitstring-3.1.7.tar.gz
Collecting cryptography>=2.1.4 (from esptool==3.*)
  File was already downloaded /home/carlos/workspace/mu/mu/wheels/cryptography-3.4.7.tar.gz
    Complete output from command python setup.py egg_info:
    
            =============================DEBUG ASSISTANCE==========================

            If you are seeing an error here please try the following to
            successfully install cryptography:
    
            Upgrade to the latest pip and try again. This will fix errors for most
            users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
            =============================DEBUG ASSISTANCE==========================
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-bjn709ho/cryptography/setup.py", line 14, in <module>
        from setuptools_rust import RustExtension
    ModuleNotFoundError: No module named 'setuptools_rust'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-bjn709ho/cryptography/
```

With pip v10:

```
Collecting esptool==3.*
  File was already downloaded /home/carlos/workspace/mu/mu/wheels/esptool-3.0.tar.gz
Collecting bitstring>=3.1.6 (from esptool==3.*)
  File was already downloaded /home/carlos/workspace/mu/mu/wheels/bitstring-3.1.7.tar.gz
Collecting cryptography>=2.1.4 (from esptool==3.*)
  File was already downloaded /home/carlos/workspace/mu/mu/wheels/cryptography-3.4.7.tar.gz
  Installing build dependencies ... done
Collecting ecdsa>=0.16.0 (from esptool==3.*)
  Downloading https://files.pythonhosted.org/packages/98/16/70be2716e24eaf5d81074bb3c05429d60292c2a96613a78ac3d69526ad2a/ecdsa-0.16.1-py2.py3-none-any.whl (104kB)
    100% |████████████████████████████████| 112kB 3.7MB/s 
  Saved ./mu/wheels/ecdsa-0.16.1-py2.py3-none-any.whl
Collecting pyserial>=3.0 (from esptool==3.*)
  File was already downloaded /home/carlos/workspace/mu/mu/wheels/pyserial-3.5-py2.py3-none-any.whl
Collecting reedsolo<=1.5.4,>=1.5.3 (from esptool==3.*)
  Downloading https://files.pythonhosted.org/packages/c8/cb/bb2ddbd00c9b4215dd57a2abf7042b0ae222b44522c5eb664a8fd9d786da/reedsolo-1.5.4.tar.gz (271kB)
    100% |████████████████████████████████| 276kB 19.5MB/s 
  Saved ./mu/wheels/reedsolo-1.5.4.tar.gz
Collecting cffi>=1.12 (from cryptography>=2.1.4->esptool==3.*)
  Using cached https://files.pythonhosted.org/packages/2b/cf/9a3b04e57191a970836aeaa8b2075574f02fbdb65d6368457a2f13213e7f/cffi-1.14.5-cp36-cp36m-manylinux1_x86_64.whl
  Saved ./mu/wheels/cffi-1.14.5-cp36-cp36m-manylinux1_x86_64.whl
Collecting six>=1.9.0 (from ecdsa>=0.16.0->esptool==3.*)
  File was already downloaded /home/carlos/workspace/mu/mu/wheels/six-1.15.0-py2.py3-none-any.whl
Collecting pycparser (from cffi>=1.12->cryptography>=2.1.4->esptool==3.*)
  Using cached https://files.pythonhosted.org/packages/ae/e7/d9c3a176ca4b02024debf82342dab36efadfc5776f9c8db077e8f6e71821/pycparser-2.20-py2.py3-none-any.whl
  Saved ./mu/wheels/pycparser-2.20-py2.py3-none-any.whl
Successfully downloaded esptool bitstring cryptography ecdsa pyserial reedsolo cffi six pycparser
You are using pip version 10.0.0, however version 21.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

Don't think we can specify pip version in setup.py (and even if we could, it would likely not work with other pip versions). Should we consider manually checking the version in the code and throw an exception with a useful message if it's below 10?

Related crash reports:
- https://github.com/mu-editor/mu/issues/1409
- https://github.com/mu-editor/mu/issues/1408
- https://github.com/mu-editor/mu/issues/1391

tjguk, 2021-04-07:
Thanks, @carlosperate  -- I was aware that we had some kind of common esptool-related issue going on, but since it wasn't hitting everyone I wasn't sure where to look first.

Couple of thoughts here:

I think we should run a "pip update pip" when we create the venv. (I always do this for my own and I seem to remember coding this into the mu module, but perhaps I removed it..). That's assuming that the various Linux-distros haven't neutered pip to the extent that it can't upgrade itself...

Also I think we could/should make the use of esptool conditional. It's an optional tool for an optional mode. Probably an additional flag on the `mode_packages` list at the head of `wheels.__main__`. If we failed to download/test-import an optional module it would simply log a warning and move on.

carlosperate, 2021-04-07:
One correction to my original message. This issue happens at the `pip download` stage, not `pip install`.

> I think we should run a "pip update pip" when we create the venv.

If you mean updating pip inside the venv, then this is already done by virtualenv, which is one of the reasons I prefer it over venv.
Also because it's able to do that update "safely", as there are cases where it can go wrong. For example with Python 3.5 and an old version of pip it will update to the latest (before the `python_requires` attribute is respected), which uses f-strings and then pip fails to run, which also means you can't use pip to downgrade itself. Things like that are taken care of with virtualenv.

However, in this case since we are using the pip module from the interpreter running Mu, and this could be the system pip if the user has not created a virtual environment to `pip install mu-editor`.

> That's assuming that the various Linux-distros haven't neutered pip to the extent that it can't upgrade itself...

I wouldn't be surprised if we find that somewhere, but so far I don't think I've encountered.

However, I wouldn't touch the system pip, I'd be really pissed off if an application I've installed messes up my Python environment.
Yes, we should encourage good Python practices with venvs, but we should never break ~~userspace~~ system Python/pip.

> Also I think we could/should make the use of esptool conditional. It's an optional tool for an optional mode. Probably an additional flag on the mode_packages list at the head of wheels.__main__. If we failed to download/test-import an optional module it would simply log a warning and move on.

This sounds sensible, specially since we left the old message in the settings in the case the module is not found in the venv.
There will be a few extra things to consider, like do we want to fail the packaging step if it cannot package the wheel? Will these optional modules appear in the settings window when installed (currently they are filtered out)? 
We will also probably need to refine the UX about explaining to the user why/how to enable that feature.

carlosperate, 2021-04-07:
As an additional note, the pip min version to download the cryptography wheel correctly is v10, but the min pip version to be able to install the available cryptography wheels correctly (without defaulting to downloading the source and trying to compile a rust project) is **v19.0**, as this is the first version to implement `manylinux2010`.

However, the PyQt wheels have stopped creating `manylinux1` wheels from v5.14 (we are still using v5.13), so as soon as we update that we'll require pip **v19.3**.

carlosperate, 2021-05-04:
We could control the pip version inside the venv by ensuring we require a minimum version of `virtualenv` in setup.py.

closed 2021-08-04 by carlosperate