mu-editor/mu #1442: Crash Report

grethe-ladegaard, 2021-04-07:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: nb_NO
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\grethela\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 442, in ensure_and_create
    self.ensure()
  File "C:\Users\grethela\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "C:\Users\grethela\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: C:\Users\grethela\AppData\Local\python\mu\mu_venv-38-20210407-084650 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\grethela\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 299, in installed
    next(iterlines)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\grethela\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\grethela\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\grethela\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "C:\Users\grethela\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 598, in create
    self.register_baseline_packages()
  File "C:\Users\grethela\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 665, in register_baseline_packages
    packages = list(self.pip.installed())
  File "C:\Users\grethela\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 305, in installed
    raise VirtualEnvironmentError("Unable to parse installed packages")
mu.virtual_environment.VirtualEnvironmentError: Unable to parse installed packages

```

tjguk, 2021-04-07:
Thanks for the crash report, @grethe-ladegaard -- are you in a position to upload the logs, please? They should be at:

`%LOCALAPPDATA%\python\mu\Logs`

That will help us to narrow down the issue. Thanks

carlosperate, 2021-05-11:
Hi @grethe-ladegaard, just a friendly reminder in case you'd be able to attach the logs? They would be really helpful for us, and you should be able to find them in `%LOCALAPPDATA%\python\mu\logs`.

carlosperate, 2021-08-10:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
Thanks for sending the crash report!

closed 2021-08-10 by carlosperate