mu-editor/mu #1449: Problem with opening the Mu editor

katarzynaozik, 2021-04-08:
I have problem with running the new version of Editor Mu under Win10. I have uninstalled the previous version, and install the new one Mu-Editor-Win64-1.1.0b3.msi and it does not start. What can I do? 
Maybe there is conflict with previous version which was installed for all users (I have unistalled it!)- this one is only for one user.

There is no error message on the screen.

carlosperate, 2021-04-08:
Hi @katarzynaozik, thanks for the report!

Hopefully there shoudn't be any conflict with previously installed editors, but we will need the log file to have a better look.

Can you attempt to one  the Mu editor, and immediately after it crashes send us the log?
It should be located in `%LOCALAPPDATA%\python\mu\Logs`

katarzynaozik, 2021-04-08:
But I do not have the folder named **python** in localization C:\Users\katarzyna\AppData\Local.

I also tried to install in new Win account and it works fine. So there is something wrong with my account.

carlosperate, 2021-04-09:
Thanks for the update @katarzynaozik. So with a new user account Mu works fine, but with the old one (possibly named `katarzyna`) it has issues, is that right?

In the account with issues, how did the problem manifest? When trying to open the Mu editor, did something opened at all? did you see a splash screen like this one?
[image]

katarzynaozik, 2021-04-12:
Yes, for the new account katarzyna2, it works, but the old one - katarzyna it doesn't. I cannot see a splash screen.
I even tried to run by command from mu folder: pythonw -m mu. There are no error messages.

carlosperate, 2021-04-12:
That's really odd, could you try uninstalling and reinstalling the Mu app in the "bad" account? Maybe something went wrong during the installation that stops Mu from launching.
Do you have any antivirus software, or something along those lines, that could be interfering?

katarzynaozik, 2021-04-13:
Success! I delete a folder in Roaming directory (C:\Users\katarzyna\AppData\Roaming) with Python. Now everything is ok. Thanks for help.

By the way is it possible to install Mu Editor for all users? It is more for convenient for using at school with different accounts?

carlosperate, 2021-04-13:
I'm glad you resolved it!
I've opened this issue to track that feature request, thanks @katarzynaozik!
https://github.com/mu-editor/mu/issues/1473

closed 2021-04-13 by carlosperate