mu-editor/mu #1451: Crash Report

zeromancer91, 2021-04-08:
[mu.log](https://github.com/mu-editor/mu/files/6280869/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: en_US
Platform: Linux 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l

```
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mu/modes/python3.py", line 109, in start_kernel
    self.repl_kernel_manager.start_kernel()
  File "/home/pi/.local/lib/python3.7/site-packages/mu/modes/python3.py", line 42, in start_kernel
    kernel_cmd, kw = self.pre_start_kernel(**kw)
AttributeError: 'MuKernelManager' object has no attribute 'pre_start_kernel'

```

carlosperate, 2021-04-10:
Thanks for the report @zeromancer91, could you provide a bit more info about your system? Like version and platform of your Linux distribution.
Looks like you might be running this from a Raspberry Pi, what installation instructions did you follow?
Could you also copy/paste the output of this terminal command?

```
python3 -m pip list
```

Thanks again!

carlosperate, 2021-04-28:
Hi @zeromancer91, just a friendly reminder if you could provide the info and run the command from my previous message?
It would really help us debug this issue to ensure it gets fixed in future releases.

Thanks!

zeromancer91, 2021-04-29:
this is what i get when using that command. now when i try to press the
REPL button i just get a black screen and cannot get it to change.

Package            Version
------------------ --------------
appdirs            1.4.3
asn1crypto         0.24.0
astroid            2.1.0
asttokens          1.1.13
automationhat      0.2.0
beautifulsoup4     4.7.1
black              20.8b1
blinker            1.4
blinkt             0.1.2
buttonshim         0.0.2
Cap1xxx            0.1.3
certifi            2018.8.24
chardet            3.0.4
click              7.1.2
colorama           0.3.7
colorzero          1.1
cookies            2.2.1
cryptography       2.6.1
cupshelpers        1.0
cycler             0.10.0
decorator          4.3.0
devscripts         2.19.5+deb10u1
distlib            0.3.1
docutils           0.14
drumhat            0.1.0
entrypoints        0.3
envirophat         1.0.0
ExplorerHAT        0.4.2
filelock           3.0.12
flake8             3.9.0
Flask              1.0.2
fourletterphat     0.1.0
gpiozero           1.5.1
guizero            0.6.0
html5lib           1.0.1
idna               2.6
importlib-metadata 3.10.0
ipykernel          4.9.0
ipython            5.8.0
ipython-genutils   0.2.0
isort              4.3.4
itsdangerous       0.24
jedi               0.13.2
Jinja2             2.10
jupyter-client     5.2.3
jupyter-core       4.4.0
keyring            17.1.1
keyrings.alt       3.1.1
kiwisolver         1.0.1
lazy-object-proxy  1.3.1
logilab-common     1.4.2
lxml               4.3.2
MarkupSafe         1.1.0
matplotlib         3.0.2
mccabe             0.6.1
meson              0.52.1
microdotphat       0.2.1
mote               0.0.4
motephat           0.0.3
mu-editor          1.1.0b3
mypy               0.670
mypy-extensions    0.4.3
nudatus            0.0.4
numpy              1.16.2
oauthlib           2.1.0
olefile            0.46
pantilthat         0.0.7
parso              0.3.1
pathspec           0.8.1
pexpect            4.6.0
pgzero             1.2
phatbeat           0.1.1
pianohat           0.1.0
picamera           1.13
pickleshare        0.7.5
picraft            1.0
piglow             1.2.5
pigpio             1.78
Pillow             5.4.1
pip                18.1
prompt-toolkit     1.0.15
psutil             5.5.1
pycairo            1.16.2
pycodestyle        2.7.0
pycrypto           2.6.1
pycups             1.9.73
pyflakes           2.3.1
pygame             1.9.4.post1
Pygments           2.3.1
PyGObject          3.30.4
pyinotify          0.9.6
PyJWT              1.7.0
pylint             2.2.2
pyOpenSSL          19.0.0
pyparsing          2.2.0
pyserial           3.4
pysmbc             1.0.15.6
python-apt         1.8.4.3
python-dateutil    2.7.3
pyudev             0.21.0
pyxdg              0.25
pyzmq              17.1.2
qtconsole          4.7.4
QtPy               1.9.0
rainbowhat         0.1.0
regex              2021.3.17
reportlab          3.5.13
requests           2.21.0
requests-oauthlib  1.0.0
responses          0.9.0
roman              2.0.0
RPi.GPIO           0.7.0
RTIMULib           7.2.1
scrollphat         0.0.7
scrollphathd       1.2.1
SecretStorage      2.3.1
semver             2.13.0
Send2Trash         1.5.0
sense-emu          1.1
sense-hat          2.2.0
setuptools         40.8.0
shortcut           0.0.2
simplegeneric      0.8.1
simplejson         3.16.0
six                1.12.0
skywriter          0.0.7
sn3218             1.2.7
soupsieve          1.8
spidev             3.4
ssh-import-id      5.7
thonny             3.3.6
toml               0.10.2
tornado            5.1.1
touchphat          0.0.1
traitlets          4.3.2
twython            3.7.0
typed-ast          1.4.2
typing-extensions  3.7.4.3
uflash             1.2.4
unicornhathd       0.0.4
urllib3            1.24.1
virtualenv         20.4.3
wcwidth            0.1.7
webencodings       0.5.1
Werkzeug           0.14.1
wheel              0.32.3
wrapt              1.10.11
zipp               3.4.1

carlosperate, 2021-05-04:
Thanks @zeromancer91! This looks related to issue https://github.com/mu-editor/mu/issues/1516 and PR https://github.com/mu-editor/mu/pull/1517.

The easiest way to fix this would be to update the `jupyter_client` package to a newer version. However, since the reason you have an old version is because some other application in your computer is using it, there is a risk that updating it could cause issues in that other application.

You can install Mu in a virtual environment, this isolates the Mu installation, so that its dependencies are not combined with the system packages. 
The main inconvenience from using a virtual environment is that it has to be "installed" somewhere (a folder where all the dependencies are stored), and then every time you want to use Mu you'll have to "activate" it.

These are the steps:
1) In a terminal, navigate to a directory where you want to install the virtual environment
2) Run this to create the virtual environment in a folder named `mu-venv`: `virtualenv --python=python3 mu-venv`
    - If you are using a Raspberry Pi then it needs to be `virtualenv --python=python3 mu-venv --system-site-packages` instead (this adds the extra `--system-site-packages` flag)
3) Run this to activate it: `source mu-venv/bin/activate`
4) Run this to install Mu in the virtual environment: `pip install mu-editor==1.1.0b4`
5) Run Mu from the virtual environment: `mu-editor`

Every time you want to run Mu you'd have to do steps 1, 3, and 5 (2 and 4 are only required to install it).

Alternatively I like to use the pipx tool: https://pipxproject.github.io/pipx/
If you install it you can then use pipx to automatically install Python applications in their own isolated virtual environment, and they can be run without manually activating the environment:
- Install pipx
    - Update pip: `python3 -m pip install --upgrade pip` 
    - Run: `python3 -m pip install --user pipx`
    - Run: `python3 -m pipx ensurepath`
- After installation you might need to close all your terminals and open a new one (or maybe even reboot, the last command should print out information)
- Now that is pipx is installed and configured you can use it to install Mu: `pipx install mu-editor==1.1.0b4`
    - If you are using a Raspberry Pi this command should be `pipx install mu-editor==1.1.0b4 --system-site-packages` instead
- And you can run Mu: `mu-editor`

carlosperate, 2021-06-23:
Hi @zeromancer91, I was wondering if you managed to resolve this issue?

carlosperate, 2021-08-10:
Hi @zeromancer91,

We've updated the jupyter requirements in the latest betas. Would you be able to check if beta.5 works for you and let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!