mu-editor/mu #1452: Rapport d’accident

mehdoui983, 2021-04-08:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.2
Locale: fr_FR
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\MEHDOUI MONDHER\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\MEHDOUI MONDHER\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\MEHDOUI MONDHER\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 441, in ensure_and_create
    self.ensure()
  File "C:\Users\MEHDOUI MONDHER\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 469, in ensure
    self.ensure_interpreter_version()
  File "C:\Users\MEHDOUI MONDHER\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 522, in ensure_interpreter_version
    process = subprocess.run(
  File "C:\Users\MEHDOUI MONDHER\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['C:\\Users\\MEHDOUI MONDHER\\AppData\\Local\\python\\mu\\mu_venv-38-20210328-002638\\scripts\\python.exe', '-c', 'import sys; print("%s%s" % sys.version_info[:2])']' returned non-zero exit status 101.

```

tjguk, 2021-04-09:
This looks like an example of https://github.com/mu-editor/mu/issues/1372 -- which I can't currently reproduce (but which is clearly affecting people!)

@mehdoui983 -- are you in a position to try Mu on a user account *without* a space in the username?

carlosperate, 2022-06-06:
Hi @mehdoui983

We think we've got a fix for this issue. Would you be able to check if this version of Mu works for you?
If you could test this it would be really useful to us, to confirm everything is working correctly before we release it.

The installer can be downloaded the bottom of this page, it's be the middle file named `mu-editor-Windows-ab489a0c09e29a55bb7a158f380a10789eb6e2d5` (you need to be logged into GitHub for the download link to work):
https://github.com/mu-editor/mu/actions/runs/2444128277
[image]

Thank you!